oliexdev/openScale #448: Crush aftrer update from f=droid

chenl, 2019-03-28:
Build version: 2.0.2 
Build date: 1979-11-30 02:00:00 
Current date: 2019-03-28 06:18:23 
Device: OnePlus A0001 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2778)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
	at android.app.ActivityThread.-wrap11(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:686)
	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2057)
	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1847)
	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1802)
	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1709)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2609)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1334)
	at android.app.Activity.performStart(Activity.java:7029)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
	... 9 more

jvisionsdev, 2019-03-28:
I've got the same issue andere the same stack trace on Android 8.1.0.

Build version: 2.0.2 
Build date: 1979-11-30 01:00:00 
Current date: 2019-03-28 06:00:30 
Device: Samsung SM-J530F

hirnsushi, 2019-03-28:
I'm experiencing the same behaviour:

Build version: 2.0.2 
Build date: 1979-11-30 01:00:00 
Current date: 2019-03-28 06:06:06 
Device: LGE Nexus 4 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2778)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
	at android.app.ActivityThread.-wrap11(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:686)
	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2057)
	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1847)
	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1802)
	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1709)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2609)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1334)
	at android.app.Activity.performStart(Activity.java:7029)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
	... 9 more

hirnsushi, 2019-03-28:
Sorry, didn't check solved issues, dev version works.

steko, 2019-03-29:
Should be already fixed as per #446

jvisionsdev, 2019-04-01:
I updated to version 2.0.3 via F-Droid this morning.
This error seams to be fixed. Well done OliE.

Lesson learned (one more time): I have to configure scheduled backups to export all the values into another database.

oliexdev, 2019-04-01:
@jvisionsdev you could enable the auto backup function in openScale.

closed 2019-04-01 by oliexdev

chenl, 2019-04-09:
The new version 2.0.3 posted on f-droid fixed this issue.