oliexdev/openScale #451: Senssun Scale still broken

quthla, 2019-03-29:
**Describe the bug**
Very often it will only take a measurement after multiple failed tries and it changes gender on scale to female when it takes a measurement then.

1.9.3 still without issues.

**To Reproduce**
Steps to reproduce the behavior:

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.0.3

**Expected behavior**

**Additional context**

**Debug log**

oliexdev, 2019-03-30:
@quthla I am wondering you wrote it works in https://github.com/oliexdev/openScale/pull/427#issuecomment-470214096 Why don't you report the problem back then? I don't own this scale, so if nobody complains I guessed it works.

quthla, 2019-03-30:
I only tried it once back then and there was no issue but I was still using 1.9.3 until I upgraded few days ago and noticed the issues.

Right now I'm back at 1.9.3 again

oliexdev, 2019-04-22:
ok then I will close this issue

closed 2019-04-22 by oliexdev

quthla, 2019-05-01:
Just tried again. Made me a woman once more. The fat measurements etc are obviously completely wrong then.

1.9.3 works without issues. 

@voyager @gmarco can you confirm these issues with your Senssun Scale on 2.0.3?

voyager, 2019-05-01:
@quthla I don't have those issues with v2.x.x, never actually had them.
I switched a couple of days ago to the Pro version, but I don't think that makes any difference.

quthla, 2019-05-01:
@oliexdev There seems to be an issue with https://github.com/oliexdev/openScale/blob/v1.9.3/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothSenssun.java#L64

2.0.3: `BluetoothCommunication: Invoke write bytes [A5 10 01`
1.9.3: `BluetoothCommunication: Write characteristic 0xfff2: A5 10 F1`

https://github.com/oliexdev/openScale/blob/v1.9.3/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothSenssun.java#L70
F1 = male, 01 = female

My user in openScale is set to male.

Can you fix openScale so it sends correct gender byte?