oliexdev/openScale #452: Feature request: Option to restrict rolling graph timespan

hirnsushi, 2019-03-30:
**Is your feature request related to a problem? Please describe.**
I really love the new graph engine, but having ~2 years of daily measurements in the database makes the rolling graph kind of unwieldy to use. Scrolling back several years in daily view is rarely useful and increases loading time. Having lost weight in the past, the weekly data view gets squashed down to almost a flat line, leaving lots of whitespace.

**Describe the solution you'd like**
Implement an option to restrict the timespan used for the rolling graph, maybe default to a month for daily view, 3 month for weekly view?

**Describe alternatives you've considered**
Restrict based on # of measurements, cap the rolling graph at say 50 or 100 db entries.

**Additional context**
Example of squashed weekly view:
[image]

oliexdev, 2019-04-22:
Yeah I know the problem I already tried to fix it but I was not successful with the library. If anybody has a working solution I would be grateful for a PR.

oliexdev, 2019-05-12:
the squashed graph line should be fixed with https://github.com/oliexdev/openScale/commit/00fc884590af2584698dd2bfc3593fa801763788 please try out the latest dev version.

closed 2019-05-12 by oliexdev

hirnsushi, 2019-05-12:
Works great! 🙂👍