oliexdev/openScale #454: SBF70 Unexpected bluetooth error / can write to characteristic / new with 2.0.3pro

lenucksi, 2019-03-31:
**Describe the bug & How to reproduce**
- Have a SBF-75, changed from 2.0.3 to 2.0.3 pro from the Google Store as mandated by the 2.0.3 version
- Everything worked with 2.0.2 of the app.
- Moved over to 2.0.3pro, recreated the user (which should really have been automatically migrated during this change), imported a backup (needed two import attempts) and tried to sync historic measurements still in the scale.
- OpenScale asked for location permission and got it.
- Scale connected successfully, attempted to fetch measures, however always fails with a toast showing this: ```Bluetooth unexpected error: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb``` and a second toast being hidden by the first and thus unreadable.
- This is independent of activated location services, it fails with services on and off.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.0.3pro is equivalent with regards to code as of today according to committ log.

**Expected behavior**
Measures should be received without failure.

**Debug log**
Is attached
[openScale_2019-03-31_15-16.txt](https://github.com/oliexdev/openScale/files/3026631/openScale_2019-03-31_15-16.txt)
[openScale_2019-03-31_15-17.txt](https://github.com/oliexdev/openScale/files/3026632/openScale_2019-03-31_15-17.txt)

ptr-jul, 2019-04-03:
I am experiencing the exact same issue on SBF70.
If you need any further debug logs than the ones lenucksi has already provided, please let me know.

nazco, 2019-04-08:
I'm experiencing the same problem too.

lucaga, 2019-04-16:
The scale and the app worked in the past. With the new version (2.0.3-pro 44) the scale can be connected but data is not being delivered

nazco, 2019-04-17:
For me synchronisation works again with 2.0.3 (44). It still fells kind of very slow, but it works again.

oliexdev, 2019-04-22:
> Everything worked with 2.0.2 of the app.

There is no difference between 2.0.2 and 2.0.3 regarding to Bluetooth, so its strange that its stopped working!?

> Might be related to #433

It is related to issue #433 there I also stated that this is probably a timing problem of the scale/app. Sometimes it works and sometimes not. I bought this scale one time and it works perfectly but it seems some smartphones and/or Android version doesn't work well with the SBF70 scale but it is more a really bad implementation on the scale side.

So idk how to resolve this problem... feel free to send a PR!

lenucksi, 2019-04-23:
Thanks for the insight, I've given the scale another go yesterday and today and it imported measurements (albeit very slowly) on version 44. I think that the issues around this scale are definitely often a case of "bad hardware implementation".
Maybe it would be worthwhile adding this to the scale's Wiki entry combined with a list of the issues around it.

danielfaust, 2019-04-24:
> Moved over to 2.0.3pro, recreated the user (which should really have been automatically migrated during this change), imported a backup (needed two import attempts) and tried to sync historic measurements still in the scale.

That was also my case.

And I can't get BT to work with the SilverCrest SBF75, same issue as always.

But I noticed that sometimes the App says "Connected to...", but the BT-icon on the scale doesn't light up. As if the App only thinks it connected successfully, but it actually didn't.

danielfaust, 2019-04-24:
I am trying to communicate with this scale (SBF75) on Python, so I am starting to use `gatttool` to do a manual step-by-step readout of the scale.

For this I used an old debug version of this openScale (aka "the App" and added a lot of additional log output comments.

When using `gatttool` on a Raspberry Pi I've noted that I cannot connect to the scale if I haven't tapped on it first, in order to wake it up and accept connections, which is a bit odd because the App is able to connect to the scale even when it has been disconnected for some time and I haven't tapped it.

So what I've started doing is the following:
Let the scale turn itself off (the display). Completely terminate the App via the App dialog, and turn off Bluetooth. Start the App, tap on the scale so that the display shows, then press the connect button in the App, and confirm that it should enable BT. Then there is a good chance to get the App to communicate correctly.

Could someone try to confirm this?

With the Raspberry Pi, when I do a `gatttool -I -b xx;xx...` and then tap the scale and issue a `connect`, it connects immediately and I can issue commands at will, and they work. The scale won't even disconnect if the display then turns off, so the connection is very stable.

There appears to be something broken inside of RxBle, or in the flow-logic of the App concerning connections and disconnections.

engeeaitch, 2019-04-24:
@danielfaust I've just tried your method of connecting and it fails for me every time at the moment - I must have tried 10 - 15 times.  It does occasionally work for me, but not tonight!  I'm sure that an earlier version of the App worked much more reliably.

lenucksi, 2019-04-24:
I can confirm the need to tap the scale to make it turn on to be able to fetch any metrics.

danielfaust, 2019-04-25:
I just noticed that when the state machine is running (hello, set time, get scale data, get user list, get measurements, ...) and an active measurement packet comes in, that the scale then aborts the current command.

For example, if you're in the middle of `get user list` or `get scale data` and you put pressure on the scale, the scale will stop sending the remaining notifications for the current command and issue a weight notification.

This could probably leave the app in an state where it is waiting for the remaining notifications. Although maybe the app handles this properly and re-issues the command, I haven't looked at the code for this.

In any case:
 (Optionally: Force close the app, disable BT)
Tap the scale, don't stand on it.
Press the BT-Connect button in the app
Wait a couple of seconds after the connected message appears (probably better to check for the symbol on the scale), then stand on the scale.

Maybe it would be helpful if the app shows a message when it issued the "CMD_DO_MEASUREMENT"

IIRC, I had this problem appear for the first time when I made a ton of measurements and never connected it to the app. Then it wouldn't work, so I installed the app that came with the scale. There I downloaded all the measurements (quite a few), and this did take a considerable amount of time. Afterwards I had almost no issues with openScale.

@engeeaitch Maybe you have a lot of measurements in the scale's memory, and you think it's ok to start measuring your weight and stand on the scale, and this aborts the download of the measurement, but leaving the app in a state where it is still expecting some measurements (the app will be told how many there are, so it knows that and waits for it). What if you just do all these steps, but never stand on the scale, and leave it for a minute or two (it can remain connected when the display is off, so the display is no indication of an existing connection, only the BT-symbol). And possibly repeat these steps 2 or 3 times?

danielfaust, 2019-04-25:
Assuming the app behaves like the readout I can perform via a Python script, with 1 user on the scale and the storage buffer having 30 entries, which is the maximum amount of storage space it has, fetching all those 30 entries takes 12 seconds, and the total time passed between connecting to the scale and starting to wait for the active measurements (the toast indicating that measurement can be startet (get on the scale)) is 13 seconds.

This script does not print, does not store, it only fetches the data into a Python dictionary and prints it out at the end.

Assuming each new measurement takes 200 ms to get stored in the app, that would be 6 seconds added, for a total of 19, let's say 20 seconds.

This means that if you connect to the scale and it has a lot of measurements stored on it, if you step on it to measure your weight and you haven't waited at least 20 seconds, that may cause problems. So it may be really important to wait for the Toast which indicates the start of a measurement, or to check if the name appears on the lower left of the scale, that's when the app has indicated the scale that it should wait for that person to step on the scale to take a measurement.

In any case, I noticed that it is **significantly** more performant  (7 seconds vs 18 seconds with no measurements stored on the device) to not use the app for live measuring, but just to step on the scale, and then at some later time (about 5 minutes later or hours or days later, remember, it can store up to 30 measurements) tap on it and connect the app, and leave the app alone without stepping on the scale, possibly until it times out because of inactivity (60 seconds I think). This would also probably help with battery life.

engeeaitch, 2019-04-25:
@danielfaust Assuming that the scale deletes the measurement after it has been transferred to the App, then I don't have any measurements on the scale (but I do have other users on the scale that might have measurements - do you think that is relevant?). 
It's really hard to establish a pattern - sometimes it connects, but most of the time it does not - but I can't isolate what causes it to fail.  On occasions, I have been able to download a measurement that I took earlier, and I have also been able to connect to the scale for a live measurement.
Keep up the good work.

danielfaust, 2019-04-25:
@engeeaitch After I posted that comment earlier, I tried to connect and do a sync, and it wouldn't want to work at all.

The app won't download other users measurements. The app deletes only the measurements for one user (all which are associated to the user) by issuing a delete user's measurements (by user-id) after it has downloaded all the measurements for that user.

You could try to take up to 31 measurements on the scale without using the app at all, at some point you will get the message "FULL" on the scale. By counting them you might be able to figure out if there is some other user which has a lot of measurements on the scale. But since the app only downloads the ones of the current user, that shouldn't really matter.

I'm clueless. I've now nearly finished the Python script then I will export all data from the app and insert it into a database, and from then on the script will insert the measurements into the database, so I won't be needing the app anymore.

I really hope that any of the information I've provided was useful.

mapotn, 2019-04-26:
Hi Guys, have same issue with Sanitas SBF70 and connecting to app v2.0.3-pro (44). I had same issues in previous versions as well, so decided to buy PRO version to prevent these issues. Unfortunately issue with connecting comes back in various versions. Anyway found out that version 1.8.2 is still working well (but restore back up with latest version is not supported). Can developers take a look on it why latest versions are not that stable?

Thank you

doctorjames, 2019-04-26:
@danielfaust would you be willing to share the Python script when you are happy with it?

danielfaust, 2019-04-26:
@doctorjames sure. I'll let you know when it's done. Probably tomorrow or on sunday.

pplucky, 2019-04-26:
> @engeeaitch After I posted that comment earlier, I tried to connect and do a sync, and it wouldn't want to work at all.
>
> The app won't download other users measurements. The app deletes only the [...]

Could you by chance share your python script to communicated with Sanitas SBF70?

I haven't been able to find any valid script to actually do it.

> @doctorjames sure. I'll let you know when it's done. Probably tomorrow or on sunday.

Looking forward to it also...

danielfaust, 2019-04-27:
I'll probably commit the code tomorrow, after reviewing it. But it is finished and will be accessible here
https://github.com/danielfaust/bt-scale
The readme is already there, so you can see what to expect.

oliexdev, 2019-05-25:
please try out the blesses-android branch version at https://github.com/oliexdev/openScale/releases/tag/travis-dev_blessed-android-build
I replaced the Bluetooth backend with the [blesses-android](https://github.com/weliem/blessed-android) library, maybe this resolves the problem!?

TheAbsoluteTruth, 2019-05-26:
I can confirm this is working, thank you olie! 
please let me know if you need additional information to verify this.

oliexdev, 2019-05-26:
@TheAbsoluteTruth thanks, could you please test this version in every condition with your scale? Add a new user. Weight for different user different weights and then sync it with openScale and so on. Do you have the SBF70 or 75? Could you attach a openScale log file (enable it under settings->about)

danielfaust, 2019-05-26:
Didn't work for me

```
2019-05-26 10:38:25.534 Debug [2] AboutPreferences: Debug log enabled, openScale (dev_blessed-android) v2.0.3-dev_753adb6f_2019-05-25 (1558815134), SDK 28, Google Pixel 3
2019-05-26 10:38:25.542 Debug [2] AboutPreferences: Selected user id(1) ===========================
2019-05-26 10:38:35.982 Debug [2] OpenScale: Trying to connect to bluetooth device [50:33:8B:EC:61:6E] (SBF75)
2019-05-26 10:38:35.989 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-05-26 10:38:36.229 Debug [2] BluetoothCommunication: Found peripheral 'SBF75'
2019-05-26 10:38:37.235 Debug [2] BluetoothCommunication: Try to connect to BLE device 50:33:8B:EC:61:6E
2019-05-26 10:38:42.899 Debug [2] BluetoothCommunication: connected to 'SBF75'
2019-05-26 10:38:42.901 Debug [2] BluetoothCommunication: Step Nr 0
2019-05-26 10:38:42.902 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-05-26 10:38:42.904 Debug [2] BluetoothCommunication: Step Nr 1
2019-05-26 10:38:42.906 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2019-05-26 10:38:42.908 Debug [2] BluetoothCommunication: Stop machine state
2019-05-26 10:38:42.909 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-05-26 10:38:42.922 Debug [2] MainActivity: Bluetooth connection successful established
2019-05-26 10:38:42.962 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2019-05-26 10:38:47.976 Debug [2] BluetoothCommunication: disconnected 'SBF75' with status 8
2019-05-26 10:39:07.012 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-05-26 10:39:07.014 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-05-26 10:39:07.035 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-05-26 10:39:09.226 Debug [2] OpenScale: Trying to connect to bluetooth device [50:33:8B:EC:61:6E] (SBF75)
2019-05-26 10:39:09.228 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-05-26 10:39:09.254 Debug [2] BluetoothCommunication: Found peripheral 'SBF75'
2019-05-26 10:39:10.258 Debug [2] BluetoothCommunication: Try to connect to BLE device 50:33:8B:EC:61:6E
2019-05-26 10:39:12.095 Debug [2] BluetoothCommunication: connected to 'SBF75'
2019-05-26 10:39:12.096 Debug [2] BluetoothCommunication: Step Nr 0
2019-05-26 10:39:12.098 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-05-26 10:39:12.100 Debug [2] BluetoothCommunication: Step Nr 1
2019-05-26 10:39:12.102 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2019-05-26 10:39:12.104 Debug [2] BluetoothCommunication: Stop machine state
2019-05-26 10:39:12.105 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-05-26 10:39:12.122 Debug [2] MainActivity: Bluetooth connection successful established
2019-05-26 10:39:12.161 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2019-05-26 10:39:17.168 Debug [2] BluetoothCommunication: disconnected 'SBF75' with status 8
2019-05-26 10:39:27.914 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-05-26 10:39:27.915 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-05-26 10:39:27.924 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-05-26 10:39:30.996 Debug [2] OpenScale: Trying to connect to bluetooth device [50:33:8B:EC:61:6E] (SBF75)
2019-05-26 10:39:30.999 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-05-26 10:39:31.245 Debug [2] BluetoothCommunication: Found peripheral 'SBF75'
2019-05-26 10:39:32.250 Debug [2] BluetoothCommunication: Try to connect to BLE device 50:33:8B:EC:61:6E
```

Then it crashed and a dialog popped up, here's what I copied to the clipboard:

```
Build version: 2.0.3-dev_753adb6f_2019-05-25 
Current date: 2019-05-26 10:39:42 
Device: Google Pixel 3 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.TypedArray android.content.Context.obtainStyledAttributes(android.util.AttributeSet, int[], int, int)' on a null object reference
	at android.preference.Preference.<init>(Preference.java:239)
	at android.preference.Preference.<init>(Preference.java:339)
	at android.preference.Preference.<init>(Preference.java:357)
	at android.preference.Preference.<init>(Preference.java:366)
	at com.health.openscale.gui.preferences.BluetoothPreferences.onDeviceFound(BluetoothPreferences.java:172)
	at com.health.openscale.gui.preferences.BluetoothPreferences.access$000(BluetoothPreferences.java:55)
	at com.health.openscale.gui.preferences.BluetoothPreferences$1.onDiscoveredPeripheral(BluetoothPreferences.java:88)
	at com.welie.blessed.BluetoothCentral$2$1.run(BluetoothCentral.java:129)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6718)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

Have you considered using Crashlytics in these test builds?

oliexdev, 2019-05-26:
@danielfaust considering your crash log it looks it had happend during the Bluetooth search in the settings!?

Could you also test more, to see if it works in every condition? Especially with multiply users?

closed 2019-06-02 by oliexdev

danielfaust, 2019-06-02:
Hi, my apologies for not getting back to you. I have not tested multiple users, as I'm no longer using the app, but when I do so, I will let you know how it went.