oliexdev/openScale #455: Blood Pressure measurement

revulent, 2019-04-04:
Given the different physiological measurements the app can track, it would make sense to include blood pressure. I think a blood pressure tracker would fit in with the app well and it would be nice to be able to track my blood pressure with the same app I track my weight.

oliexdev, 2019-04-22:
do you mean by adding it manually or to support Bluetooth blood pressure?
Then we have two new measurements (Systolic (mmHg) and Diastolic (mmHg)) maybe three if we include the heart rate!?

gappleto97, 2019-04-27:
Entering it manually would be a good start. I am not aware that there even *are* Bluetooth-enabled blood pressure cuffs.

A compromise might be to allow users to create custom measurements. This way you do not have to support as much when it comes to calculation or specialized functions, but you can still allow for some special use cases

oliexdev, 2019-04-29:
yeah some blood pressure devices supports Bluetooth e.g. Omrom devices.

creating custom measurements is not easy to implement. You could use the comment field to add some information for special use cases.

gappleto97, 2019-04-29:
That is not very satisfying though. The comment field cannot be graphed, and if you want to do any analysis on it it would require parsing.

That is what I am doing in the meantime, but it would be very nice to support it as a measurement. Given the number of optional measurements already, I dont think there is any crisis in adding a few more, especially since I cannot think of any other reasonable measurements this app is missing.

Killom, 2019-07-10:
I would also like to request you, to add such a feature.

The Omron (i.e. M700) devices are using BLE profiles.

FTL1, 2020-03-12:
> do you mean by adding it manually or to support Bluetooth blood pressure?
> Then we have two new measurements (Systolic (mmHg) and Diastolic (mmHg)) maybe three if we include the heart rate!?

Yes, please!

Systolic (mmHg)
Diastolic (mmHg)
Heart Rate
+ Body Temperature

These measures, dumped conveniently into the OpenScale Pro db....with corresponding time and weight..   VERY useful!!  All the other apps seem to suck...and none of them are interested in playing ball on the same court :-)