oliexdev/openScale #456: SBF70 Bluetooth

jroth-haj, 2019-04-10:
Updated to 2.0.2 pro.

Got the same BT error I described here 
https://github.com/oliexdev/openScale/issues/433#issuecomment-473467965

MainActivity: Bluetooth unexpected error: Disconnected from D0:B5:C2:D2:9B:2A with status 8 (GATT_INSUF_AUTHORIZATION)
2019-03-15 23:53:07.974 Error [2] MainActivity: Bluetooth unexpected error: Cannot set local notification (code 1) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb

jroth-haj, 2019-04-13:
I refund the payment of openscale pro because it's unusable. So long I stay at Version 1.8.2.
It's free and it's working with the SBF70.

closed 2019-04-22 by oliexdev

jroth-haj, 2019-04-22:
Duplicate?
Can't find the duplicate issue in open issues.
And it's still not solved.

oliexdev, 2019-04-22:
See issue #454 or read properly the readme or search for sbf 70 in the issues... there are quite a lot of the same issues

jroth-haj, 2019-04-22:
They all have different error messages.
What's changed since Version 1.8.2?
With this Version it's running.