oliexdev/openScale #457: please provide a way to reassign measurements to a different user

TjrGithub, 2019-04-11:
**Is your feature request related to a problem? Please describe.**
I have 5 family members with an openscale user each, and pages full of measurements. But most of the measurements are assigned to the default user with no obvious way to change that. Two users are very close in weight, so I sometimes need to reassign new measurements.

It happened this way: For two years, my wife has been using the scale. I got a new phone, rediscovered openscale, created users for both of us, and connected to the scale. Wohoo, all the data is still there on the Xiaomi v1 scale. Most the data got dumped into the primary user. On top of that, the children have grown drastically in that time, now are reasonably similar in weight, and completely confuse the auto-assignment feature.

**Describe the solution you'd like**
Long-pressing on a measurement currently brings up the edit menu. I'd like for this menu to have an entry, reassign user.

**Describe alternatives you've considered**
A workaround that doesn't work is: Uninstall the app, reinstall it, create the 5 users with suitable starting weights, and re-download the data. This does not work because the kids' weight trajectories overlap. E.g. they started at 5kg and 8kg, now are at 11kg and 12kg, and you need the timestamp to decide which kid was 10kg. 

**Additional context**
This is somewhat related to #43, but this bug concerns already-imported measurements instead of initialization.

oliexdev, 2019-04-22:
thanks I will put it on the to-do list. If anybody want to implement this feature are welcome to send a PR.

oliexdev, 2019-05-26:
I added a user measurement view to reassign an user, so please test the latest dev version of openScale.

closed 2019-06-02 by oliexdev

TjrGithub, 2019-06-05:
I'm glad to help test it.

Is there an apk file available? (I see the source code at https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

Can I install the test APK over the FDroid apk without losing data?