oliexdev/openScale #458: Openscale sync not able to sync

NDR008, 2019-04-25:
Hello,

I had been using openscale and a couple of months ago I bought openscale pro.
Today I bought openscale sync.
But I cannot sync to Google health

Build version: 0.2 
Current date: 2019-04-25 08:28:01 
Device: Samsung SM-G950F 
 
Stack trace:  
java.lang.IllegalArgumentException: Data point does not have the timestamp set: DataPoint{[83.65]@[0, 0,raw=0,insert=0](r:weight:com.health.openscale.sync: N/A)}
	at d.a.a.a.b.b.s.a(:9)
	at com.google.android.gms.fitness.data.DataSet.a(:16)
	at d.b.a.a.a.b.k.a(:24)
	at d.b.a.a.b.h.onClick(:23)
	at android.view.View.performClick(View.java:7352)
	at android.widget.TextView.performClick(TextView.java:14177)
	at android.view.View.performClickInternal(View.java:7318)
	at android.view.View.access$3200(View.java:846)
	at android.view.View$PerformClick.run(View.java:27801)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7045)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

Thanks.
Nadir

NDR008, 2019-04-25:
I found the source of this issue.
I had an entry in my data - dated from 1970 (I do not know how it was created, because I started using openscale for measurements from 2016).

This 1970 had several blanks, which cause the sync error - the hint was:
`java.lang.IllegalArgumentException: Data point does not have the timestamp set: DataPoint{[83.65]@[0, 0,raw=0,insert=0](r:weight:com.health.openscale.sync: N/A)}`

Perhaps an improvement would be to check for invalid entries first (I think the issue was that there was a null entry)?

oliexdev, 2019-04-25:
thanks but it is a bit strange because you can't have a measurement without a timestamp!? Which fields had the blanks?

NDR008, 2019-04-25:
Unfortunately I do not know.
I suspect the problem occurred from an import.

I won't be able reproduce but the one was obviously generated abnormally
since I was not both in 1970...

oliexdev, 2019-04-26:
we will see if this problem occurs more often

closed 2019-04-26 by oliexdev