oliexdev/openScale #459: Xiaomi Mi Scale v2 bluetooth not staying connected

potatoflakes, 2019-04-27:
**Describe the bug**
Not sure if this is important, but the Aliexpress listing called this the "global version" of the Mi Scale 2. When I try to connect Bluetooth to the scale I get "connecting to: MIBCS" followed by "connection established" immediately followed by "bluetooth connection closed". The openscale app must have successfully communicated something because it switches the weight measurement on the scale between pounds and kilograms depending on the app settings. It doesn't take any data from the scale though.

I tried three different devices, each running a different version of lineageos (14.1, 15.1, 16.0). All of them had the same connection issue with the scale disconnecting from this app. I tried the official Mi Fit app on the device running 15.1 to verify the scale is functional and had no issues downloading the data from it.

**To Reproduce**
Steps to reproduce the behavior:
the scale shows up under "your bluetooth scale" as MIBCS
It connects successfully, then immediately disconnects 
tried this with the latest version of the app (2.0.3)

**Debug log**
The app never crashed during this process. As far as I can tell it never provided an error of any kind either.

oliexdev, 2019-04-29:
that's ok, internally a mi scale sets a id. The mi fit app sets a unique id to always identify you, openScale uses a random id to protect your privacy. You have to use more often the app to the scale. The first time the random id is set. Weight yourself and then after it use openScale to fetch the measurements.
You can enable the openScale log file (under settings->about) while your weighting to see what happened in openScale.

closed 2019-04-29 by oliexdev

potatoflakes, 2019-04-30:
I finally got it to write something to the debug log, but I have no idea what to make of it. No matter what I do I get the same connect and disconnect message. Is the app supposed to connect automatically or is there something else I need to do?
[openScale_2019-04-30_11-50.txt](https://github.com/oliexdev/openScale/files/3132256/openScale_2019-04-30_11-50.txt)

potatoflakes, 2019-05-12:
Was my comment missed?

oliexdev, 2019-05-12:
there is no live support. Weight yourself -> connect with openScale afterwards to sync the measurements that's all but you don't should switch between Mi Fit and openSale because of the description above. You could also read the wiki which is described in detail.

potatoflakes, 2019-05-12:
I haven't been able to find any instructions on the wiki for using the mi scale 2. Can you please provide a link?

I followed what you said in your comment from 13 days ago with no luck.

Here's the most recent thing I've tried:
1)turn off Bluetooth using the button within the app
2)weigh myself
3)wait about 5 seconds after everything finishes blinking
4)Turn Bluetooth back on
5)try to fetch the measurements

I've tried every way of connecting that I can think of. I've attached another log based on what I described above.

I've tried just standing on the scale for a a few minutes after the lights stop blinking and restart the app a few times. I've also tried restarting the app after stepping off the scale and trying to see if it would get the measurements then.

No matter what I do I get the same connect and disconnect message with no measurement imports.

I don't have the mi fit installed on the phone I'm trying to connect the scale to.

[openScale_2019-05-12_14-37.txt](https://github.com/oliexdev/openScale/files/3170488/openScale_2019-05-12_14-37.txt)

potatoflakes, 2019-05-25:
Does the log show anything useful?

I tried to look through it and noticed this warning towards the end (this same warning shows up in other logs for the different ways I've tried to connect the scale):

2019-05-12 14:38:25.006 Warning [646] RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
java.lang.InterruptedException
	at java.lang.Object.wait(Native Method)
	at com.polidea.rxandroidble2.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
	at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:71)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:458)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)

potatoflakes, 2019-06-09:
Please let me know if the openscale log shows anything so that I know whether or not I need to buy a new scale? The latest update hasn't fixed connectivity issue.

oliexdev, 2019-06-10:
please attach the log file

potatoflakes, 2019-06-11:
log for the latest version, waited until the scale finished measurements before trying to fetch

[openScale_2019-06-10_22-14.txt](https://github.com/oliexdev/openScale/files/3274643/openScale_2019-06-10_22-14.txt)

oliexdev, 2019-06-16:
please try out with the latest dev version of openScale and attach the log file

potatoflakes, 2019-06-23:
Connects successfully, but there seems to be some issue with history retrieval and a couple of the measurements.

On try 1 it did not retrieve the measurement that had just been taken, but downloaded one from April 27, which was probably the first time I used the scale. On the second try it only retrieved the measurement that had just been taken.

The measurement from April and measurement from today both read 80.00% for the muscle measurements. Also, 49.77% body fat seems pretty high. Is there a document that explains these percentages? 

[openScale_2019-06-23_01-59.txt](https://github.com/oliexdev/openScale/files/3317560/openScale_2019-06-23_01-59.txt)