oliexdev/openScale #464: I want 4 scales to work together for wheel scales for racing Karts Kartscales is project

ponytl, 2019-05-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ponytl, 2019-05-04:
I'm a completely new... my last real programming was in DOS before windows... but I want to use 4 inexpensive bluetooth $20ebay scales  to  weigh my sons racing kart (and share this with other racers) currently a set of 4 scales is over $1000 which is outside the budget of many weekend racers, 
With one scale under each wheel/tire on a level surface: 
What we need to know is:
A. total weight from all 4 scales combined
B. front weight (front 2 tires) rear weight (rear 2 tires)
C. side weight left side, right side
E. Cross weight  left rear and right front
and most importantly is the percentages of each in relationship to the total weight
[image]

Thank you in advance for any help and direction

oliexdev, 2019-05-11:
You could read the wiki and search in the issues to get an idea how to reverse engineer a scale.

closed 2019-05-11 by oliexdev

ponytl, 2019-05-11:
thanks...  I just received 4 of the scales... they each have a different  bluetooth address?  i just need a way to gather info from each and assemble it in a useful manor