oliexdev/openScale #465: Can't connect to SBF75 after upgrade to pro

Boddlnagg, 2019-05-07:
I really don't want to give you a bad review on Google Play, but I hope you can help me here:

The non-pro version said that Bluetooth is no longer supported. Okay. So I bought the pro version. But when I want to connect with the pro version it doesn't work either, failing with the message: *"Max. number of concurrent scale users reached"*. But I have only one user registered on the scale, which is the one I used with openScale before, and I used the same user name when I created the user in the pro version.

Now the Wiki says this (https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#how-do-i-get-on-my-sbf70-scale-the-body-metrics-water-muscle-fat-percentage):
> Please make first a backup of your measurements before you continue! To resolve this problem please delete all users in the vendors app (or reset the scale with the button on the back) and then open openScale and add a new user. After you synchronise to the SBF70 scale with openScale you should receive all other body metrics.

... but I don't want to do a hard reset because then I'd probably lose measurements from the last three weeks that I hadn't synced yet.

Is it a problem that I have already imported the history from the old app into the pro version before I created a user?

engeeaitch, 2019-05-09:
Hi - I had a similar problem.  You have to setup the username exactly as it was before (caps, spaces etc.), otherwise it creates a new user.  I found that I had already had other users that I wasn't aware of (probably setup from the vendor app).  I'm guessing that you backed up your data by doing an export from the Table data?  This only saves the scale measurements. A better solution is to do a full export backup, because this saves the user details, as well as the actual readings - go to Settings, then Backup, Export Backup.  Then import this to the pro-version of the App.
BUT... the app only works very intermittently (for me and some others at least).  Be prepared for a number of repeated attempts before you can connect to the scale.  Hopefully this will get fixed at some point.

oliexdev, 2019-05-11:
you could also check into the openScale log file to check which users are on your scale (enable it under settings->about)

Boddlnagg, 2019-05-12:
> A better solution is to do a full export backup

Thanks, I hadn't realized that this exists! That solved it for me.

closed 2019-05-12 by Boddlnagg