oliexdev/openScale #466: Table view: Font size too small

jottr, 2019-05-12:
Text in table view and in graph annotations is unreadable because it is too small.

See relative font sizes in the screenshots below: 

[image]

[image]

App version: `v2.0.3(44)`

Android 9
Xiaomi Mi A2

Display: 5.99-inch / 2160px x 1080px resolution

Please add option to set the font size in the app or adhere to device wide font size settings.

oliexdev, 2019-05-12:
well, I can read that. If I increase the font size then the table columns are too big but you can send a PR with your proposal font size then we will see.

jottr, 2019-05-13:
This is a screenshot. You'd need to scale it down to the real screen size of the Mi A2 to see the real font size. 
Not every user of your app will have your great eyesight. For them, the table view is currently unreadable. Which renders the app difficult or even impossible to use for this part of the user base.
From my perspective this is a UX bug. 

I won't create a PR since I'm not an Android dev. But I'm happy to contribute and help improve the project by creating tickets.

jottr, 2019-05-13:
Why not simply make the table view scrollable? This would allow the font size to be increased.

oliexdev, 2019-05-19:
@jottr a horizontal scrollable is not easy to implement but I made the font size of the graph a bit bigger. Anyway if you can't read it you can still click on each measurement to get the measurement dialog.

closed 2019-05-19 by oliexdev

jottr, 2019-05-19:
Why did you close the issue? May I suggest reopening it and adding a `help-wanted` label? 
Maybe someone else is capable of coming up with a solution.

oliexdev, 2023-01-29:
I refactored the overview and table layout, the font-size should be larger. Please try out the latest dev-build.