oliexdev/openScale #470: Crash with Beurer BF700

CodeFinder2, 2019-06-06:
App (v2.1.0 from F-droid) crashes with the following stack trace while taking a measurement. I am unable to complete it and it does not get logged / stored:

Build version: 2.1.0 
Build date: 1979-11-30 01:00:00 
Current date: 2019-06-07 00:51:31 
Device: OnePlus ONEPLUS A3003 
 
Stack trace:  
java.lang.NullPointerException: Attempt to read from field 'boolean com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.isNew' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onNextStep(BluetoothBeurerSanitas.java:237)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:434)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.access$000(BluetoothCommunication.java:44)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicWrite(BluetoothCommunication.java:327)
	at com.welie.blessed.BluetoothPeripheral$1$9.run(BluetoothPeripheral.java:563)
	at android.os.Handler.handleCallback(Handler.java:789)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6798)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

Happens all the time. :-( Sometimes it also seems to take quite long until it says "connected" 🤔...

oliexdev, 2019-06-08:
thanks should be fixed in the latest dev build

closed 2019-06-08 by oliexdev

CodeFinder2, 2019-06-30:
Thanks! Would you mind making a new minor release with this fix included?