oliexdev/openScale #471: Cannot connect to scale (SBF70) without location permission in 2.1.0

Olangu, 2019-06-11:
**Describe the bug**
After update to 2.1.0 it's no longer possible to connect to Bluetooth scale Sanitas SBF70 without location permission. 

**To Reproduce**
Revoke lcation permission from app,
try to connect.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Fdroid 2.1.0

**Expected behavior**
For the app to connect even without location permission.

**Debug log**
See next in thread.

Olangu, 2019-06-11:
[openScale_2019-06-11_10-20.txt](https://github.com/oliexdev/openScale/files/3275684/openScale_2019-06-11_10-20.txt)

weliem, 2019-06-12:
Since you revoked the location permission, Openscale is trying a direct connection using the mac address of the scale. But that fails because that mac address is not in Android's bluetooth cache. So you must first scan to get it in the cache again and then it will work again...

Olangu, 2019-06-13:
You're correct. After connecting with location permission and revoking it again it does work.

Why can't the app connect to any MAC without it being in the cache? Is it due to horrible design decisions from Google?

oliexdev, 2019-06-14:
read https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect#connect-without-a-scan for further information

closed 2019-06-14 by oliexdev