oliexdev/openScale #472: No longer getting data from Yunmai Mini

gappleto97, 2019-06-16:
Since the latest app update I have no longer been able to connect. The app thinks that it is connected, but the scale only thinks so for about a quarter second, then the Bluetooth status icon disappears on it. The phone still thinks it is connected after that.

As a short term workaround, until we can figure out what the bug is, would you mind putting out a copy of the last release with the version number bumped? Android no longer lets you downgrade even with ADB unless the package is marked as debugable. 

I am happy to provide any logfiles you would like.

oliexdev, 2019-06-16:
then post the openScale log file

gappleto97, 2019-06-18:
```
2019-06-18 08:00:42.758 Debug [2] AboutPreferences: Debug log enabled, openScale v2.1.0 (45), SDK 27, Google Pixel 2 2019-06-18 08:00:42.773 Debug [2] AboutPreferences: Selected user id(1) name(Gabe) birthday(Sun Jan 12 00:00:00 PST 1997) age(22) body height(179.07) scale unit(lb) gender(male) initial weight(96.62) goal weight(72.01) goal date(Tue Dec 31 00:00:00 PST 2019) measure unt(in) activity level(0) 2019-06-18 08:00:56.001 Debug [2] OpenScale: Disconnecting from bluetooth device 2019-06-18 08:00:56.002 Debug [2] BluetoothCommunication: Bluetooth disconnect 2019-06-18 08:00:56.008 Debug [2] MainActivity: Bluetooth connection successful disconnected 2019-06-18 08:00:57.630 Debug [2] OpenScale: Trying to connect to bluetooth device [04:A3:16:3D:28:EF] (YUNMAI-SIGNAL-M1WS) 2019-06-18 08:00:57.631 Debug [2] BluetoothCommunication: No location permission, connecting without LE scan 2019-06-18 08:00:58.632 Debug [2] BluetoothCommunication: Try to connect to BLE device 04:A3:16:3D:28:EF 2019-06-18 08:01:01.230 Debug [2] BluetoothCommunication: connected to 'YUNMAI-SIGNAL-M1WS' 2019-06-18 08:01:01.231 Debug [2] BluetoothCommunication: Step Nr 0 2019-06-18 08:01:01.232 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 11 AF B3 01 16 55 5A 00 00 02 03 17] on 0xffe9 2019-06-18 08:01:01.233 Debug [2] BluetoothCommunication: Step Nr 1 2019-06-18 08:01:01.234 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 08 FC AD 00 00 00 00 00 18] on 0xffe9 2019-06-18 08:01:01.235 Debug [2] BluetoothCommunication: Step Nr 2 2019-06-18 08:01:01.236 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4 2019-06-18 08:01:01.237 Debug [2] BluetoothCommunication: Stop machine state 2019-06-18 08:01:01.238 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered 2019-06-18 08:01:01.243 Debug [2] MainActivity: Bluetooth connection successful established 2019-06-18 08:01:01.277 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 11 AF B3 01 16 55 5A 00 00 02 03 17> to <0000ffe9-0000-1000-8000-00805f9b34fb> 2019-06-18 08:01:06.317 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-SIGNAL-M1WS' with status 8 2019-06-18 08:01:17.455 Debug [2] OpenScale: Disconnecting from bluetooth device 2019-06-18 08:01:17.456 Debug [2] BluetoothCommunication: Bluetooth disconnect 2019-06-18 08:01:17.459 Debug [2] MainActivity: Bluetooth connection successful disconnected 2019-06-18 08:01:19.621 Debug [2] OpenScale: Trying to connect to bluetooth device [04:A3:16:3D:28:EF] (YUNMAI-SIGNAL-M1WS) 2019-06-18 08:01:19.622 Debug [2] BluetoothCommunication: No location permission, connecting without LE scan 2019-06-18 08:01:20.624 Debug [2] BluetoothCommunication: Try to connect to BLE device 04:A3:16:3D:28:EF 2019-06-18 08:01:25.470 Debug [2] BluetoothCommunication: connected to 'YUNMAI-SIGNAL-M1WS' 2019-06-18 08:01:25.471 Debug [2] BluetoothCommunication: Step Nr 0 2019-06-18 08:01:25.473 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 11 AF B3 01 16 55 5A 00 00 02 03 17] on 0xffe9 2019-06-18 08:01:25.476 Debug [2] BluetoothCommunication: Step Nr 1 2019-06-18 08:01:25.478 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 08 FC C5 00 00 00 00 00 70] on 0xffe9 2019-06-18 08:01:25.478 Debug [2] BluetoothCommunication: Step Nr 2 2019-06-18 08:01:25.478 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4 2019-06-18 08:01:25.479 Debug [2] BluetoothCommunication: Stop machine state 2019-06-18 08:01:25.479 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered 2019-06-18 08:01:25.485 Debug [2] MainActivity: Bluetooth connection successful established 2019-06-18 08:01:25.511 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 11 AF B3 01 16 55 5A 00 00 02 03 17> to <0000ffe9-0000-1000-8000-00805f9b34fb> 2019-06-18 08:01:30.548 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-SIGNAL-M1WS' with status 8 2019-06-18 08:01:32.588 Debug [2] OpenScale: Disconnecting from bluetooth device 2019-06-18 08:01:32.588 Debug [2] BluetoothCommunication: Bluetooth disconnect 2019-06-18 08:01:32.592 Debug [2] MainActivity: Bluetooth connection successful disconnected 2019-06-18 08:02:25.642 Debug [2] OpenScale: Trying to connect to bluetooth device [04:A3:16:3D:28:EF] (YUNMAI-SIGNAL-M1WS) 2019-06-18 08:02:25.644 Debug [2] BluetoothCommunication: Do LE scan before connecting to device 2019-06-18 08:02:26.685 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-SIGNAL-M1WS' 2019-06-18 08:02:27.689 Debug [2] BluetoothCommunication: Try to connect to BLE device 04:A3:16:3D:28:EF 2019-06-18 08:02:29.559 Debug [2] BluetoothCommunication: connected to 'YUNMAI-SIGNAL-M1WS' 2019-06-18 08:02:29.559 Debug [2] BluetoothCommunication: Step Nr 0 2019-06-18 08:02:29.560 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 11 AF B3 01 16 55 5A 00 00 02 03 17] on 0xffe9 2019-06-18 08:02:29.561 Debug [2] BluetoothCommunication: Step Nr 1 2019-06-18 08:02:29.561 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 08 FD 05 00 00 00 00 00 B1] on 0xffe9 2019-06-18 08:02:29.562 Debug [2] BluetoothCommunication: Step Nr 2 2019-06-18 08:02:29.562 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4 2019-06-18 08:02:29.563 Debug [2] BluetoothCommunication: Stop machine state 2019-06-18 08:02:29.563 Debug [2] BluetoothCommunication: Successful Bluetooth se [... cut ...] 3 17> to <0000ffe9-0000-1000-8000-00805f9b34fb> 2019-06-18 08:02:34.586 Debug [2] OpenScale: Disconnecting from bluetooth device 2019-06-18 08:02:34.587 Debug [2] BluetoothCommunication: Bluetooth disconnect 2019-06-18 08:02:34.595 Debug [2] MainActivity: Bluetooth connection successful disconnected 2019-06-18 08:02:34.609 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-SIGNAL-M1WS' with status 0
```

_Sent from my Pixel 2 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

oliexdev, 2019-06-20:
could you properly paste the log file into GitHub or attach it as a file?

gappleto97, 2019-06-20:
Sorry about that. It looked fine from mobile but I can see that its not on desktop.

[openScale_2019-06-18_08-00.txt](https://github.com/oliexdev/openScale/files/3310744/openScale_2019-06-18_08-00.txt)

oliexdev, 2019-06-23:
it seems the problem is the following line

> 2019-06-18 08:01:30.548 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-SIGNAL-M1WS' with status 8

could you try again while you giving the location permission (turn on GPS)

gappleto97, 2019-06-23:
The app had location permission. The last attempt in the logfile had it with location services turned on. I tried it again to be sure, and I see the same symptoms

oliexdev, 2019-06-25:
please look into the log file if the following is written

> 2019-06-18 08:00:57.631 Debug [2] BluetoothCommunication: No location permission, connecting without LE scan

then the location permission was not granted or GPS was not on.

gappleto97, 2019-06-26:
[openScale_2019-06-25_20-12.txt](https://github.com/oliexdev/openScale/files/3328068/openScale_2019-06-25_20-12.txt)

Location on:

[image]

Permission on:

[image]

weliem, 2019-06-28:
The scale is disconnecting after the first write operation. Difficult to say why...

Did you try pulling the batteries from the scale or renewing the batteries? Bluetooth devices often start behaving weird when batteries are almost empty. Make sure you also reboot your phone.

gappleto97, 2019-06-28:
I did reboot the phone. I will try replacing the batteries when I get home

gappleto97, 2019-06-29:
Looks like new batteries fixed it. Sorry for the trouble, I just didnt think it would fail the Bluetooth part only because of low battery. Thank you for the help.

gappleto97, 2019-06-29:
It now looks like that success was one time only. I recharged the batteries and it failed after the first time. I recharged again and it continued to fail ever time. Logs attached below

[openScale_2019-06-29_15-14.txt](https://github.com/oliexdev/openScale/files/3342345/openScale_2019-06-29_15-14.txt)

oliexdev, 2019-07-01:
try out this dev build https://github.com/oliexdev/openScale/releases/tag/travis-dev_yunmai_mini_fix-build

weliem, 2019-07-06:
Does you scale work properly with the original Yunmai scale?

gappleto97, 2019-07-07:
I dont know, as I have never used it. I dont intend to, either, considering how many unnecessary permissions they demand. From what I have read they seem to phone home a lot of information, and you can't use it without making an account. I will try the dev build tomorrow if I can.

gappleto97, 2019-07-07:
Bench build seems to work. Logs attached below.

[openScale_2019-07-07_14-54.txt](https://github.com/oliexdev/openScale/files/3366162/openScale_2019-07-07_14-54.txt)

gappleto97, 2019-07-08:
Also, I had never noticed before but it toggles the units on the scale to match the one set on the app, which is really cool. Major props to whoever thought of that.

closed 2019-07-14 by oliexdev

gappleto97, 2019-09-26:
Would you mind putting this in a release?

gappleto97, 2019-10-03:
Just saw it drop on F-Droid. Thank you!