oliexdev/openScale #474: Xiaomi Smart Weight Scale 2 not regocnized

FaBjE, 2019-06-19:
I bought a "Xiaomi Smart Weight Scale 2" planning it to use it with openScale.
shop link: https://nl.aliexpress.com/item/33007689938.html

It shows up in the bluetooth device list, but it says "device not supported" under it.
I find that strange because the scale is listed as supported in the wiki https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale

I included a debug log below with the device (characteristic) info.

Is this a new version of the scale that is not yet supported or?

**Debug log**

Debug [2] AboutPreferences: Debug log enabled, openScale v2.1.0 (45), SDK 28, OnePlus ONEPLUS A6003
Debug [2] BluetoothPreferences: Found unsupported device MI SCALE2 [C8:47:8C:9C:14:XX]
Debug [2] BluetoothCommunication: Do LE scan before connecting to device
Debug [2] BluetoothCommunication: Found peripheral 'MI SCALE2'
Debug [2] BluetoothCommunication: Try to connect to BLE device C8:47:8C:9C:14:XX
Debug [2] BluetoothCommunication: connected to 'MI SCALE2'
Debug [2] BluetoothCommunication: Step Nr 0
Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
Debug [2] BluetoothCommunication: Invoke read bytes on 0x2a00 "device name"
Debug [2] BluetoothDebug: Service 0x1800 "generic access"
Debug [2] BluetoothDebug: |- characteristic 0x2a00 "device name" (#3): READ, WRITE*
Debug [2] BluetoothDebug: |- characteristic 0x2a01 "appearance" (#5): READ, WRITE*
Debug [2] BluetoothDebug: |- characteristic 0x2a02 "peripheral privacy flag" (#7): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a04 "peripheral preferred connection parameters" (#9): READ
Debug [2] BluetoothDebug: Service 0x1801 "generic attribute"
Debug [2] BluetoothDebug: |- characteristic 0x2a05 "service changed" (#12): INDICATE, READ
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: Service 0x180a "device information"
Debug [2] BluetoothDebug: |- characteristic 0x2a25 "serial number string" (#16): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a27 "hardware revision string" (#18): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a28 "software revision string" (#20): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a23 "system id" (#22): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a50 "pnp id" (#24): READ
Debug [2] BluetoothDebug: Service 0x181d "weight scale"
Debug [2] BluetoothDebug: |- characteristic 0x2a2b "current time" (#27): READ, WRITE*
Debug [2] BluetoothDebug: |- characteristic 0x2a9e (#29): READ
Debug [2] BluetoothDebug: |- characteristic 0x2a9d "weight measurement" (#31): INDICATE
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: |- characteristic 00002a2f-0000-3512-2118-0009af100700 (#34): NOTIFY, WRITE*
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: Service 00001530-0000-3512-2118-0009af100700
Debug [2] BluetoothDebug: |- characteristic 00001531-0000-3512-2118-0009af100700 (#38): NOTIFY, WRITE*
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: |- characteristic 00001532-0000-3512-2118-0009af100700 (#41): WRITE_NO_RESPONSE*
Debug [2] BluetoothDebug: |- characteristic 0x2a04 "peripheral preferred connection parameters" (#43): NOTIFY, READ, WRITE*
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: |- characteristic 00001542-0000-3512-2118-0009af100700 (#46): NOTIFY, READ, WRITE*
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothDebug: |- characteristic 00001543-0000-3512-2118-0009af100700 (#49): NOTIFY, READ, WRITE*
Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
Debug [2] BluetoothCommunication: Bluetooth disconnect
Debug [2] OpenScale: Disconnecting from bluetooth device
Debug [2] BluetoothCommunication: Bluetooth disconnect
Debug [2] BluetoothCommunication: disconnected 'MI SCALE2' with status 0

FaBjE, 2019-06-19:
Just a little update. I just tested it with the Nordic nRF connect app and using the Service 0x181d "weight scale" the app as able to read the weight data correctly from the scale.

oliexdev, 2019-06-20:
try out the latest dev version of openScale

FaBjE, 2019-06-20:
I wanted to write "I tried that and it didnt work", but then I saw you just did a commit to (probably) fix this.
Thanks 👍 
I will test it tonight and report back

FaBjE, 2019-06-20:
I can select the scale now and it seems to connect. But no measurement is appearing.
I tried the + button, but that just manually adds an measurement.

Log:

019-06-20 20:25:26.054 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_de55107d_2019-06-20 (1561018557), SDK 28, OnePlus ONEPLUS A6003
2019-06-20 20:25:32.939 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-06-20 20:25:32.942 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-06-20 20:25:32.962 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-06-20 20:25:32.992 Debug [2] BluetoothCommunication: disconnected 'MI SCALE2' with status 0
2019-06-20 20:25:37.239 Debug [2] OpenScale: Trying to connect to bluetooth device [C8:47:8C:9C:14:XX] (MI SCALE2)
2019-06-20 20:25:37.244 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-06-20 20:25:37.548 Debug [2] BluetoothCommunication: Found peripheral 'MI SCALE2'
2019-06-20 20:25:38.559 Debug [2] BluetoothCommunication: Try to connect to BLE device C8:47:8C:9C:14:XX
2019-06-20 20:25:40.199 Debug [2] BluetoothCommunication: connected to 'MI SCALE2'
2019-06-20 20:25:40.201 Debug [2] BluetoothCommunication: Step Nr 0
2019-06-20 20:25:40.204 Debug [2] BluetoothCommunication: Invoke write bytes [06 04 00 00] on 00001542-0000-3512-2118-0009af100700
2019-06-20 20:25:40.206 Debug [2] BluetoothCommunication: Step Nr 1
2019-06-20 20:25:40.209 Debug [2] BluetoothCommunication: Invoke write bytes [E3 07 06 14 14 19 28 03 00 00] on 0x2a2b "current time"
2019-06-20 20:25:40.210 Debug [2] BluetoothCommunication: Step Nr 2
2019-06-20 20:25:40.212 Debug [2] BluetoothCommunication: Invoke set notification on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:25:40.213 Debug [2] BluetoothCommunication: Step Nr 3
2019-06-20 20:25:40.216 Debug [2] BluetoothCommunication: Invoke write bytes [01 FF FF 8F 2D] on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:25:40.218 Debug [2] BluetoothCommunication: Step Nr 4
2019-06-20 20:25:40.219 Debug [2] BluetoothCommunication: Invoke write bytes [02] on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:25:40.219 Debug [2] BluetoothCommunication: Stop machine state
2019-06-20 20:25:40.220 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-06-20 20:25:40.234 Debug [2] MainActivity: Bluetooth connection successful established
2019-06-20 20:25:40.297 Debug [2] BluetoothCommunication: SUCCESS: Writing <06 04 00 00> to <00001542-0000-3512-2118-0009af100700>
2019-06-20 20:26:XX.865 Debug [2] BluetoothCommunication: disconnected 'MI SCALE2' with status 19
2019-06-20 20:26:40.260 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2019-06-20 20:26:40.265 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-06-20 20:26:40.305 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-06-20 20:27:05.897 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-06-20 20:27:05.899 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-06-20 20:27:07.206 Debug [2] OpenScale: Trying to connect to bluetooth device [C8:47:8C:9C:14:XX] (MI SCALE2)
2019-06-20 20:27:07.211 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-06-20 20:27:07.895 Debug [2] BluetoothCommunication: Found peripheral 'MI SCALE2'
2019-06-20 20:27:08.906 Debug [2] BluetoothCommunication: Try to connect to BLE device C8:47:8C:9C:14:XX
2019-06-20 20:27:15.241 Debug [2] BluetoothCommunication: connected to 'MI SCALE2'
2019-06-20 20:27:15.243 Debug [2] BluetoothCommunication: Step Nr 0
2019-06-20 20:27:15.246 Debug [2] BluetoothCommunication: Invoke write bytes [06 04 00 00] on 00001542-0000-3512-2118-0009af100700
2019-06-20 20:27:15.247 Debug [2] BluetoothCommunication: Step Nr 1
2019-06-20 20:27:15.250 Debug [2] BluetoothCommunication: Invoke write bytes [E3 07 06 14 14 1B 0F 03 00 00] on 0x2a2b "current time"
2019-06-20 20:27:15.251 Debug [2] BluetoothCommunication: Step Nr 2
2019-06-20 20:27:15.252 Debug [2] BluetoothCommunication: Invoke set notification on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:27:15.253 Debug [2] BluetoothCommunication: Step Nr 3
2019-06-20 20:27:15.255 Debug [2] BluetoothCommunication: Invoke write bytes [01 FF FF 8F 2D] on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:27:15.256 Debug [2] BluetoothCommunication: Step Nr 4
2019-06-20 20:27:15.258 Debug [2] BluetoothCommunication: Invoke write bytes [02] on 00002a2f-0000-3512-2118-0009af100700
2019-06-20 20:27:15.259 Debug [2] BluetoothCommunication: Stop machine state
2019-06-20 20:27:15.259 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-06-20 20:27:15.268 Debug [2] MainActivity: Bluetooth connection successful established
2019-06-20 20:27:15.294 Debug [2] BluetoothCommunication: SUCCESS: Writing <06 04 00 00> to <00001542-0000-3512-2118-0009af100700>
2019-06-20 20:27:46.579 Debug [2] BluetoothCommunication: disconnected 'MI SCALE2' with status 19
2019-06-20 20:28:15.274 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2019-06-20 20:28:15.277 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-06-20 20:28:15.315 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-06-20 20:28:31.938 Debug [2] AboutPreferences: Debug log disabled

weliem, 2019-06-21:
@oliexdev , you added 'Mi Scale2' to the BluetoothMiScale2 class, but by the look of the service UUID's @FaBjE lists in the inital post, it is not a BluetoothMiScale2 but a BluetoothMiScale. So it you add it there it will probably work...

FaBjE, 2019-06-22:
It is indeed the "simple" version without the body impedance measurement.

oliexdev, 2019-06-23:
then please try again

FaBjE, 2019-06-23:
New dev version works a treat! thank you (both) so much for your effort!

closed 2019-06-25 by oliexdev

KerstinMaur, 2019-09-11:
The [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) works great, thank you so much!

[image]
[image]
[image]
[image]