oliexdev/openScale #477: Xiaomi Body Fat Scale v2

lesavonfou, 2019-07-03:
Hello everyone,
I have bought a Xiaomi Body Fat Scale 2 recently and it is not recognized by openScale.
aliexpress link: www.aliexpress.com/store/product/-/1872811_33008790826.html

The scale shows up as MIBFS, here is the log:
```
2019-04-30 11:50:24.552 Debug [2] AboutPreferences: Debug log enabled, openScale v2.0.3 (44), SDK 28, OnePlus ONEPLUS A3000
2019-04-30 11:50:24.553 Debug [2] AboutPreferences: Selected user id(2) name(test) birthday(Sat Jul 19 11:49:27 CDT 1997) age(21) body height(177.80) scale unit(lb) gender(male) initial weight(94.80) goal weight(72.57) goal date(Wed Oct 30 11:49:27 CDT 2019) measure unt(in) activity level(0)
2019-04-30 11:50:29.446 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-04-30 11:50:29.447 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-04-30 11:50:44.108 Debug [2] OpenScale: Trying to connect to bluetooth device [CE:8E:E5:5B:58:B6] (MIBCS)
2019-04-30 11:50:44.112 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-04-30 11:50:44.115 Debug [2] RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(54351201)
2019-04-30 11:50:44.116 Debug [2401] RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(54351201)
2019-04-30 11:50:44.119 Info [2407] RxBle#QueueOperation: Scan operation is requested to start.
2019-04-30 11:50:44.139 Debug [2401] RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(54351201) in 22 ms
2019-04-30 11:50:44.298 Debug [2] BluetoothCommunication: Stop Le san
2019-04-30 11:50:44.300 Info [2407] RxBle#CancellableDisposable: Scan operation is requested to stop.
2019-04-30 11:50:45.305 Debug [2] BluetoothCommunication: Try to connect to BLE device CE:8E:E5:5B:58:B6
2019-04-30 11:50:45.310 Debug [2] BluetoothCommunication: Step Nr 0
2019-04-30 11:50:45.315 Debug [2] BluetoothCommunication: Invoke write bytes [06 04 00 01] on 00001542-0000-3512-2118-0009af100700
2019-04-30 11:50:45.332 Debug [2] BluetoothCommunication: Step Nr 1
2019-04-30 11:50:45.334 Debug [2411] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(46091349)
2019-04-30 11:50:45.337 Debug [2] BluetoothCommunication: Invoke write bytes [E3 07 04 1E 0B 32 2D 03 00 00] on 0x2a2b "current time"
2019-04-30 11:50:45.339 Debug [2401] RxBle#ClientOperationQueue: STARTED  ConnectOperation(46091349)
2019-04-30 11:50:45.342 Debug [2] BluetoothCommunication: Step Nr 2
2019-04-30 11:50:45.346 Debug [2] BluetoothCommunication: Invoke set notification on 00002a2f-0000-3512-2118-0009af100700
2019-04-30 11:50:45.348 Verbose [2407] RxBle#BleConnectionCompat: Connecting without reflection
2019-04-30 11:50:45.608 Debug [2429] RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-30 11:50:45.616 Debug [2401] RxBle#ClientOperationQueue: FINISHED ConnectOperation(46091349) in 278 ms
2019-04-30 11:50:45.627 Debug [2] MainActivity: Bluetooth connection successful established
2019-04-30 11:50:45.637 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(200628433)
2019-04-30 11:50:45.642 Debug [2440] RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(200628433)
2019-04-30 11:50:46.555 Debug [2429] RxBle#BluetoothGatt$1: onServicesDiscovered status=0
2019-04-30 11:50:46.561 Verbose [2411] RxBle#ServiceDiscoveryOperation: Preparing services description
2019-04-30 11:50:46.580 Verbose [2411] RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
PERIPHERAL ADDRESS: CE:8E:E5:5B:58:B6
PERIPHERAL NAME: MIBCS
-------------------------------------------------------------------------

Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
Instance ID: 1
-> Characteristics:
	* Device Name (00002a00-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ WRITE ]
	* Appearance (00002a01-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ ]
	* Peripheral Privacy Flag (00002a02-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ WRITE ]
	* Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ ]
	* Reconnection Address (00002a03-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ WRITE_NO_RESPONSE WRITE ]

Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
Instance ID: 12
-> Characteristics:
	* Service Changed (00002a05-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ INDICATE ]
	  -> Descriptors: 
		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)

Primary Service - Device Information (0000180a-0000-1000-8000-00805f9b34fb)
Instance ID: 16
-> Characteristics:
	* Serial Number String (00002a25-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ ]
	* Software Revision String (00002a28-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ ]
	* Hardware Revision String (00002a27-0000-1000-8000-00805f9b34fb)
[... 220 lines cut ...]
[NEW] Descriptor
	/org/bluez/hci0/dev_0C_95_41_C7_55_35/service0022/char002b/desc002d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Characteristic
       /org/bluez/hci0/dev_0C_95_41_C7_55_35/service0022/char002e
	00001543-0000-3512-2118-0009af100700
	Vendor specific
[NEW] Descriptor
	/org/bluez/hci0/dev_0C_95_41_C7_55_35/service0022/char002e/desc0030
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device 0C:95:41:C7:55:35 UUIDs: 00001530-0000-3512-2118-0009af100700
[CHG] Device 0C:95:41:C7:55:35 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 0C:95:41:C7:55:35 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 0C:95:41:C7:55:35 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device 0C:95:41:C7:55:35 UUIDs: 0000181b-0000-1000-8000-00805f9b34fb
[CHG] Device 0C:95:41:C7:55:35 ServicesResolved: yes
[CHG] Device 0C:95:41:C7:55:35 Appearance: 0x0c80
[CHG] Device 0C:95:41:C7:55:35 Modalias: bluetooth:v0157p0019d0100
```

This scale looks at first sight like the smart scale v2 but it seems to have some differences:
- MIBFS as identifier
- Different GATT services

oliexdev, 2019-07-05:
does your scale support body measurements, see issue #474 ?

lesavonfou, 2019-07-06:
I am afraid not, the only service showing up is Body Composition (0x181B) where can be found 2 characteristics:
- Body Composition Feature (0x2A9B)
- Body Composition Measurement (0x2A9C)

Is there a way to sniff the Bluetooth communication between the MI fit and the scale? On Android or over the air?

lesavonfou, 2019-07-06:
Maybe it looks unrelated but I noticed that this service:
```
Primary Service - Unknown service (00001530-0000-3512-2118-0009af100700)
Instance ID: 38
-> Characteristics:
	* Unknown characteristic (00001531-0000-3512-2118-0009af100700)
	  Properties: [ WRITE NOTIFY ]
	  -> Descriptors: 
		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
	* Unknown characteristic (00001532-0000-3512-2118-0009af100700)
	  Properties: [ WRITE_NO_RESPONSE ]
	* Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
	  Properties: [ READ WRITE NOTIFY ]
	  -> Descriptors: 
		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
	* Unknown characteristic (00001542-0000-3512-2118-0009af100700)
	  Properties: [ READ WRITE NOTIFY ]
	  -> Descriptors: 
		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
	* Unknown characteristic (00001543-0000-3512-2118-0009af100700)
	  Properties: [ READ WRITE NOTIFY ]
	  -> Descriptors: 
		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
```

Is related the Xiaomi Mi Band according the Gadgetbridge code:
https://github.com/Freeyourgadget/Gadgetbridge/blob/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/devices/miband/MiBandService.java
```
    public static final UUID UUID_SERVICE_MIBAND_SERVICE = UUID.fromString(String.format(BASE_UUID, "FEE0"));
    public static final UUID UUID_SERVICE_MIBAND2_SERVICE = UUID.fromString(String.format(BASE_UUID, "FEE1"));
    public static final UUID UUID_SERVICE_HEART_RATE = GattService.UUID_SERVICE_HEART_RATE;
    public static final String UUID_SERVICE_WEIGHT_SERVICE = "00001530-0000-3512-2118-0009af100700";

    public static final UUID UUID_CHARACTERISTIC_DEVICE_INFO = UUID.fromString(String.format(BASE_UUID, "FF01"));

public static final UUID UUID_CHARACTERISTIC_DEVICE_NAME = UUID.fromString(String.format(BASE_UUID, "FF02"));
```

Seems not really used though

lesavonfou, 2019-07-06:
Never mind it looks like this service is used for firmware upgrades:
https://medium.com/@yogeshojha/i-hacked-miband-3-and-here-is-how-i-did-it-part-ii-reverse-engineering-to-upload-firmware-and-b28a05dfc308

The service id is quite similar

oliexdev, 2019-07-06:
please read the wiki how to reverse engineer a scale

weliem, 2019-07-06:
Looks like this is the new version of the Body Composition scale. Adding the new advertisement string to the handler might do the trick...

oliexdev, 2019-07-14:
according your logs it is not identifier as MIBFS!
Please upgrade to the latest openScale version and this issue should be resovled.

closed 2019-07-14 by oliexdev

lesavonfou, 2019-07-17:
The latest dev release (301153f) does not solve the recognition problem.
The following zip contains btsnoop logs: 
[btsnoop_logs.zip](https://github.com/oliexdev/openScale/files/3403055/btsnoop_logs.zip)

oliexdev, 2019-07-17:
please post the openScale debug log files with the latest version

Cyberrain, 2019-08-31:
2019-09-01 00:51:31.787 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_31a1712a_2019-08-31 (1567222012), SDK 28, Xiaomi Mi 9 SE
2019-09-01 00:51:31.788 Debug [2] AboutPreferences: Selected user id(1) name(Den) birthday(Sat Sep 01 00:00:00 GMT+03:00 1984) age(35) body height(174.00) scale unit(kg) gender(male) initial weight(76.00) goal weight(70.00) goal date(Sun Mar 01 00:34:41 GMT+03:00 2020) measure unt(cm) activity level(0)
2019-09-01 00:51:40.849 Debug [2] BluetoothPreferences: Found unsupported device MIBFS [0C:95:41:3E:EE:CC]
2019-09-01 00:51:41.164 Debug [2] BluetoothPreferences: Found supported device MI_SCALE [C8:0F:10:C0:A5:A3] (driver: Xiaomi Mi Scale v1)
2019-09-01 00:51:44.575 Debug [2] OpenScale: Trying to connect to bluetooth device [0C:95:41:3E:EE:CC] in debug mode
2019-09-01 00:51:44.580 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-09-01 00:51:44.598 Debug [2] BluetoothCommunication: Stop machine state
2019-09-01 00:51:44.945 Debug [2] BluetoothCommunication: Found peripheral 'MIBFS'
2019-09-01 00:51:45.962 Debug [2] BluetoothCommunication: Try to connect to BLE device 0C:95:41:3E:EE:CC
2019-09-01 00:51:47.845 Debug [2] BluetoothCommunication: connected to 'MIBFS'
2019-09-01 00:51:47.849 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-09-01 00:51:47.852 Debug [2] BluetoothCommunication: Invoke read bytes on 0x2a00 "device name"
2019-09-01 00:51:47.856 Debug [2] BluetoothDebug: Service 0x1800 "generic access"
2019-09-01 00:51:47.859 Debug [2] BluetoothDebug: |- characteristic 0x2a00 "device name" (#3): READ, WRITE*
2019-09-01 00:51:47.862 Debug [2] BluetoothDebug: |- characteristic 0x2a01 "appearance" (#5): READ
2019-09-01 00:51:47.864 Debug [2] BluetoothDebug: |- characteristic 0x2a04 "peripheral preferred connection parameters" (#7): READ
2019-09-01 00:51:47.866 Debug [2] BluetoothDebug: Service 0x1801 "generic attribute"
2019-09-01 00:51:47.869 Debug [2] BluetoothDebug: |- characteristic 0x2a05 "service changed" (#10): INDICATE, READ
2019-09-01 00:51:47.871 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.872 Debug [2] BluetoothDebug: Service 0x180a "device information"
2019-09-01 00:51:47.873 Debug [2] BluetoothDebug: |- characteristic 0x2a25 "serial number string" (#14): READ
2019-09-01 00:51:47.874 Debug [2] BluetoothDebug: |- characteristic 0x2a28 "software revision string" (#16): READ
2019-09-01 00:51:47.875 Debug [2] BluetoothDebug: |- characteristic 0x2a27 "hardware revision string" (#18): READ
2019-09-01 00:51:47.876 Debug [2] BluetoothDebug: |- characteristic 0x2a23 "system id" (#20): READ
2019-09-01 00:51:47.877 Debug [2] BluetoothDebug: |- characteristic 0x2a50 "pnp id" (#22): READ
2019-09-01 00:51:47.877 Debug [2] BluetoothDebug: Service 0x181b "body composition"
2019-09-01 00:51:47.879 Debug [2] BluetoothDebug: |- characteristic 0x2a2b "current time" (#25): READ, WRITE*
2019-09-01 00:51:47.880 Debug [2] BluetoothDebug: |- characteristic 0x2a9b (#27): READ
2019-09-01 00:51:47.880 Debug [2] BluetoothDebug: |- characteristic 0x2a9c "body composition measurement" (#29): INDICATE
2019-09-01 00:51:47.881 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.882 Debug [2] BluetoothDebug: |- characteristic 00002a2f-0000-3512-2118-0009af100700 (#32): NOTIFY, WRITE*
2019-09-01 00:51:47.883 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.884 Debug [2] BluetoothDebug: Service 00001530-0000-3512-2118-0009af100700
2019-09-01 00:51:47.885 Debug [2] BluetoothDebug: |- characteristic 00001531-0000-3512-2118-0009af100700 (#36): NOTIFY, WRITE*
2019-09-01 00:51:47.886 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.886 Debug [2] BluetoothDebug: |- characteristic 00001532-0000-3512-2118-0009af100700 (#39): WRITE*
2019-09-01 00:51:47.887 Debug [2] BluetoothDebug: |- characteristic 0x2a04 "peripheral preferred connection parameters" (#41): NOTIFY, READ, WRITE*
2019-09-01 00:51:47.888 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.889 Debug [2] BluetoothDebug: |- characteristic 00001542-0000-3512-2118-0009af100700 (#44): NOTIFY, READ, WRITE*
2019-09-01 00:51:47.890 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.891 Debug [2] BluetoothDebug: |- characteristic 00001543-0000-3512-2118-0009af100700 (#47): NOTIFY, READ, WRITE*
2019-09-01 00:51:47.891 Debug [2] BluetoothDebug: |--- descriptor 0x2902 "client characteristic configuration"
2019-09-01 00:51:47.892 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-09-01 00:51:47.894 Debug [2] BluetoothCommunication: Resume machine state
2019-09-01 00:51:47.895 Debug [2] BluetoothCommunication: Step Nr 0
2019-09-01 00:51:47.896 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2019-09-01 00:51:47.897 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-09-01 00:51:47.898 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-09-01 00:51:47.935 Debug [2] BluetoothCommunication: disconnected 'MIBFS' with status 0
2019-09-01 00:51:53.712 Debug [2] OpenScale: Trying to connect to bluetooth device [0C:95:41:3E:EE:CC] in debug mode
[... cut ...]
2019-09-01 00:51:55.881 Debug [2] BluetoothCommunication: Step Nr 0
2019-09-01 00:51:55.881 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2019-09-01 00:51:55.884 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-09-01 00:51:55.885 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-09-01 00:51:55.923 Debug [2] BluetoothCommunication: disconnected 'MIBFS' with status 0
2019-09-01 00:52:05.883 Debug [2] BluetoothPreferences: Found unsupported device MIBFS [0C:95:41:3E:EE:CC]

oliexdev, 2019-09-02:
@Cyberrain Do you have two version of the mi scale? mi scale v1 and v2?
Please try out the latest dev version of openScale.

Cyberrain, 2019-09-02:
My version is Mi body composition scale 2 v2. I reinstall latest dev builds, this is work, thanks.