oliexdev/openScale #479: Sync to Garmin account

ajfriesen, 2019-07-08:
Garmin does not charge a 5000$ fee for using their API anymore.

So synchronization of weight and possibly other data with Garmin would be awesome.

oliexdev, 2019-07-14:
maybe in the openScale sync version if I have time and it is not too complex

flatherskevin, 2019-09-07:
In my experience working with the Garmin API, it is a one way street. You can pull data from them, but not allowed to push data up. I could be wrong on this, but their Health API documentation doesn't seem to make any mention of updating data. All that said, it's OAuth1.0a for authentication, and they offer webhooks if there is a backend powering the application. If not, there is a standard RESTful API that they offer.

FifiTheReal, 2019-09-23:
[Weight Logger](https://play.google.com/store/apps/details?id=org.kochka.android.weightlogger) syncronise fine with Garmin, but I can't import data from openScale :-(

[WeightLogger is on Github.](https://github.com/kochka/WeightLogger)

oliexdev, 2019-09-27:
I've read my way into the Garmin Health API now and it looks a bit fishy. Even they stated it is free but in their [request form](https://www.garmin.com/en-US/forms/health-api-request/) the last question is:

> If your intended program is not revenue-bearing, please explain how the proposed relationship would offer strategic value to Garmin.

You have to be an "enterprise partner". The decision if you get access to the API is based on Garmin. Honestly, it is too much information what they want from me and the [basic idea](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale#how-its-started-) of openScale was to get rid of the company clouds for privacy reason.

So, no I am not intend to support synchronization to Garmin.

@flatherskevin regarding to this [code snippet](https://github.com/kochka/WeightLogger/blob/master/app/src/main/java/org/kochka/android/weightlogger/tools/GarminConnect.java) you can push data via a .fit file to Garmin.

closed 2019-09-27 by oliexdev