oliexdev/openScale #481: Bluetooth Sync seems broken after update to version 2.1.0 (Xiaomi Body Fat Scale v1)

sanjaay, 2019-07-09:
**Describe the bug**
After getting my weight with my scale I can not retrieve the data with Open Scale.

**To Reproduce**
Steps to reproduce the behavior:
1. Weigh yourself
2. Start Open Scale and start sync via Bluetooth
3. Open Scale connects properly and syncs for 20-30 seconds
4. No new data was retrieved

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.1.0

**Expected behavior**
Open Scale should show the new data.

**Additional context**
The problem started after updating to version 2.1.0. After I downgraded to version 2.0.3 the problem did not occur anymore.

jmpn89, 2019-07-10:
hi there,

I was having the same issues as described by you, but with a Sanitas SBF70.
the connection was established, but nothing was happening and no BT icon was shown on the scale screen, so most probably the connection was in fact not properly working.

I have been trying during few days to disable wifi and/or GPS before syncing -> not working
I tried also downgrading to 2.0.3 and got it working at the second attempt. 
I see already a feature that was lost with 2.1.0 -> error messages shown as toast notification.

additional info: 
-  using version from F-Droid
-  Xiaomi Mi6 with xiaomi.eu rom (v10.3.1.0)
-  Sanitas SBF70, with two users (second user works with no problem, Honor 9)

thanks!

oliexdev, 2019-07-14:
please provide the openScale log files so we can look into it to see what might the problem is.

jmpn89, 2019-07-16:
could you please tell me how to get the logs? I tried once to troubleshoot myself but I haven't succeed.

thanks

oliexdev, 2019-07-17:
@jmpn89 I updated the FAQ please read https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#how-to-get-the-openscale-debug-log-file

jmpn89, 2019-07-18:
hi, 

I managed to get the log:

`2019-07-18 17:27:13.906 Debug [2] AboutPreferences: Debug log enabled, openScale v2.1.0 (45), SDK 28, Xiaomi MI 6
2019-07-18 17:27:13.908 Debug [2] AboutPreferences: Selected user id(1) name(JMP) birthday(Wed Sep 27 00:00:00 GMT+01:00 1989) age(29) body height(178,00) scale unit(kg) gender(male) initial weight(93,00) goal weight(88,00) goal date(Fri Jul 26 00:00:00 GMT+02:00 2019) measure unt(cm) activity level(0)
2019-07-18 17:27:25.595 Debug [2] OpenScale: Trying to connect to bluetooth device [50:F1:4A:51:9E:80] (SANITAS SBF70)
2019-07-18 17:27:25.603 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-07-18 17:27:33.895 Debug [2] BluetoothCommunication: Found peripheral 'SANITAS SBF70'
2019-07-18 17:27:34.907 Debug [2] BluetoothCommunication: Try to connect to BLE device 50:F1:4A:51:9E:80
2019-07-18 17:27:50.227 Debug [2] BluetoothCommunication: connected to 'SANITAS SBF70'
2019-07-18 17:27:50.229 Debug [2] BluetoothCommunication: Step Nr 0
2019-07-18 17:27:50.231 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-07-18 17:27:50.231 Debug [2] BluetoothCommunication: Stop machine state
2019-07-18 17:27:50.233 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-07-18 17:27:50.246 Debug [2] MainActivity: Bluetooth connection successful established
2019-07-18 17:27:50.311 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2019-07-18 17:27:50.312 Debug [2] BluetoothCommunication: Resume machine state
2019-07-18 17:27:50.313 Debug [2] BluetoothCommunication: Step Nr 1
2019-07-18 17:27:50.314 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2019-07-18 17:27:50.314 Debug [2] BluetoothCommunication: Stop machine state
2019-07-18 17:27:55.307 Debug [2] BluetoothCommunication: disconnected 'SANITAS SBF70' with status 8
2019-07-18 17:28:50.272 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2019-07-18 17:28:50.273 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-07-18 17:28:50.300 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-07-18 17:28:56.236 Debug [2] AboutPreferences: Debug log disabled`

oliexdev, 2019-07-19:
@jmpn89 try out the latest dev version! :expressionless:

Tertiush, 2019-07-22:
I can confirm the latest dev release fixes this issue for my Yunmai mini. Any idea when you will be pushing this fix to the play store? The pro app is kinda in limbo without it. Thanks

oliexdev, 2019-08-11:
then I will close this issue. Currently I don't have any release plan.

closed 2019-08-11 by oliexdev

jmpn89, 2019-08-15:
> then I will close this issue. Currently I don't have any release plan.

how to sync with google fit from the dev version?