oliexdev/openScale #486: Runtastic Libra support?

Robyer, 2019-08-10:
I see in readme that Runtastic Libra is supported, but when I try to search for it in application, it found "Libra-B [mac address]" and on second line says "Device not supported". 

What am I doing wrong?

oliexdev, 2019-08-11:
please post the openScale log file, read the FAQ for more details

Robyer, 2019-08-13:
Here is the log. I tried to search for scales, then clicked on the Libra-B which was found (even though it said it's unsupported), waiting dialog was showed and then closed automatically. I clicked it once again with same result.

[openScale_2019-08-13_10-40.txt](https://github.com/oliexdev/openScale/files/3496067/openScale_2019-08-13_10-40.txt)

oliexdev, 2019-08-13:
your device name is a bit different than the other Libra scales. I added your device name, please try out the latest dev build of openScale and give feedback if it works or not.

Robyer, 2019-08-13:
Thanks, it works now.

closed 2019-08-13 by Robyer