oliexdev/openScale #487: Support for Shape Sense Connect 200

G0nzzzz, 2019-08-12:
Logs:

[shape_sense_connect_200-btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3491668/shape_sense_connect_200-btsnoop_hci.log)

[openScale_2019-08-07_08-37.txt](https://github.com/oliexdev/openScale/files/3491686/openScale_2019-08-07_08-37.txt)

I did three weight measurements: 
136,5kg (0555 big endian) found in package no. 1393:

> Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e001090a331c**0555**01dc019d

136,7kg (0557 big endian) not found - propably to similiar to the first measurement.

138,1kg (0565 big endian) found in package no. 7287:

> Bluetooth Attribute Protocol
>     Opcode: Handle Value Notification (0x1b)
>         0... .... = Authentication Signature: False
>         .0.. .... = Command: False
>         ..01 1011 = Method: Handle Value Notification (0x1b)
>     Handle: 0x0043 (Unknown: Unknown)
>         [Service UUID: 352e300028e940b8a3616db4cca4147c]
>         [UUID: 352e300128e940b8a3616db4cca4147c]
>     Value: 300100**0565**00000000

There was no body composition data shown in the app, allthough the scale shows muscle, fat and water percentages.

I used DCODE to find timestamps in the values before and after the weight string. Some Hex 32bit formats show weird time stamps, like from the year 1907 or 2050, but I did not find the exact date (May 27th, 2019, 9:33 MESZ)

If needed I will add more data. Just let me know.

G0nzzzz, 2019-08-13:
I redid the measurement to hopefully come up with more clear results.
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3496040/btsnoop_hci.log)

I did manage to identify the age/sex/height data that is asked for by the app, a hex string with todays date (August 13th 2019) - both in combination with the hex string of the measured weight.

What I did not found were hex strings of fat, water and muscle which were measured at 36,1%, 45,3% and 34,5% through all three measurements.

measurement1: weight 136,1kg hex:0551
found in package 996:

```
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101290000c5055102
                 ^^age
                   ^^sex?
                     ^^^^height
			 ^^^^weight
```

found in package 1325:
```
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101290000c5055102
                 ^^age
                   ^^sex?
                     ^^^^height
			 ^^^^weight
```

found in package 1413:
```
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e3080d090b00055111ff15dc
               ^^^^year
                   ^^month
                     ^^day
                       ^^hour
                         ^^minutes
                           ^^seconds
                             ^^^^weight
```

found in package 2852:
```
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
[... 141 lines cut ...]
                 ^^age
                   ^^sex?
                     ^^^^height
			 ^^^^weight
```

found in package 245:
```
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e3080d090f31059712000000
               ^^^^year
                   ^^month
                     ^^day
                       ^^hour
                         ^^minutes
                           ^^seconds
                             ^^^^weight
```

oliexdev, 2019-08-13:
could you make the values more clearly what is what? With the ^^^ marks is a bit difficult to read. Also please write down the true values for sex, age, measurement time and so on. Also would be great if you could do a measurement with a larger different.

> What I did not found were hex strings of fat, water and muscle which were measured at 36,1%, 45,3% and 34,5% through all three measurements.

Maybe only the impedance value are transmitted and the calculation is done in the vendors app!?

G0nzzzz, 2019-08-13:
> could you make the values more clearly what is what? With the ^^^ marks is a bit difficult to read. 
> Also please write down the true values for sex, age, measurement time and so on. 

value of package 996: 0c0101290000c5055102
`values: 0c0101(unknown), 29(age 41), 00(sex? male), 00c5(height 197cm), 0551(weight 136,1kg), 02(unknown)`

I'm not quit sure whether height is encoded in 4 digits (00c5) or only 2 digits (c5). If its encoded in two digits, the purpose/meaning of the other two is unknown.

value of package 1413: 090107e3080d090b00055111ff15dc
`values: 0901(unknown), 07e3(year 2019), 08(month 08), 0d(day 13), 09(hour 09), 0b(minutes 11), 00(seconds 00), 0551(weight 136,1kg), 11ff15dc(unknown)`

> Also would be great if you could do a measurement with a larger different.

Yeah, I will get a case of beer today and repeat the measurement :)

> Maybe only the impedance value are transmitted and the calculation is done in the vendors app!?

Can those three values be derived from only one impedance value? In the packages with the timestamp, there are 8 unknown digits at the end of the value string which stay very much the same over the course of all measurements.

oliexdev, 2019-08-14:
>  values: 0c0101(unknown), 29(age 41), 00(sex? male), 00c5(height 197cm), 0551(weight 136,1kg), 02(unknown)

you could easily chekc if 00 male or female if you switch the sex in the vendors app and log again.

> Yeah, I will get a case of beer today and repeat the measurement :)

But don't drink while your measuring :wink: 

> Can those three values be derived from only one impedance value? In the packages with the timestamp, there are 8 unknown digits at the end of the value string which stay very much the same over the course of all measurements. Maybe the impedance or fat/water/muscle value is encoded there.

Yes combined with your sex, age, height you can calculate the other values, see for example YunmaiLib.java or OneByoneLib.java classes.

oliexdev, 2019-08-15:
I added a first test implementation with commit https://github.com/oliexdev/openScale/commit/87b62ed6e2e30799eba17b70a0ffba54f47e574c please test it and post the log files with the [branch dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_SoehnleScale200-build).

I looks like they are using the BLE weight standard profile, see https://www.bluetooth.com/specifications/gatt/characteristics

But on the first look they are transmitting the impedance value and do the calculation in the app itself.

@G0nzzzz are you java developer?

G0nzzzz, 2019-08-16:
Wow that was fast :) I'll give it a try on sunday and report back.

I've done some basic Java stuff, yeah.

oliexdev, 2019-08-18:
@G0nzzzz I have some formula for the body fat, water and muscle but I need three different proper measurements from you with the following data to evaluate it:

* weight
* height
* age
* activityLevel
* attach hci log file for this measurement

please get those values from latest vendors app.

G0nzzzz, 2019-08-19:
@oliexdev I've tried the dev version this morning and noticed:

- Pairing works fine, scale is discovered and connection is established, see also openScale log: 
[openScale_2019-08-19_09-06.txt](https://github.com/oliexdev/openScale/files/3515080/openScale_2019-08-19_09-06.txt)
- when I step on the scale after connecting, a weight measurement is taken and the app chrashes with the exception:

```
Build version: 2.1.0-dev_87b62ed6_2019-08-15
Current date: 2019-08-19 09:09:41
Device: Samsung SM-A520F

Stack trace:
java.util.UnknownFormatConversionException: Conversion = 'i'
at java.util.Formatter$FormatSpecifier.conversion(Formatter.java:2730)
at java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2759)
at java.util.Formatter$FormatSpecifierParser.<init>(Formatter.java:2591)
at java.util.Formatter.parse(Formatter.java:2524)
at java.util.Formatter.format(Formatter.java:2472)
at java.util.Formatter.format(Formatter.java:2426)
at java.lang.String.format(String.java:2626)
at com.health.openscale.core.bluetooth.BluetoothSoehnle.handleBodyCompositionMeasurement(BluetoothSoehnle.java:160)
at com.health.openscale.core.bluetooth.BluetoothSoehnle.onBluetoothNotify(BluetoothSoehnle.java:89)
at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:338)
at com.welie.blessed.BluetoothPeripheral$1$7.run(BluetoothPeripheral.java:481)
at android.os.Handler.handleCallback(Handler.java:751)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:154)
at android.app.ActivityThread.main(ActivityThread.java:6186)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

> @G0nzzzz I have some formula for the body fat, water and muscle but I need three different proper measurements from you with the following data to evaluate it: [...]

I will provide you with new measurements this week.

G0nzzzz, 2019-08-21:
Additional measurements, as promised:

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3525954/btsnoop_hci.log)

measurement 136,3kg -> 0553
found in package 914

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101270100c5055302

found in package 1044

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e30815081f37055312000000

found in package 1864

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e30815081f37055312000000

found in package 2225

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e30815081f37055312000000

measurement 140,3kg -> 057B
found in package 1450

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101270100c5057b02

found in package 1768

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101270100c5057b02

found in package 1865

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e3081508201c057b11ff0000

found in package 1866

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e30815082105057b12020000

measurement 144,8kg -> 05A8
found in package 2143

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 0c0101270100c505a802

found in package 2228

Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1011 = Method: Handle Value Notification (0x1b)
    Handle: 0x0043 (Unknown: Unknown)
        [Service UUID: 352e300028e940b8a3616db4cca4147c]
        [UUID: 352e300128e940b8a3616db4cca4147c]
    Value: 090107e3081508212705a812010000

oliexdev, 2019-08-21:
@G0nzzzz thats not what I need please read again https://github.com/oliexdev/openScale/issues/487#issuecomment-522334749

G0nzzzz, 2019-08-21:
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3526032/btsnoop_hci.log)

``` 
weight 136,3kg
height 197cm
age 41
activityLevel 2
fat in percent 45,3%
water in percent 43,1%
muscle in percent 31,4%
```
```
weight 140,3kg
height 197cm
age 41
activityLevel 2
fat in percent 45,3%
water in percent 43,1%
muscle in percent 31,4%
```
```
weight 144,8kg
height 197cm
age 41
activityLevel 2
fat in percent 45,3%
water in percent 43,1%
muscle in percent 31,4%
```

> please get those values from latest vendors app.

I used the latest vendor app version available to me, which is Soehnle_Connect_5.1.2

G0nzzzz, 2019-08-21:
> @G0nzzzz I have some formula for the body fat, water and muscle but I need three different proper measurements from you with the following data to evaluate it

Or do you mean: three different measurements over the course of several days, without adding artificial weight?

oliexdev, 2019-08-21:
can you double check you fat, water, muscle values? Did you weight yourself with barefoot? Because it can't be always the same values...

G0nzzzz, 2019-08-21:
Propably it's the same value because I was not standing on the scale long enough to take a new body impedance measurement. Will do again.

oliexdev, 2019-08-21:
@G0nzzzz thanks but the impedance measurements are the important information which I need now.

G0nzzzz, 2019-08-22:
@oliexdev here are three new measurements with varying fat/water/muscle values:

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3530949/btsnoop_hci.log)

```
weight 137kg
height 197cm
age 41
activityLevel 2
fat in percent 47,2%
water in percent 41,6%
muscle in percent 29,2%
```

```
weight 137kg
height 197cm
age 41
activityLevel 2
fat in percent 44,1%
water in percent 44,3%
muscle in percent 31,8%
```

```
weight 144,8kg
height 197cm
age 41
activityLevel 2
fat in percent 44,7%
water in percent 43,8%
muscle in percent 31,3%
```

oliexdev, 2019-08-22:
@G0nzzzz thanks my formula works. the last bytes are the impedance values. I added in your comment the associated hex bvtes.
Could you log the user creation process? Please delete for that every user in the vendor app and reset your scale. Then log the Bluetooth communication if you **first time** connect to the scale. Please write down which sex, age, height and weight activity level you choose for the creation.

G0nzzzz, 2019-08-22:
@oliexdev I'm not at home during the weekend, but will do as soon as I'm back.

oliexdev, 2019-09-02:
@G0nzzzz any updates?

G0nzzzz, 2019-09-03:
@oliexdev I deleted all previous users and created a new one:
Name: yoogiibaaeer
Birthday: January, 1st, 1980
Sex: male
Height: 197cm
Activity Level: 2
Weight: 137,5kg

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3568169/btsnoop_hci.log)

oliexdev, 2019-09-08:
@G0nzzzz thanks please try out the latest [soehnle dev branch](https://github.com/oliexdev/openScale/releases/tag/travis-dev_SoehnleScale200-build) and post the openScale log file.
Before you test it please reset your scale (or delete all scale users), so that openScale need to create a new user for the scale.

G0nzzzz, 2019-09-15:
Deleted the existing user with Soehnle app: 
[btsnoop_hci-deleteuser.log](https://github.com/oliexdev/openScale/files/3613605/btsnoop_hci-deleteuser.log)

soehnle dev branch log: 
[openScale_2019-09-15_11-13.txt](https://github.com/oliexdev/openScale/files/3613606/openScale_2019-09-15_11-13.txt)

oliexdev, 2019-09-15:
could you weight again and note down your weight, height, age, activityLevel, fat in percent, water in percent and muscle in percent. Need to check something.

Do you see something on the scale if the user was successful created on the scale?

G0nzzzz, 2019-09-17:
@oliexdev User is created, but height is not correct (should be 197, is 50432), see screenshot.

Name: yogibaer
Birthday: January, 1st, 1980
Sex: male
Height: 197cm
Activity Level: 2
Weight: 137,5kg

[image]

oliexdev, 2019-09-17:
thanks the body height should be fixed in the latest dev build, please try it out. 

Please also weight again and note down your weight, height, age, activityLevel, fat in percent, water in percent and muscle in percent and attach the openScale log file. I need to check if the hex values are transmitted.

G0nzzzz, 2019-09-18:
Body height is corrected.

weight 136,5kg
height 197cm
age 39
activityLevel 2
fat in percent 36,4%
water in percent 45,2%
muscle in percent 33,8%

[openScale_2019-09-18_08-01.txt](https://github.com/oliexdev/openScale/files/3625302/openScale_2019-09-18_08-01.txt)

oliexdev, 2019-09-18:
please try out the latest dev build again and post the openScale log file with the corresponding weight and so on like before.

G0nzzzz, 2019-09-24:
weight 136,0kg
height 197cm
age 39
activityLevel 2
fat in percent 34,9%
water in percent 46,3%
muscle in percent 34,7%

[openScale_2019-09-24_08-26.txt](https://github.com/oliexdev/openScale/files/3647190/openScale_2019-09-24_08-26.txt)

oliexdev, 2019-09-27:
the attached log file looks very strange. They don't go to any steps. Could you double check it and do the logging again?

G0nzzzz, 2019-10-21:
@oliexdev

I've done the logging again, see attached file. I noticed two strange things: the bluetooth connection is unsuccessful as long as the location permissions aren't set/enabled. The bluetooth connection is successfully enabled then, but no values of measurements are transmitted to the app.

weight 135,4kg
height 197cm
age 39
activityLevel 2
fat in percent 36,7%
water in percent 44,2%
muscle in percent 33,5%

oliexdev, 2019-11-05:
@G0nzzzz thanks could you test the latest dev branch version? It should work now.

G0nzzzz, 2019-11-13:
@oliexdev Wight, BMI, water, muscle and fat values are transfered to the app now. BMR and TDEE are not displayed correctly:

[openScale_2019-11-13_08-03.txt](https://github.com/oliexdev/openScale/files/3840266
/openScale_2019-11-13_08-03.txt)

[image]

G0nzzzz, 2019-11-14:
Oh, and there seems to be an issue with the date. In the second last row of the screenshot you can see the year displayed as "58119"

oliexdev, 2019-11-14:
@G0nzzzz thanks the TDEE and BMR are calculated with the official formula from wikipedia. 
Did the date problem happens every time? In the log file you have the correct time. It is a bit strange.
Could you post a photo of your scale with the same camera angle like in our wiki gallery?

oliexdev, 2019-11-20:
the date bug should be fixed now

closed 2019-11-20 by oliexdev