oliexdev/openScale #488: Reverse engineering Sanitas SBF70

retrodimi, 2019-08-14:
I want to reverse engineer the Sanitas SBF70 scale.

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/3500536/btsnoop_hci.log)

I tried to follow a similar procedure as described by https://github.com/oliexdev/openScale/issues/71

Otherwise I was able to decipher one part of the values so far.

Here are some examples:

`1b:2e:00:e7:4c:02:01:5d:57:76:b1:07:07:01:e2:00:b9:02`
`1b:e2:00:e7:59:03:02:5d:51:76:b1:07:07:01:e2:00:00:00`
`1b:2e:00:e7:59:03:03:00:00:00:00:00:00:00:00:00:00:00`
`'1b:2e:00:e7:4c:02:02:20:01:99:01:18:08:29:0c:3e:00:f1`

Bytes: 0-3 do not change
Bytes: 4-6 are changing
Bytes: 7-10 unix timestamp or Date which can be converted from hex to dec to unix / real time date
Bytes: 12-17 are all changing again.

For e.g 5d:51:76:b1 is 12.08.2019 - 16:24:49 which is the accurate date of one measurement.

I do assume that 14-15 could be the body fat, by pressuming the pattern is similar to the Yunmai scale reverse engineering process.

What I am missing is the following information

Weight in kg 
Body fat %
Muscle  %
Water %
Bone mass kg

I am not sure if other information could also be inside the values, from the profile I am using the health coach app from Sanitas, which is tedious to use. I tested the scale with openscale and I can't measure my body weight there, yet I receive values in the logs, yet the value itself is not stored in the app, all it shows is a small message below with the current weight the scale measures.

So, I am not sure if the following metrics are inside the values above:
Age
Height
"Weight"
Gender

Which are displayed in the profile.

Did anyone else test this scale and received different results, as well as has any idea on how to find out the values for instance of body fat, weight etc ?

I will continue to work on it, till I find some of the values. Any idea or ressources on how to gain access to the values above?

So far I compared the results from the yunmai scale and tried to find out the numbers via the hex converter https://www.binaryhexconverter.com/hex-to-decimal-converter, as described in the guide by openscale.

oliexdev, 2019-08-14:
we already have done it? See https://github.com/oliexdev/openScale/wiki/Beurer-Sanitas

closed 2019-08-14 by oliexdev

retrodimi, 2019-08-20:
> we already have done it? See https://github.com/oliexdev/openScale/wiki/Beurer-Sanitas

Thank your for mentioning it ! I tried to still reverse engineer it via the link provided and wanted to ask in case it is fine to ask, if you found the 22 Byte Measurement.

22 bytes of data:
<timestamp> <weight> <impedance> <fat> <water> <muscle> <bone> <BMR> <AMR> <BMI>

Can this be found in the btsnoop_hci.log ? The logs with 22 Bytes did not contain the values above, and I can't also find a log which has the values. I used different formulars, yet the values are a bit off. 

---------------------------------------------------------------------------------------------------------------

I tried the link below and. I have every value besides <water> <muscle> <bone> <BMR> <AMR> <BMI> and I can't finde the log with 22 Measurement of bytes which could contain the values, any help with that?

https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

oliexdev, 2019-08-27:
the 22 bytes should be found in the btsnoop_hci.log. What do you have already programmed? Please paste your code here and what do you want to achieve / what is your project?

retrodimi, 2019-08-28:
I am not allowed to code, yet I talked with the professor yet he wants me to have ideally all of the measurements first. My project is to learn how to reverse engineer scales and I received the Sanitas SBF70 scale and implement it into the campuses fitness app, the professor is not there, yet we are skyping regularly, because it is his project. I will receive a couple more scales in a week approx.

I searched for the 22 Measurement and looked into all of the measurement with 22 Byte available, and what I did not find is:

Muscle %
Water %
Bone mass kg

I did find the weight and the body fat as well as the impedance and I am using this repository now to find different formulars, I can't use openScale I looked through the openScale code. 

https://github.com/oliexdev/openScale/issues/71

I can't use open scale to compare the formulars that I found in the repository as well as on the net and I am still missing muscle % as well as bone mass in kg, if I would find the 22 Byte measurement. I could start coding, instead of looking for these things for a couple of weeks now. Because apparently all of the measurements are included in the 22 Byte measurement.

I looked into most 22 Byte measurements of the btsnoop_hci.log and tested different logs and user profiles to receive different values, yet none of them yielded any results about muslce %, water % and bone mass etc. I could not test this with openScale since I was not able to use it successfully with the Santias SBF70 scale. So, I used the vendors app "Health Coach" which costs a lot of time to just get a few measurements. 

I keep looking for formulars and try to find a solution or an approximation to the values of the health coach.

oliexdev, 2019-08-28:
@retrodimi which professor, campuses fitness app and which university do you mean? Do you have a project site?

Could you post an openScale log file while you trying to connect to the scale?

retrodimi, 2019-08-28:
Well, yes I am not sure if the professor matters. I can send you the website of the project you can also download it on the app store. The campus is in Germany. 

https://www.nsoria.io

The app can also be downloaded on the app store.

https://play.google.com/store/apps/details?id=com.nsoria.fit&hl=de

I will send the openScale log below. My task is to reverse engineer the scales during the internship here, and I can do my bachelor thesis after that, so it would be great if I can make progress.

[openScale_2019-08-28_15-08.txt](https://github.com/oliexdev/openScale/files/3551182/openScale_2019-08-28_15-08.txt)

The file contains the first successful measurement with openScale dev version it just suddenly worked, I will check the hci_log now and also create a synonymous user on the "HealthCoach App" since I've read here in the repositories that his is the right prodcedure is the procedure correct ?

oliexdev, 2019-08-28:
Do you know that you might run into a legal issue? Please be aware that you need to comply with the GPLv3 license with your task if you implement it into n*soria which is closed source.

retrodimi, 2019-08-29:
> Do you know that you might run into a legal issue? Please be aware that you need to comply with the GPLv3 license with your task if you implement it into n*soria which is closed source.

I am not going to implement it into the project I am going to learn how to reverse engineer scales and to learn programming with android studio. So, I will receive some test code and learn how it could be done. 

I've read the GPLv3 now and I from what I understand the license is included for people who want to protect their product and or service and the lincense holder (company) has to distrubute the license for each user or applicant of the app/software. Yet, what I perceive is that I am neither modifying their source code nor do any coding based on their covered work. 

I am reading the bluetooth protocols and implement it and will receive test code based on Nsoria's* research. Also, we are protected by the goverment as "Freiheit der Lehre" by the constitution. As far as I know.

I understand that the GPLv3 protected services can be used in open source and not closed source, yet I am not doing anything that falls under the "protection" of the license. Because I am reading out the bluetooth protocols, which is legal and is neither covered work nor object code or a conveyance of some sort. 

I mean you can see where the data goes to with Wireshark and if the company protects it properly I can't even reverse engineer it from what I read. So, I don't quite get why this could be a legal issue.

oliexdev, 2019-08-29:
> I am not going to implement it into the project I am going to learn how to reverse engineer scales and to learn programming with android studio. So, I will receive some test code and learn how it could be done.

If you don't implement it into n*soria then it is fine.

> I've read the GPLv3 now and I from what I understand the license is included for people who want to protect their product and or service and the lincense holder (company) has to distrubute the [...]

The basic idea is that if you use/modify GPLv3 source-code the derived work have to be under the same GPLv3 license and so it must be also open-sourced. So the license will basically protect that someone will copy your source code and use it in a closed source product. You get something from the open-source community and you will return something to the open-source community back, isn't it fair?

> I am reading the bluetooth protocols and implement it and will receive test code based on Nsoria's* research. Also, we are protected by the goverment as "Freiheit der Lehre" by the constitution. As far as I know.

The "Freie Lehre" has nothing to do with the GPLv3 license. You still has to follow the license rules.

> I understand that the GPLv3 protected services can be used in open source and not closed source, yet I am not doing anything that falls under the "protection" of the license. Because I am reading [...]

You already read the openScale source-code or based on our information, which you stated before, to try to implement it. Again, I have no problem if your version is under GPLv3 but reading out the source code or gain information from an open-source project to implement it into a closed project is illegal and morally doubtful.

Your professor Mr Hendrick Speck contacted me in January 2019. He already wanted to profit from  openScale for n*soria. I already stated that we can cooperate if n*soria were open-source. In the end he was rethinking about his license concept vs. to monetize n*soria. I find it really interesting that he let his internship and students do the reverse engineer work which I now find here asking for help? What would you do in my position @retrodimi ?

retrodimi, 2019-08-29:
> If you don't implement it into n*soria then it is fine.

Well, yes I am learning all of this still, so I am not going to implement it. 

> The basic idea is that if you use/modify GPLv3 source-code the derived work have to be under the same GPLv3 license and so it must be also open-sourced. So the license will basically protect that [...]

Well, I do understand that I found a couple of things mainly through the repositories and based on Nsoria's* research, so I did not modify or us any GPLv3 source-code. Is looking at it alone falling under the GPLv3 license ? When I implement nothing based on it ? Or looking at the repositories ?

> The "Freie Lehre" has nothing to do with the GPLv3 license. You still has to follow the license rules.
Yes, I just read the Wiki entry about this for a bit and you are right.

> You already read the openScale source-code or based on our information, which you stated before, to try to implement it. Again, I have no problem if your version is under GPLv3 but reading out the [...]

I've read it but I did not implement anything nor did I intend to I don't quiet understand why reading the source code for information, yet implement it based on ones own research is stricitly forbidden and morally doubtful. It's not that code is copy and pasted. Of course I could say it's morally doubtful since it seems like code is modified and still I would not understand the concept of open-source to some degree. For sure it's great that a community can exchange information within it's own community, yet I don't quite understand why others can not be inspired by it and use it in their own projects to make profit for instance or provide a quality product for people, instead of manufactures providing apps that are horrible to use. Since, they don't invest.

> Your professor Mr Hendrick Speck contacted me in January 2019. He already wanted to profit from openScale for n_soria. I already stated that we can cooperate if n_soria were open-source. In the end [...]

I can understand your position as a developer of openScale putting in a lot of effort into building a product and sharing it as open-source to share it with others and have a community which gives support and get's support. I can understand that if the license is not respected that it feels unfair and morally disgusting, to see ones source-code implemented in for instance samsung health etc. Or that you prefer for things to be ones own proprietary, and that ultimately you want to decide what happens to the product/service as well as you may don't want the product to be abused or misused ? As well as open-soruce should be open-source ? So closed-source has no access to it ?

One thing I don't understand is sure it's good that the community is protected and people can benefit from each other, yet that seems rather esoteric in it's literal definition of the word or in other words ethnocentric. I could say the same vice-versa with closed source projects. Also, I am not familair with the license etc. since you are also selling it on the appstore, yet offer the source code for free on the net. It is considered open-source is that correct ? Sure I am personally for the decision that open-source should be protected when I put myself in their shoes or simply empathizes, yet I am not implementing anything from the source-code and reading it to look for a solution, yet implementing it on ones own without the source-code would already be a modification ? Not an inspiration or way to find alternative ideas ? 

The project was offered by the professor and I was interested in it because I really like fitness and android programming, so it was ideal. I work out a lot and many apps are just a travesty, especially the scale apps I've seen so far. 

For sure I can ask him for help, yet he really wants students to learn something, so it's a bit different from other professors I would not have choosen this professor, if he would not want his students to learn something and a lot of students went afterwards to very good companies etc. And I just had a horrible professor. Not every prof, takes care so much and wants students to learn something.

I can't tell you what to do and what not to do, all I wanted to know is where I can find the 22 byte measurement since I did not find it in wireshark, and I have to find the formulars on my own on the net, I've already found some stuff, to get close to the values through sheer approximation. So, yes I was curious where I can find a solution for the wireshark issue.

As I said above again, I am not implementing it, I wanted to know the 22 Byte Measurement so I can start programming and reverse engineer different scales. Based on what Nsoria* has done, not on openScale and based on what I can find out through reading the Bluetooth protocols.

oliexdev, 2019-08-29:
> Well, yes I am learning all of this still, so I am not going to implement it.

You wrote you are not implement it but then you wrote at the same time:

> ...so I can start programming...

or 

> ..and implement it into the campuses fitness app..

strange isn't it?

> Well, I do understand that I found a couple of things mainly through the repositories and based on Nsoria's* research, so I did not modify or us any GPLv3 source-code. Is looking at it alone falling under the GPLv3 license ? When I implement nothing based on it ? Or looking at the repositories?

Sure you can copy source-code just by looking at it, even if you implement it into another programming language it is copying, see https://www.gnu.org/licenses/gpl-faq.html#TranslateCode. It is like If you write your Bachelor thesis and read some really interesting research paper and you copy the structure or even whole sentences or the content you have to cite it and in this case you have to follow the GPLv3 license.

> For sure it's great that a community can exchange information within it's own community, yet I don't quite understand why others can not be inspired by it and use it in their own projects to make [...]

I agree the vendors app are horrible from the privacy aspect and your are welcome to fork or modify the openScale source code under the terms of the GPLv3.

> I can understand your position as a developer of openScale putting in a lot of effort into building a product and sharing it as open-source to share it with others and have a community which gives [...]

Yes that's the main point for the GPLv3 open-source should be open-source! They are different open-source licences out there that can be used in closed source like n*soria using the open-source library MPAndroidChart but this is based on Apache 2.0 license which is allowed. I recommend that you read the GPLv3 FAQ https://www.gnu.org/licenses/gpl-faq.html . In your case especially https://www.gnu.org/licenses/gpl-faq.html#WhatIfSchool

> One thing I don't understand is sure it's good that the community is protected and people can benefit from each other, yet that seems rather esoteric in it's literal definition of the word or in [...]

How you could say that vice-versa with closed source? How can another project benefit from the n*soria source code or their modifications? You can earn money with open-source software as long as you provide the source-code.

> The project was offered by the professor and I was interested in it because I really like fitness and android programming, so it was ideal. I work out a lot and many apps are just a travesty, [...]

Well, I've never met a professor who didn't want that his students learn something :wink: 

> I can't tell you what to do and what not to do, all I wanted to know is where I can find the 22 byte measurement since I did not find it in wireshark, and I have to find the formulars on my own on [...]

All I said is that you have to be aware of the GPLv3 license. We had in the past a copyright violation, see issue #63

retrodimi, 2019-08-29:
To clarify: My professor did insist, that I shall NOT use any code from OpenScale. He actually clarified that I shall not write any line of code at all (for the moment). I am working with Wireshark & Co. and there is certainly no violation of any kind as you have incorrectly assumed. Our academic research project n*soria has independent algorithms, interfaces, functions, statistics, focus, and goals. N*soria is absolut transparent in regards to sources, licenses, transparency report, canary, privacy policy etc, and can actually serve as an example for other apps (as you are aware).

The problem with the (more complex) n*soria framework (where weight is only one subsection) are the currently lacking OpenSource classes, GPL incompatible functions, or services that need to be licensed by third parties. This includes for instance classes for ANT+ (Bike Computers), tracks, elevation or weather data, or several APIs from third parties and platforms, which will work only with a proper key / agreement / license.

A "clean" GPL variant would lead. As you are fully aware, to a reduced functionality, devices that can not be integrated, lacking data sets, worse results. This might in the long term promote the open source idea, but it would punish users at the moment, who might be looking for an alternative, noncommercial, modular, manufacturer agnostic project. n*soria does not follow any hidden political agenda, but defends the interests of the users, offers decentralized data storage, options and informational self determination. 

That being said, my professor just informed me, that he did offer you access to the source base of n*soria, following the spirit of academic exchange. Your statement that he wanted to "profit" from your code is thus a bit limited. The same offer stands for the other devices and scales that I will be working on in the next weeks in my practical training here in our university, 
We are willing to share, and we have nothing to hide :-). 

I asked you politely as I am still learning how to reverse engineer these scales. My only interest is in understanding the principles, so that I can apply these skills to the other scales waiting in the lab.

I am a student at the university and I enjoy the privilege to choose and propose the topics for my practical training and the bachelor thesis (as this is customary in Germany). So its not the professor "who lets me do this", but I was free to decide on my own will.

Aside from the procedure of reverse engineering (where I am a beginner) the discussion focuses (only) on 22 Bytes of a Bluetooth protocol of the scales, which are implemented by the manufacturers of chipsets / middleware, distributed by the different brands. We started with HRM, power, cadence, blood pressure, are now working on scales and will then look into integrated smart watches / fitness trackers etc. 

From IPRHelpdesk.eu "Copyright protection [in Europe] extends to any element of expression of the creativity of its author but not to the ideas behind it, procedures, methods of operation, or mathematical concepts as such.

In other words, an algorithm is not eligible for copyright protection, because it will be considered to be of a factual nature, and therefore not an expression of the creativity of its author.
Following the aforementioned, copyright will protect only the computer program in the form written by a programmer i.e. its source code. Neither the functionality of a computer program, nor the programming language nor the format of data files used in a computer program in order to exploit certain of its functions constitute a form of expression of that program, and thus these are not protected by copyright."

Translated: Your code, your design, your interface seem rightfully to enjoy the copyright / copyleft claim you are making (as long as they are your own, which we assume). The same argument however does (at least in Europe) not apply to procedures, methods of operation or mathematical concepts - especially when they in reality originate from the manufacturers of the scales.

It would be great to have you in one of our coding classes or in a seminar, it would be great if you would cooperate this is a only about a 22 byte measurement and I found most of the other metrics, besides the muscle mass in % for now. We could also hold a skype session to discuss and clarify potential issues. I am very interested in learning how to reverse engineer, so talking to you would be great !

Thank you !

oliexdev, 2019-09-02:
> To clarify: My professor did insist, that I shall NOT use any code from OpenScale. He actually clarified that I shall not write any line of code at all (for the moment). I am working with Wireshark [...]

Maybe your professor is full aware of the issue and that's why he insist you do not use the openScale source code!? I already mentioned your potentially GPLv3 violation in the previous post. 
N*soria is *not* transparent regarding his developing, source-code, financial support, his goal, function and is actually a great example how an app benefits from open-source and try to make profit of it in a closed source program (as your aware).

> The problem with the (more complex) n*soria framework (where weight is only one subsection) are the currently lacking OpenSource classes, GPL incompatible functions, or services that need to be [...]

Then why not support the open-source community and develop those classes? If everybody thinks like you we would stuck. There are open-source licenses out there which are compatible with closed sourced third party libraries, you can choose for instance the MIT license. You can particularly open some functions. Or you can do it like openScale provide a closed sync app and an open-source app. There are many ways to do it. So this is a false argument and your professor knows that. 

> That being said, my professor just informed me, that he did offer you access to the source base of nsoria, following the spirit of academic exchange. Your statement that he wanted to "profit" from [...]

What should I do with the nsoria code when it has no license which I can use? Publish your source-code with a open source license! Publish your information from your other information which you already gained from other devices? Publish white papers or research papers? Isn't nsoria a research project or a commercial project? You project github  (which follow me since months) has no public repository or information. When was n*soria founded? 2015? So time could not be reason do not to do it.

> I asked you politely as I am still learning how to reverse engineer these scales. My only interest is in understanding the principles, so that I can apply these skills to the other scales waiting in [...]

And now openScale gives you a great starting point. 

> Aside from the procedure of reverse engineering (where I am a beginner) the discussion focuses (only) on 22 Bytes of a Bluetooth protocol of the scales, which are implemented by the manufacturers of [...]

> From IPRHelpdesk.eu "Copyright protection [in Europe] extends to any element of expression of the creativity of its author but not to the ideas behind it, procedures, methods of operation, or [...]

As I already mention it's not only my copyright (and btw the copyright from the openScale collaborator too!) you have to follow the GPLv3 license and this includes the source code as well. 

> It would be great to have you in one of our coding classes or in a seminar, it would be great if you would cooperate this is a only about a 22 byte measurement and I found most of the other metrics, [...]

Again I have no problem we are working together but it can't be a one sided. Gain trust in the open-source community and work together with open-source community! nsoria have to do their steps!

Btw I did a quick google search about n*soria and your professor Hendrick Speck and found the following statements:

>  Das bringt uns zurück auf das Thema Hürden bei der Entwicklung: n*soria ist ja eine Free Open Source Software. 

source https://www.hs-aalen.de/de/news/2525

Where is n*soria is open source software? 

> doch jeder, der will,kann die Software verwenden, um neue, ei-gene apps zu programmieren, unabhängigvon Großkonzernen. Frei nach dem motto:mein lauf gehört mir.

source http://magazin.spiegel.de/EpubDelivery/spiegel/pdf/134337115

How can I use your software to make my own app?

> Datenschutz, Verschlüsselung und Cloudanbindung deutlich von ihren Konkurrenten absetzen und die datenschutzrechtlichen Interessen der Nutzer berücksichtigen.

source https://www.pressebox.de/pressemitteilung/fachhochschule-kaiserslautern/Hochschule-Kaiserslautern-stellt-auf-CeBIT-aus/boxid/730746

If I read your data privacy your fundamental privacy principle is:
"We aggregate certain information about our users use of the Services for business and development purposes and we sometimes publish that information or share it with others."

which sound like any other company. 

What gain more trust to the people if n*soria publish their source-code, so that everybody can look into it and check if the nsoria app don't share my personal health data?