oliexdev/openScale #490: Support request yunmai mini2

codepuregit, 2019-08-21:
dev log.

2019-08-21 22:24:12.177 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_d6d17cdf_2019-08-20 (1566309609), SDK 28, Xiaomi POCOPHONE F1
2019-08-21 22:24:12.190 Debug [2] AboutPreferences: Selected user id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)
2019-08-21 22:24:16.141 Debug [2] OpenScale: Trying to connect to bluetooth device [6C:C3:74:D9:51:3F] (YUNMAI-ISM2-W)
2019-08-21 22:24:16.157 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-08-21 22:24:16.166 Debug [2] BluetoothCommunication: Stop machine state
2019-08-21 22:24:16.728 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-ISM2-W'
2019-08-21 22:24:17.736 Debug [2] BluetoothCommunication: Try to connect to BLE device 6C:C3:74:D9:51:3F
2019-08-21 22:24:22.305 Debug [2] BluetoothCommunication: connected to 'YUNMAI-ISM2-W'
2019-08-21 22:24:22.307 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-08-21 22:24:22.309 Debug [2] BluetoothCommunication: Resume machine state
2019-08-21 22:24:22.310 Debug [2] BluetoothCommunication: Step Nr 0
2019-08-21 22:24:22.317 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B] on 0xffe9
2019-08-21 22:24:22.318 Debug [2] BluetoothCommunication: Step Nr 1
2019-08-21 22:24:22.320 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 5D 46 06 00 00 00 00 00 5C] on 0xffe9
2019-08-21 22:24:22.321 Debug [2] BluetoothCommunication: Step Nr 2
2019-08-21 22:24:22.322 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4
2019-08-21 22:24:22.322 Debug [2] BluetoothCommunication: Stop machine state
2019-08-21 22:24:22.329 Debug [2] MainActivity: Bluetooth connection successful established
2019-08-21 22:24:22.350 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-21 22:24:27.435 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-ISM2-W' with status 8
2019-08-21 22:25:00.550 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-08-21 22:25:00.551 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-08-21 22:25:00.563 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2019-08-22:
@codepuregit 

could you test out the ideas by @weliem:

* The scale is out of battery and hence behaving erratic. Please replace the batteries with new ones
* The scale is a 'bad state'. Factory resetting it or pulling the batteries briefly may solve the issue.
* The Bluetooth stack on the phone is messed up. Reboot your phone and try again.

codepuregit, 2019-08-26:
> * The scale is out of battery and hence behaving erratic. Please replace the batteries with new ones
No effect
> * The scale is a 'bad state'. Factory resetting it or pulling the batteries briefly may solve the issue.
No effect
> * The Bluetooth stack on the phone is messed up. Reboot your phone and try again.
No effect

2019-08-26 21:29:50.606 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_d6d17cdf_2019-08-20 (1566309609), SDK 28, Xiaomi POCOPHONE F1
2019-08-26 21:29:50.617 Debug [2] AboutPreferences: Selected user id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)
2019-08-26 21:29:53.216 Debug [2] OpenScale: Trying to connect to bluetooth device [6C:C3:74:D9:51:3F] (YUNMAI-ISM2-W)
2019-08-26 21:29:53.221 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-08-26 21:29:53.230 Debug [2] BluetoothCommunication: Stop machine state
2019-08-26 21:29:54.204 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-ISM2-W'
2019-08-26 21:29:55.212 Debug [2] BluetoothCommunication: Try to connect to BLE device 6C:C3:74:D9:51:3F
2019-08-26 21:29:57.250 Debug [2] BluetoothCommunication: connected to 'YUNMAI-ISM2-W'
2019-08-26 21:29:57.253 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-08-26 21:29:57.254 Debug [2] BluetoothCommunication: Resume machine state
2019-08-26 21:29:57.256 Debug [2] BluetoothCommunication: Step Nr 0
2019-08-26 21:29:57.261 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B] on 0xffe9
2019-08-26 21:29:57.262 Debug [2] BluetoothCommunication: Step Nr 1
2019-08-26 21:29:57.264 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 63 D0 C5 00 00 00 00 00 37] on 0xffe9
2019-08-26 21:29:57.265 Debug [2] BluetoothCommunication: Step Nr 2
2019-08-26 21:29:57.265 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4
2019-08-26 21:29:57.266 Debug [2] BluetoothCommunication: Stop machine state
2019-08-26 21:29:57.271 Debug [2] MainActivity: Bluetooth connection successful established
2019-08-26 21:29:57.319 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-26 21:30:02.346 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-ISM2-W' with status 8
2019-08-26 21:30:35.292 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-08-26 21:30:35.293 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-08-26 21:30:35.303 Debug [2] MainActivity: Bluetooth connection successful disconnected

The difference is only one line
5 days ago
2019-08-21 22:24:22.320 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 5D 46 06 00 00 00 00 00 5C] on 0xffe9

today
2019-08-26 21:29:57.264 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 63 D0 C5 00 00 00 00 00 37] on 0xffe9

oliexdev, 2019-08-27:
please try out the latest [openScale dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and post the log file (please use the code tags in markups!)

codepuregit, 2019-08-29:
oops.

Is this because Yunmai mini 2 is in Chinese Version? ;(

2019-08-29 07:35:58.299 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_7b7e58ae_2019-08-27 (1566914388), SDK 28, Xiaomi POCOPHONE F1
2019-08-29 07:35:58.310 Debug [2] AboutPreferences: Selected user id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)
2019-08-29 07:36:01.113 Debug [2] OpenScale: Trying to connect to bluetooth device [6C:C3:74:D9:51:3F] (YUNMAI-ISM2-W)
2019-08-29 07:36:01.118 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-08-29 07:36:01.126 Debug [2] BluetoothCommunication: Stop machine state
2019-08-29 07:36:01.656 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-ISM2-W'
2019-08-29 07:36:02.663 Debug [2] BluetoothCommunication: Try to connect to BLE device 6C:C3:74:D9:51:3F
2019-08-29 07:36:05.215 Debug [2] BluetoothCommunication: connected to 'YUNMAI-ISM2-W'
2019-08-29 07:36:05.217 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-08-29 07:36:05.218 Debug [2] BluetoothCommunication: Resume machine state
2019-08-29 07:36:05.219 Debug [2] BluetoothCommunication: Step Nr 0
2019-08-29 07:36:05.223 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B] on 0xffe9
2019-08-29 07:36:05.224 Debug [2] BluetoothCommunication: Step Nr 1
2019-08-29 07:36:05.226 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 67 01 D5 00 00 00 00 00 F2] on 0xffe9
2019-08-29 07:36:05.227 Debug [2] BluetoothCommunication: Step Nr 2
2019-08-29 07:36:05.227 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4
2019-08-29 07:36:05.228 Debug [2] BluetoothCommunication: Stop machine state
2019-08-29 07:36:05.233 Debug [2] MainActivity: Bluetooth connection successful established
2019-08-29 07:36:05.284 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-29 07:36:10.308 Error [2] BluetoothCommunication: ERROR: Failed writing <0D 0D 11 5D 67 01 D5 00 00 00 00 00 F2> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-29 07:36:10.313 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-ISM2-W' with status 8
2019-08-29 07:37:02.649 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-08-29 07:37:02.650 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-08-29 07:37:02.658 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2019-08-29:
does your scale work with the vendors app? Maybe you bought a faulty device?

codepuregit, 2019-08-29:
Success Log
```
2019-08-29 21:12:25.152 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.0-dev_7b7e58ae_2019-08-27 (1566914388), SDK 28, Xiaomi POCOPHONE F1
2019-08-29 21:12:25.163 Debug [2] AboutPreferences: Selected user id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)
2019-08-29 21:12:27.491 Debug [2] OpenScale: Trying to connect to bluetooth device [6C:C3:74:D9:51:3F] (YUNMAI-ISM2-W)
2019-08-29 21:12:27.496 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-08-29 21:12:27.507 Debug [2] BluetoothCommunication: Stop machine state
2019-08-29 21:12:27.749 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-ISM2-W'
2019-08-29 21:12:28.774 Debug [2] BluetoothCommunication: Try to connect to BLE device 6C:C3:74:D9:51:3F
2019-08-29 21:12:52.538 Debug [2] BluetoothCommunication: connected to 'YUNMAI-ISM2-W'
2019-08-29 21:12:52.541 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2019-08-29 21:12:52.542 Debug [2] BluetoothCommunication: Resume machine state
2019-08-29 21:12:52.543 Debug [2] BluetoothCommunication: Step Nr 0
2019-08-29 21:12:52.546 Debug [2] BluetoothCommunication: Invoke write bytes [0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B] on 0xffe9
2019-08-29 21:12:52.548 Debug [2] BluetoothCommunication: Step Nr 1
2019-08-29 21:12:52.551 Debug [2] BluetoothCommunication: Invoke write bytes [0D 0D 11 5D 67 C1 44 00 00 00 00 00 A3] on 0xffe9
2019-08-29 21:12:52.552 Debug [2] BluetoothCommunication: Step Nr 2
2019-08-29 21:12:52.553 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe4
2019-08-29 21:12:52.554 Debug [2] BluetoothCommunication: Stop machine state
2019-08-29 21:12:52.577 Debug [2] MainActivity: Bluetooth connection successful established
2019-08-29 21:12:52.627 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 12 10 01 00 00 9E 15 AB 01 24 55 5A 00 00 01 03 0B> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-29 21:12:52.725 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 0D 11 5D 67 C1 44 00 00 00 00 00 A3> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-29 21:12:52.816 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe4-0000-1000-8000-00805f9b34fb
2019-08-29 21:12:52.817 Debug [2] BluetoothCommunication: Resume machine state
2019-08-29 21:12:52.818 Debug [2] BluetoothCommunication: Step Nr 3
2019-08-29 21:12:52.821 Debug [2] BluetoothCommunication: Invoke write bytes [0D 05 13 00 16] on 0xffe9
2019-08-29 21:12:52.822 Debug [2] BluetoothCommunication: Step Nr 4
2019-08-29 21:12:52.823 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2019-08-29 21:12:52.911 Debug [2] BluetoothCommunication: SUCCESS: Writing <0D 05 13 00 16> to <0000ffe9-0000-1000-8000-00805f9b34fb>
2019-08-29 21:12:52.913 Debug [2] BluetoothCommunication: Step Nr 4
2019-08-29 21:12:52.913 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2019-08-29 21:13:02.948 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 5D 67 C1 4E 00 00 9E 15 1C BD 01 A1 08 C0 FE]
2019-08-29 21:13:02.951 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 73.57, fat: 22.40, resistance: 417]
2019-08-29 21:13:02.955 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)]
2019-08-29 21:13:02.961 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Thu Aug 29 21:13:02 GMT+09:00 2019, WEIGHT: 73.57, FAT: 22.40, WATER: 56.34, MUSCLE: 54.33, LBM: 0.00, WAIST: 0.00, HIP: 0.00, BONE: 3.07, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: ]
2019-08-29 21:13:02.967 Debug [2] OpenScale: assign measurement to the nearest measurement with the user もか (smartUserAssignment=on)
2019-08-29 21:13:02.972 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Thu Aug 29 21:13:02 GMT+09:00 2019, WEIGHT: 73.57, FAT: 23.53, WATER: 40.16, MUSCLE: 54.33, LBM: 56.41, WAIST: 0.00, HIP: 0.00, BONE: 3.07, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: 
2019-08-29 21:13:03.738 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 5D 67 C1 4E 00 00 9E 15 1C BD 01 A1 08 C0 FE]
2019-08-29 21:13:03.740 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 73.57, fat: 22.40, resistance: 417]
2019-08-29 21:13:03.741 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(もか) birthday(Thu Jan 06 00:00:00 GMT+09:00 1983) age(36) body height(171.00) scale unit(kg) gender(male) initial weight(71.00) goal weight(69.00) goal date(Sat Jan 11 06:56:54 GMT+09:00 2020) measure unt(cm) activity level(0)]
2019-08-29 21:13:03.743 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Thu Aug 29 21:13:02 GMT+09:00 2019, WEIGHT: 73.57, FAT: 22.40, WATER: 56.34, MUSCLE: 54.33, LBM: 0.00, WAIST: 0.00, HIP: 0.00, BONE: 3.07, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: ]
[... cut ...]
2019-08-29 21:15:50.281 Debug [2] BluetoothCommunication: Found peripheral 'YUNMAI-ISM2-W'
2019-08-29 21:15:51.302 Debug [2] BluetoothCommunication: Try to connect to BLE device 6C:C3:74:D9:51:3F
2019-08-29 21:16:26.488 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-ISM2-W' with status 8
2019-08-29 21:16:34.650 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-08-29 21:16:34.653 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-08-29 21:16:34.680 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2019-08-29:
so its looks more like a device problem or timing problem!?

Btw if you are Japanese it would be great if you could help us translate the app into Japanese, see https://hosted.weblate.org/projects/openscale/strings/ja/

closed 2019-08-29 by oliexdev