oliexdev/openScale #493: Trend line should use the last measurements instead the last days

naoliv, 2019-09-11:
**Is your feature request related to a problem? Please describe.**
It seems that the trend lines are using the last `N` (I am using `N` because I am unsure if you are using `7`) days to be calculated, instead using the last `N` measurements.

**Describe the solution you'd like**
Trend lines should be calculated with the last `N` weight/data measurements, no matter if we skipped to measure them in some days.

**Additional context**
For example, today was the first day of the month where I did a measurement and inserted it in openScale.
It's possible to see that there are only flat trend lines, even when artificially inserting my weigth with +20kg than my last measurement from the last month:

[image]

Using the last `N` measurements should give us a trend line even if we measure only once per month, for example.

ati3, 2019-12-17:
This seems like a good solution, but what if you skip 6 months of measurements? I suggest to calculate the trendlines from the last month or 2 with increasing statistical weight. https://github.com/oliexdev/openScale/issues/436 should fix this.

oliexdev, 2020-12-02:
should be implemented now.

closed 2020-12-02 by oliexdev