oliexdev/openScale #495: Question: Support of YUNMAI SE

white-gecko, 2019-09-11:
As it seems in https://github.com/oliexdev/openScale/issues/284 the YUNMAI SE is supported while in the wiki (https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale) it is listed as:

Scale | Initialisation process | History data | Body metrics | Remarks / Known Issues
-- | -- | -- | -- | --
Yunmai SE | o | n/a | n/a | Weight unit is currently not set via openScale

Does this mean it is supported in general but just the initialisation process has to be performed with the App from Yunmai?

If so, what are the problems with the initialisation process?

white-gecko, 2019-09-13:
Yes, it works like magic.

Just unbox and connect the scale via the openScale setting using bluetooth and everything is set.

Setting the unit works as well: If I set kg for the user the scale switches to kg, if I set lb for the user the scale switches to lb, and if I set st for the user the scale switches to lb but the app seems to do the calculation.

I think 100% of the features that the scale offers are covered by openScale.

oliexdev, 2019-09-13:
thx for testing I corrected the wiki entry accordingly.

closed 2019-09-13 by oliexdev