oliexdev/openScale #497: Unable to sync to google

lbr88, 2019-09-18:
**Describe the bug**
Unable to use openscale sync. Throws an error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
[image]

Reproduced with latest openscale sync version on play store 

**Expected behavior**
Expected to sync

**Additional context**
I figure it crashed because there is some error in my data 

**Debug log**
Build version: 0.2 
Current date: 2019-09-18 06:32:42 
Device: OnePlus ONEPLUS A5010 
 
Stack trace:  
java.lang.IllegalArgumentException: Data point does not have the timestamp set: DataPoint{[142.65]@[0, 0,raw=0,insert=0](r:weight:com.health.openscale.sync: N/A)}
	at d.a.a.a.b.b.s.a(:9)
	at com.google.android.gms.fitness.data.DataSet.a(:16)
	at d.b.a.a.a.b.k.a(:24)
	at d.b.a.a.b.h.onClick(:23)
	at android.view.View.performClick(View.java:6669)
	at android.view.View.performClickInternal(View.java:6638)
	at android.view.View.access$3100(View.java:789)
	at android.view.View$PerformClick.run(View.java:26145)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6898)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

lbr88, 2019-09-18:
I deleted an entry in the openscale pro app that had the date of January1. 1970 with the weight from the bugreport and it now works. But maybe it would still be worth fixing so it doesn't crash on people after they have paid for the app.

oliexdev, 2019-09-18:
thanks for the report. Would be great if you post the openScale sync log file to see more into the details (You can enable it on the upper right three dots -> enable log)

closed 2019-09-18 by oliexdev

lbr88, 2019-09-19:
2019-09-18 06:34:13.521 Debug [2] MainActivity: Debug log enabled, openScale sync v0.2 (2), SDK 28, OnePlus ONEPLUS A5010
2019-09-18 06:34:15.471 Debug [2] MainActivity: GoogleFit successful sign in
2019-09-18 06:34:15.482 Debug [2] MainActivity: openScale version 45 with content provider API version 1
2019-09-18 06:34:15.484 Debug [2] MainActivity: openScale version is ok(com.health.openscale.pro)
2019-09-18 06:34:15.493 Debug [2] MainActivity: openScale user found(Lars Bo Rasmussen)
2019-09-18 06:34:15.494 Debug [2] View: Manual sync openScale measurements -> GoogleFit

Measurements sync lines removed for privacy

Last line:
2019-09-18 06:34:15.547 Debug [2] View: openScale measurement date Thu Jan 01 01:00:00 GMT+01:00 1970 weight 142.65 added to GoogleFit