oliexdev/openScale #498: MQTT Crashes app

DENightOne, 2019-09-30:
Updated to latest build this morning, I enabled MQTT on Samsung A7 2018 (Android Pie) phone and as soon as I clicked on the username field and press backspace a few times the app crashes... When restarting the app crashes immediatley . I have to uninstall the app completely and reinstall the app to get it to work again.

Have done it 4 times now and it always crashes when I take away the username.

Error Log:
 
Build version: 0.3 
Current date: 2019-09-30 12:40:22 
Device: Samsung SM-A750FN 
 
Stack trace:  
java.lang.IllegalArgumentException
at d.c.a.a.a.c.d.d(:3)
at d.c.a.a.a.c.d.a(:2)
at d.c.a.a.b.b.b$c.afterTextChanged(:5)
at android.widget.TextView.sendAfterTextChanged(TextView.java:10990)
at android.widget.TextView$ChangeWatcher.afterTextChanged(TextView.java:14162)
at android.text.SpannableStringBuilder.sendAfterTextChanged(SpannableStringBuilder.java:1273)
at android.text.SpannableStringBuilder.replace(SpannableStringBuilder.java:576)
at android.text.SpannableStringBuilder.replace(SpannableStringBuilder.java:506)
at android.text.SpannableStringBuilder.replace(SpannableStringBuilder.java:36)
at android.view.inputmethod.BaseInputConnection.replaceText(BaseInputConnection.java:849)
at android.view.inputmethod.BaseInputConnection.setComposingText(BaseInputConnection.java:618)
at com.android.internal.view.IInputConnectionWrapper.executeMessage(IInputConnectionWrapper.java:396)
at com.android.internal.view.IInputConnectionWrapper$MyHandler.handleMessage(IInputConnectionWrapper.java:85)
at android.os.Handler.dispatchMessage(Handler.java:106)
at android.os.Looper.loop(Looper.java:214)
at android.app.ActivityThread.main(ActivityThread.java:7078)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

oliexdev, 2019-09-30:
thanks, I didn't test that if the login credentials are empty. Should be fixed in the openScale sync version 0.3.1, which I just released a few seconds ago.

closed 2019-09-30 by oliexdev

DENightOne, 2019-09-30:
MQTT does not allow no username and password (field required)....  I have two MQTT servers....one dev one with anonymous access for testing, and then a live one that has authentication....works fine on live with username and password.

[Image]

Closed #498<https://github.com/oliexdev/openScale/issues/498>.