oliexdev/openScale #501: Auto daily backup is not complete

FifiTheReal, 2019-10-05:
**Describe the bug**
The automatic daily backup is only completed when I enter the Settings/Backup menu.

**To Reproduce**
[See my screen capture.](https://www.dropbox.com/s/qd7jlnqzn4rg6wh/2019-10-05%2008.28.03.mp4)

oliexdev, 2019-10-12:
The backup time is inexact scheduled by Android, see https://developer.android.com/training/scheduling/alarms for more information. openScale will save the time and date based on your settings when you exit the Settings/Backup menu, so the alarm manager might be triggered with the new settings but it should automatically triggered every day/week/month and so on.

FifiTheReal, 2019-10-16:
Thanks for the reply!

I think the problem was that the phone didn't let the program run in the background. I enabled running openscale in the background and now everything is fine:
[image]

closed 2019-10-16 by FifiTheReal