oliexdev/openScale #502: Support for Beurer BF 950

PiQuer, 2019-10-05:
Hi!
Thanks for developing this great app! I recently bought a BF 950 and would like to help getting it supported by OpenScale.

[btsnoop_hci_2019-10-05_01.log](https://github.com/oliexdev/openScale/files/3693111/btsnoop_hci_2019-10-05_01.log)
date: 2019-10-05
time: 09:21
sex: male
weight: 72.5 kg
body fat: 13.2%
water: 61.7%
muscles: 44.6%
bones: 3.2 kg

[btsnoop_hci_2019-10-05_02.log](https://github.com/oliexdev/openScale/files/3693113/btsnoop_hci_2019-10-05_02.log)
date: 2019-10-05
time: 11:58
sex: male
weight: 74.7 kg
body fat: 13.0%
water: 60.8%
muscles: 44.1%
bones: 3.3 kg

[btsnoop_hci_2019-10-03.log](https://github.com/oliexdev/openScale/files/3693114/btsnoop_hci_2019-10-03.log)
date: 2019-10-05
time: 12:10
sex: male
weight: 75.8 kg
body fat: 14.9%
water: 60.3%
muscles: 43.8%
bones: 3.3 kg

[btsnoop_hci_2019-10-05_04.log](https://github.com/oliexdev/openScale/files/3693118/btsnoop_hci_2019-10-05_04.log)
date: 2019-10-05
time: 12:15
sex: male
weight: 77.0 kg
body fat: 14.9%
water: 59.7%
muscles: 43.5%
bones: 3.3 kg

Debug log: 
[openScale_2019-10-05_11-53.txt](https://github.com/oliexdev/openScale/files/3693128/openScale_2019-10-05_11-53.txt)

For the first three logs I could not find the corresponding weight values in the data stream, neither in big endian nor little endian format. The last value 77.0 kg is a bit unfortunate: it is 03 02 or 02 03 in hex, there are plenty occurrences, but I think those have to do with the bluetooth protocol rather then the weight value.

Please let me know if I can help by doing more measurements.

oliexdev, 2019-10-12:
If I have time I will have a look. Maybe it is the same like BF600, see issue #219

PiQuer, 2019-10-13:
Thanks, I will read through that other issue.

PiQuer, 2019-10-13:
I installed the development environment and successfully built the master branch, reading for testing. Will test the `BluetoothStandardWeightProfile` with the BF950 and report back.

PiQuer, 2019-10-14:
Debug log with `BluetoothStandardWeightProfile`:

[openScale_2019-10-14_02-12.txt](https://github.com/oliexdev/openScale/files/3722661/openScale_2019-10-14_02-12.txt)

PiQuer, 2019-11-03:
@oliexdev to me the log looks like with this modification there is already a lot of communication going on between the scale and the app, and now some kind of authentication is needed. Anything I can do to bring this issue forward? Thanks!

dror3go, 2019-11-11:
I also own a Beurer BF 950 scale and am available to help with solving the issue.

felixandersen, 2020-04-24:
Looking at buying this scale. Any progress on this issue? Would be willing to get one to provide info for adding it to the app.

oliexdev, 2020-04-28:
@felixandersen not really but the scale use the Bluetooth standard protocol which is good. Unfortunately, somebody has to make some effort to implement the authorization pairing process. I don't own this scale so it is to troublesome to do it on my own.

felixandersen, 2020-04-28:
@oliexdev Thanks for getting back on the question! I would be willing to put in some hours to get this working. I have limited experience with Java and Android, but somewhat broad development experience in general. What would I need apart from the scale? An android phone or tablet I guess?

oliexdev, 2020-04-28:
Yes an Android phone or tablet is needed and I would recommend that you have some some basic reverse engineering skills, see https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale
The basic protocol is implemented in BluetoothStandardWeightProfile.java but you have to do at least two things:

1. You need to send the consent code for a user (it supports up to 8 users). Currently it is hardcoded. You can find this code by pressing the buttons on the scale itself. So we need to ask for that 4 digit code when the user tries to pair the scale with openScale. Then we need to send that code instead of the hardcoded PIN.

2. If we haven't bonded the scale yet, we need to call createBond directly after the service discovery is complete.

It is possible to support this scale. Mainly because it uses the standard Bluetooth protocol, but without the scale it is a bit difficult.

felixandersen, 2020-04-28:
Great, thanks! I will order the scale and get on it soon.

oliexdev, 2020-04-28:
But I can't guarantee it if that works, so if you buying this scale it is on your own risk :wink:

felixandersen, 2020-04-28:
For sure! I expect no support or promises. Just happy if I can contribute.

PiQuer, 2020-05-01:
I'm also still around to test or provide data. Currently I'm importing the values manually into openScale, which is a bit cumbersome...

oliexdev, 2020-06-16:
@felixandersen any progress?

dror3go, 2020-09-27:
@felixandersen Have you looked into it? What did you find?

bcdiaconu, 2020-12-08:
Are there any news for this scale?

dror3go, 2020-12-08:
@bcdiaconu See also [issue 295](https://github.com/oliexdev/openScale/issues/295), as this scale follows the Bluetooth standard protocol.

krisjans, 2021-07-26:
I made support for BF600 and BF850. And I think that BF950 could also work with the same code.

If anyone interested to test if BF960 works like BF600 - this is updated build of #753: [openScale-2.3.5-debug_bf600_bf850_bf950.zip](https://github.com/oliexdev/openScale/files/6877239/openScale-2.3.5-debug_bf600_bf850_bf950.zip)

dror3go, 2021-07-27:
I tested it with my BF 950.
I was able to connect to the scale, and noticed that the time on the scale got updated - which is great.
However, there was no popup with existing scale users once I created a new app user.

Is there anything else you'd like me to try out? Should I try to factory reset the scale?

krisjans, 2021-07-27:
@dror3go Thanks for testing my version!
Could you try again and provide openScale debug log and btsnoop_hci.log ?

Another thing that may help would be btsnoop_hci.log that shows how manufacturer's app creates new users on scale.

krisjans, 2021-07-27:
> However, there was no popup with existing scale users once I created a new app user.

@dror3go 
If there is no users on scale then openScale will atempt to create new user automatically. If scale user was created successfully then  then there should short status message that suggests that you step on scale barefoot.
Another idea to try - remove BF950 from paired devices in your phones Bluetooth settings before connecting to scale with openScale. Especially after factory reset.

krisjans, 2021-07-28:
If someone is interested in testing my latest changes related to [BF600](https://github.com/oliexdev/openScale/issues/219), [BF850](https://github.com/oliexdev/openScale/issues/531), [BF950](https://github.com/oliexdev/openScale/issues/502) and [SBF77](https://github.com/oliexdev/openScale/issues/681), then latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk), change log in #753.

If something does not work or you encounter a crash - I would highly appreciate if you leave me a comment with attached openScale debug log (if possible also  btsnoop_hci.log. [Instructions how to get logs.](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic))

krisjans, 2021-07-29:
Hello, @PiQuer, @felixandersen, @bcdiaconu !
Any Do you still have BF950? I would be extremely happy if you could test my latest debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk) with enabled openScale debug [log](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#how-to-get-the-openscale-debug-log-file).
According to @dror3go [something is working](https://github.com/oliexdev/openScale/issues/502#issuecomment-887218200), but no scale user selection dialog or pin entry dialog pops up. 
Could you test and provide debug logs? I would be happy to fix support for BF950.

dror3go, 2021-07-29:
@krisjans I've re-installed the APK after uninstalling it and unpairing the device from the Bluetooth settings. Still same behaviour.
Just to clarify: there are existing users on the scale, with some weights for each of the users.
See log attached.
[openScale_2021-07-29_11-53.txt](https://github.com/oliexdev/openScale/files/6899358/openScale_2021-07-29_11-53.txt)

krisjans, 2021-07-29:
@dror3go Thanks for debug log!
I fixed the problem and released new debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk). Would you like to test it and let me know if it works now?

Existing code was expecting that all scales have `Battery Service` and when BF950 responded with error on battery service request BluetoothStandardWeightProfile got stuck. I changed the code - now `battery service` is optional.
I also analyzed btsnoop_hci.log files (provided by @PiQuer) and discovered that BF950 is more like SBF77 (and not BF600 as I initially thought). I fixed that also. Can't wait when someone will try this out!

dror3go, 2021-07-29:
OK, we're making progress 🙂 
I think I was able to make an app-scale user connection - I entered the pin code, but there's no feedback on the app indicating that it worked.
Now I'm not sure what to expect: should I scale myself and then see the info in the app? Is there a way to sync old data to the app?

Note that the app crashes when entering a wrong pin code several times in a row:

```
 Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{9e69fbd} (e2c8b194-c8ea-4fcb-85ea-b94f4da4cbc3)} not attached to a context.
 at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
 at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
 at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
 at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
 at android.os.Handler.handleCallback(Handler.java:938)
 at android.os.Handler.dispatchMessage(Handler.java:99)
 at android.os.Looper.loop(Looper.java:233)
 at android.app.ActivityThread.main(ActivityThread.java:8010)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:631)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:978)
```

krisjans, 2021-07-29:
@dror3go Thanks for the feedback!

> I think I was able to make an app-scale user connection - I entered the pin code, but there's no feedback on the app indicating that it worked.

Yes, there is no further feedback if you selected existing user and entered correct pin code.
> Now I'm not sure what to expect: should I scale myself and then see the info in the app? Is there a way to sync old data to the app?

The scale delete old measurements as soon as the are transmitted to any app. If you used manufacturers app before openScale then there will be no measurements left in scale's memory. To get some measurements in app you can weight yourself before you connect app to the scale (first app that connects will get the measurements). Please note that  only measurements for selected user will be transmitted from scale to app. Be sure that BF950 recognizes and indicate correct user after weighting.

> Note that the app crashes when entering a wrong pin code several times in a row:

Thanks for spotting this! I newer tried to enter wrong pin code several times. Will investigate this further.

krisjans, 2021-07-29:
@dror3go May I ask you another favor? Could you create two new users with different activity levels in manufacturers app and collect btsnoop_hci.log? Bluetooth standard does not describe how to set or store "activity level" - and each scale seems to be doing it in it's own way.
If you could do this, then I will be able to finalize user creation from openScale (for BF950).

dror3go, 2021-07-30:
I'd love to help, but I'm not using the manufacturer's app. Is there anything else I can help with?

krisjans, 2021-07-31:
> I'd love to help, but I'm not using the manufacturer's app. Is there anything else I can help with?

@dror3go 
- Did you manage to get actual measurements from your BF950 to openScale?
- Could you provide openScale debug log (when measurements got transferred from scale to app)?

bcdiaconu, 2021-08-03:
@krisjans I have installed your patch. It worked flawlessly to connect to scale. I am using it for a few days now and did not receive any error. 

I will search for log(s) and upload it here

krisjans, 2021-08-03:
@bcdiaconu Thanks for the feedback!

May I ask you your help? Could you please create two new users with different activity levels in manufacturers app and collect btsnoop_hci.log? Bluetooth standard does not describe how to set or store "activity level" - and each scale seems to be doing it in it's own way.
If you could get me btsnoop_hci.log from BF950 (after manufacturer's app creates new users on scale), then I might be able to figure out UUID that is used to set "activity level" for scale user and finalize BF950 support.
Thanks in advance!

bcdiaconu, 2021-08-03:
@krisjans I am at work now but I will do that and I will get back to you soon

bcdiaconu, 2021-08-03:
@krisjans So, it is really hard to work with original app (at least on my phone) but I managed for the same user to change the activity level. I attached all the logs for that. In `scale_data_transfer.log`, you can find line 2313 where I started cycling through activity levels.

[scale_data_transfer.log](https://github.com/oliexdev/openScale/files/6925751/scale_data_transfer.log)
[data_transfer.log](https://github.com/oliexdev/openScale/files/6925752/data_transfer.log)
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6925775/btsnoop_hci.log)

To change the time stamps in my Wireshark capture files to match the time stamps in the log files:
Wireshark -> View <menu> -> Select Time Display Format -> Date and Time of Day 
or use shortcut `ctrl + alt + 1`

If necessary still, I will try creating new users with just different activity. Please let me know.

krisjans, 2021-08-03:
@bcdiaconu I'm looking at your [btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6925775/btsnoop_hci.log) and there is no packets sent from phone to scale. If I set filter `bluetooth.dst == 5c:ca:d3:00:92:4b` then there is nothing.
In scale_data_transfer.log I see the data that is being sent, but I don's see target UUID (or handle).
Maybe in your phone debug settings you set `Enable Bluetooth HCI snoop log` to `Enable filered`?  If so, then set it to `Enabled` (don't forget to disable and re-enable Bluetooth after that) and try capturing again.

bcdiaconu, 2021-08-07:
@krisjans I had a few full days. I have added two users with same name (X1) and all defaults w/ only changed activity level.
and than second time I created the user, I also cycled through the activity levels. Hopefully the frames are now captured and can be found in log:

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6948665/btsnoop_hci.log)
[data_transfer.log](https://github.com/oliexdev/openScale/files/6948666/data_transfer.log)
[scale_data_transfer.log](https://github.com/oliexdev/openScale/files/6948667/scale_data_transfer.log)
[errors.log](https://github.com/oliexdev/openScale/files/6948668/errors.log)

krisjans, 2021-08-07:
@bcdiaconu Thanks for the logs!
Unfortunately, also this time btsnoop_hci.log does not contain packets sent by phone to BF950. For some reason your phone does not log packets transmitted by phone into btsnoop_hci.log.
Maybe you have some older android phone or tablet (with older Android version)? Older Android version had less restrictions regarding Bluetooth traffic logging. I you have such older device, could you try again? Make it even simpler - just create one user from manufacturers app. That should be enough for me - all I need is to pinpoint UUID used to set "activity level".
Thanks again for your effort and help!

bcdiaconu, 2021-08-07:
@krisjans I have done another try from another newer phone. Took a look at the contents and it seems there are frames both directions between the two. Hopefully this time you find what you're looking for.

[image]

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6948904/btsnoop_hci.log)
[scale_data_transfer.log](https://github.com/oliexdev/openScale/files/6948905/scale_data_transfer.log)
[data_transfer.log](https://github.com/oliexdev/openScale/files/6948906/data_transfer.log)

krisjans, 2021-08-07:
@bcdiaconu YES! Thank you very much! This time btsnoop_hci.log contains packets in both directions.
So far I figured out following extra UUIDS:
- User activity level: handle 0x0054, service UUID 0xffff, UUID 0x0004; One byte, values from 1-5;
- User initials: handle 0x004e, service UUID 0xffff, UUID 0x0002; Three bytes, ASCII;
- Request scale measurement: 0x005d, service UUID 0xffff, UUID 0x0006; One byte, value 0 == start new measurement;

bcdiaconu, 2021-08-07:
@krisjans great. I am glad it worked in the end

krisjans, 2021-08-07:
@bcdiaconu I think I fixed "activity level" and "request measurement" for BF950.
Could you grab latest [build](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk) and test if you can create new user on BF950 from openScale app?

Detailed testing script:
1. Create new user in openScale app;
2. Select the new user as active user; 
2. Connect to BF950;
3. When `Select scale user` dialog pops up then pick `Create new user on scale` option (at the bottom of the list));
4. Notification message with text "Please step barefoot on the scale for reference measurements" should be displayed for short period of time;
5. Scale display should light up. Step barefoot on scale to complete reference measurement;
6. One measurement should appear in openScale;
7. Disconnect Bluetooth;
8. Using BF950 LCD and buttons check that newly created user has correct data (birth date, activity level, gender and height);

If possible, collect openScale debug log. Any comments welcome!
Thanks in advance!

bcdiaconu, 2021-08-09:
@krisjans 
Creation of user with exact steps you posted are working fine but there are some observations:
* If scale is already lid up, creation of a new user is impossible; the user is created on scale but when stepping on the scale (as app asks), the measurement will be added to existing user rather the new created one
* New user initials (name) are not set to the scale
* Deletion of a user inside app will not delete the user from scale

I will upload also the logs in later edit

krisjans, 2021-08-10:
@bcdiaconu Thanks for your feedback!
>New user initials (name) are not set to the scale

Correct, user initial setting is not yet implemented;
>Deletion of a user inside app will not delete the user from scale

This is difficult to decide. Do I really wanted to delete user from scale? Maybe in scale user selection dialog I accidentally selected a scale user that someone else is using?
Another aspect - what about situation when app user is being deleted when scale is not in Bluetooth range?
If you have any ideas about this, please share!

bcdiaconu, 2021-08-10:
@krisjans the first case is as simple as a pop-up asking if one wants to delete the user also from scale

Second could be similar where the user is promoted that Bluetooth device ID out of range and it can only be deleted from app if continued. Could be better options than that. So, if someone else has a better idea we are 2 to welcome

krisjans, 2021-08-24:
Anyone willing to help out and test latest changes for #753 on BF950?
Latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk).
I'm especially interested if user initials is now stored and displayed on SBF77 when you create new user from openScale.

p.s. @bcdiaconu maybe you would like to help me once more and test this build and provide your comments and debug log?

bcdiaconu, 2021-08-25:
@krisjans 

Now it throws this exception at convention with scale. I will attach logs when connecting to computer

```log
Build version: 2.3.5 Build date: 1981-01-01 01:01:02 Current date: 2021-08-25 07:27:10 Device: HMD Global Nokia 7.2 OS version: Android 10 (SDK 29)  Stack trace:  java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{fcbeb23} (db70de8b-2211-4a2c-93d3-a97a5d4e3baf)} not attached to a context.	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)	at android.os.Handler.handleCallback(Handler.java:883)	at android.os.Handler.dispatchMessage(Handler.java:100)	at android.os.Looper.loop(Looper.java:214)	at android.app.ActivityThread.main(ActivityThread.java:7397)	at java.lang.reflect.Method.invoke(Native Method)	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

bcdiaconu, 2021-08-25:
@krisjans 

Exception 2 and 3 (same stack trace with different logs):

[openScale_2021-08-25_07-43.txt](https://github.com/oliexdev/openScale/files/7043796/openScale_2021-08-25_07-43.txt)
[openScale_2021-08-25_07-44.txt](https://github.com/oliexdev/openScale/files/7043807/openScale_2021-08-25_07-44.txt)

```log
Build version: 2.3.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-08-25 07:43:55 
Device: HMD Global Nokia 7.2 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalArgumentException: could not get CCC descriptor for characteristic 00002a19-0000-1000-8000-00805f9b34fb
	at com.welie.blessed.BluetoothPeripheral.setNotify(BluetoothPeripheral.java:1390)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.setNotificationOn(BluetoothCommunication.java:329)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep(BluetoothStandardWeightProfile.java:151)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:576)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:204)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onNotificationStateUpdate(BluetoothCommunication.java:439)
	at com.welie.blessed.BluetoothPeripheral$1$2.run(BluetoothPeripheral.java:238)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7397)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

krisjans, 2021-08-25:
@bcdiaconu Thanks for your help and feedback! Much appreciated!  

> Now it throws this exception at convention with scale.
> 
> Later edit:
> [openScale_2021-08-25_07-25.txt](https://github.com/oliexdev/openScale/files/7043785/openScale_2021-08-25_07-25.txt)
> 
> ```
> Build version: 2.3.5 
> Build date: 1981-01-01 01:01:02 
> Current date: 2021-08-25 07:27:10 
> Device: HMD Global Nokia 7.2 
> OS version: Android 10 (SDK 29) 
>  
> Stack trace:  
> java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{fcbeb23} (db70de8b-2211-4a2c-93d3-a97a5d4e3baf)} not attached to a context.
> 	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
> 	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
> 	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
> 	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
> 	at android.os.Handler.handleCallback(Handler.java:883)
> 	at android.os.Handler.dispatchMessage(Handler.java:100)
> 	at android.os.Looper.loop(Looper.java:214)
> 	at android.app.ActivityThread.main(ActivityThread.java:7397)
> 	at java.lang.reflect.Method.invoke(Native Method)
> 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
> 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
> ```

I'm unable to repeat this crash. Could you describe in more detail what you do when it happens? Sequence of actions?

krisjans, 2021-08-25:
> Exceptions 2 and 3 (same stack trace with different logs but different from exception 1):
> 
> [openScale_2021-08-25_07-43.txt](https://github.com/oliexdev/openScale/files/7043796/openScale_2021-08-25_07-43.txt)
> [openScale_2021-08-25_07-44.txt](https://github.com/oliexdev/openScale/files/7043807/openScale_2021-08-25_07-44.txt)
> 
> ```
> Build version: 2.3.5 
> Build date: 1981-01-01 01:01:02 
> Current date: 2021-08-25 07:43:55 
> Device: HMD Global Nokia 7.2 
> OS version: Android 10 (SDK 29) 
>  
> Stack trace:  
> java.lang.IllegalArgumentException: could not get CCC descriptor for characteristic 00002a19-0000-1000-8000-00805f9b34fb
> 	at com.welie.blessed.BluetoothPeripheral.setNotify(BluetoothPeripheral.java:1390)
> ```
@bcdiaconu Thanks for your help!
This should be fixed now (1f1813eef6793fd4be18237003646d3c992f2451).  Could you grab latest [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk) and try again?
This was probably caused by changes in Blessed library. Previously checking that getCharacteristic() returns non-null value was enough. Now it returns non-null object even if characteristic does not exist.

bcdiaconu, 2021-08-25:
@krisjans 

Not really sure how to reproduce but I've seen in log (for first exception) that there are a few `-1,00` on second line. If this is the case there were no `-1` introduced by myself. I have however added some abnormal height, weight (don't remember the numbers possibly something like 111 cm and 111kg)

Steps I remember I've done:
1. Clean install
2. Skip tour
3. Skip creation of a user (press back button)
4. Tried enabling Bluetooth (but a no user message appeared)
5. Created user
6. Added Bluetooth device
7. Enabled Bluetooth

Hopefully this helps

krisjans, 2021-08-25:
> [openScale_2021-08-25_07-25.txt](https://github.com/oliexdev/openScale/files/7043785/openScale_2021-08-25_07-25.txt)
>
> ```
> Build version: 2.3.5
> Build date: 1981-01-01 01:01:02
> Current date: [...]

@bcdiaconu This seems to be the same crash described in https://github.com/oliexdev/openScale/issues/761#issue-955104161 and is unrelated to specific scale (BF950 in this case) support.

bcdiaconu, 2021-08-25:
@krisjans looks like it is the same bug

I found an improvement/bug with latest patch. If the name starts with a space (possibly also the same applies if it's at the end), when creating the user in scale or will throw an exception. So a simple solution would be to trim the (name) string for whitespace(s).

```
Build version: 2.3.5 Build date: 1981-01-01 01:01:02 Current date: 2021-08-25 18:02:37 Device: HMD Global Nokia 7.2 OS version: Android 10 (SDK 29)  Stack trace:  java.lang.StringIndexOutOfBoundsException: length=0; index=0	at java.lang.String.charAt(Native Method)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.buildInitialsStringFrom(BluetoothStandardWeightProfile.java:844)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.getInitials(BluetoothStandardWeightProfile.java:830)	at com.health.openscale.core.bluetooth.BluetoothSwpSBF77.writeInitials(BluetoothSwpSBF77.java:66)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.writeUserDataToScale(BluetoothStandardWeightProfile.java:214)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep(BluetoothStandardWeightProfile.java:193)	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:583)	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:204)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.handleUserControlPointNotify(BluetoothStandardWeightProfile.java:277)	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onBluetoothNotify(BluetoothStandardWeightProfile.java:248)	at com.health.openscale.core.bluetooth.BluetoothSwpSBF77.onBluetoothNotify(BluetoothSwpSBF77.java:109)	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:467)	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:276)	at android.os.Handler.handleCallback(Handler.java:883)	at android.os.Handler.dispatchMessage(Handler.java:100)	at android.os.Looper.loop(Looper.java:214)	at android.app.ActivityThread.main(ActivityThread.java:7397)	at java.lang.reflect.Method.invoke(Native Method)	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

krisjans, 2021-08-25:
@bcdiaconu Nice catch!
This should be fixed now (9a0d3fdb95164b2b745af3d0fc3ab6397c9eded2). Could you grab latest apk and try again?

The problem was specifically if first character of name was space - in this scenario split() reurned zero lengt string. Oh. but two spaces between name and family name will create the same problem..

krisjans, 2021-08-25:
@bcdiaconu now it should be fixed for real now (9012419a0bc8fa2519b4688b376d0063995a78d3) :D. My latest debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk).

krisjans, 2021-09-10:
I just made a [refactoring](https://github.com/oliexdev/openScale/pull/753/commits/3142344865a7ee7adb68d65a527776b3d98890c2) that might affect SBG77. 
I would highly appreciate if anyone could test my latest debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk) and let me know if it still works on SBF77.

bcdiaconu, 2021-09-23:
@krisjans i have installed your new version a few days ago and I did not get the exception any longer. Only some problems with connection from time to time; possibly 20% connection error rate. The behaviour is that the scale LCD does not lid the Bluetooth icon. And the app does display connecting but does not state connected message.

oliexdev, 2021-10-10:
see #753

closed 2021-10-10 by oliexdev