oliexdev/openScale #503: No connection to Sanitas SBF70 - Failed writing <E6 01>

Guzi1985, 2019-10-06:
I used the free version of Openscale - the connection to the scale after many attempts worked, after switching to the pro version I can't connect to the scale. I tried to delete users in the scale and add the user using the official scale (such a username). Batteries in scale changed, weight and phone reset.
The phone displays the connection message, the Bluetootch icon appears on the scale for a moment, no data is being downloaded. The error message in the log: ERROR: Failed writing <E6 01> to sanitas

Weight: Sanitas SBF70
Application: v2.1.1-pro (46)
Phone: Xiaomi Mi MIX 2S 
Android 9

Error log: (same error every time):
[openScale_2019-10-06_11-58.txt](https://github.com/oliexdev/openScale/files/3694260/openScale_2019-10-06_11-58.txt)

oliexdev, 2019-10-12:
Do you know how to use logcat?

Guzi1985, 2019-10-14:
Log sent to the mail.

Guzi1985, 2019-10-16:
I uninstalled the sanitas health coach and connected me the first time :) Thank you.

oliexdev, 2019-10-20:
should be fixed in the openScale version 2.1.2

closed 2019-10-20 by oliexdev

gnaghi, 2020-12-28:
I have reset my scale, removed Healthcoach app, erased data for openscale, and started the init procedure.
When connecting the scale for the first time to OpenScale, I have the same error :

2020-12-28 10:52:58.491 Debug [2] MainActivity: Bluetooth connection successful established`
2020-12-28 10:52:58.522 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2020-12-28 10:52:58.523 Debug [2] BluetoothCommunication: Resume machine state
2020-12-28 10:52:58.523 Debug [2] BluetoothCommunication: Step Nr 1
2020-12-28 10:52:58.523 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2020-12-28 10:52:58.524 Debug [2] BluetoothCommunication: Stop machine state
2020-12-28 10:52:58.525 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-12-28 10:53:03.581 Error [4308] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'GATT_ERROR'
2020-12-28 10:53:03.583 Error [2] BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-12-28 10:53:03.585 Info [4308] BluetoothPeripheral: peripheral 'SANITAS SBF70' disconnected with status 'CONNECTION_TIMEOUT' (8)
2020-12-28 10:53:03.626 Debug [2] BluetoothCommunication: disconnected 'SANITAS SBF70' with status 8

I’m using openScale v2.1.1-pro (46).