oliexdev/openScale #504: Porting to ESP32, cannot get any weight data - confused.

neocogent, 2019-10-07:
Hello,
I've created my own hacked scale using an ESP32, HX711, 1.8" TFT display. It's all working and I'm trying to port over simple code from your MCU/HM-10 device using Custom openScale connection service/characteristic.

I have got simple test code running and it advertises service/characteristic that openScale app sees, connects to, and says "connection established". That seems fine. I've tested my code with "nRF Connect" to debug and the various command chars 0,1,2,3,9 give valid data back. So I think that part is working.

What I can't understand is the openScale app never sends any commands - it never requests scale data, device info, etc. How and/or when does that happen? It doesn't seem to be triggered when you press "+" for adding new weight measurement. It isn't happening upon connection or opening app. I cannot get app to send command ever. 

Even though connection is established it seems to never do anything after that.

The ESP32 has BLE/Wifi built in so it's not using AT commands via serial to HM-10. I think I have this working properly as tested with debug "nRF Connect" app. I don't see any app code for a secondary service/characteristic so I assume just the single one used in BluetoothCustomOpenScale.java is the only comm channel. When I used nRF COnnect to send a "0" command I can read the characteristic back with device info.

Any ideas or suggestions would be most helpful. I'm happy to share ESP32 code and info afterwards. I'll probably put in my github as project. It would probably be nice if it emulated a Beurer scale instead of custom but I thought starting with "custom" made sense for simplicity.

oliexdev, 2019-10-12:
I made it simply. I transmit all saved data from my custom scale to openScale every time the scale will go into the sleep mode, see [here](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale#reducing-the-power-consumption) for more details.
So you don't need to trigger a command to receive the data. Well, it was **my** design decision, you can implement it differently of course.

I would be (and I think other too) interested in your ESP32 implementation, so give me a note if you have published it. :smile:

neocogent, 2019-10-18:
Thank you for info. I was away a bit and hadn't done any work on my scale. 

I didn't realize that it sent data when going to sleep but that's very helpful. I haven't written code yet for users and logging of weigh ins so have no historic data. So for now I'll try some code to just send data after a stable weighing. What confused me is that nothing ever shows up in the app when I send a data packet - no data points on graph or pop msg etc. I may be doing something wrong so I'll play some more. The BT service/class stuff is a bit heavy and I tend to cut reading docs as much as I can so I may be missing details.

The main reason I wanted to go via your app was to sync data to MyFitnessPal for weigh ins. But that's rather convoluted going via Google Fit and so I've got a different idea now. I will add some code to go direct to MyFirnessPal via their web api. There is a python module [available](https://github.com/coddingtonbear/python-myfitnesspal) (not official) that I've tested form python console and it works for updating weight. So I just need to adapt the http/api calls to the ESP library and add some scale screen options. 

I'll probably still want my scale to be able to send data to your scale app for completeness but I'm not personally into all the weight graphs and stats etc. Anyway, I'll add a note here when I get the code up in my repo. It'll be a while as I'm changing stuff around to better handle power down. Turns out the CP2102 serial chip on the ESP32 dev board draws about 15ma even if I deep-sleep the ESP32. That's enough to wipe out my 600ma LiPo battery in a couple days. Not great. For now I added a switch but I'll have to come up with a better option if I want it to go to sleep and wake up with scale pressure (ideal).

oliexdev, 2019-10-20:
ok then I will close this issue. If you have something to show give a hint here for reference.

closed 2019-10-20 by oliexdev