oliexdev/openScale #506: Customize export

robinx999, 2019-10-13:
Hi,
I have a problem with the exported data (it is on a german phone with german language settings this might be a problem), or maybe I am just to stupid to use the data
The Export looks like this
```
"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"
0.0,2.9,0.0,0.0,0.0,0.0,0.0,"","13.10.2019 06:29",17.6,0.0,0.0,40.5,0.0,0.0,0.0,0.0,62.7,69.3
```

When I try to import that into Libreoffice I have a little Problem with the values.
If I set the input to English. everything gets imported correctly but the date. If I set the input to German only the date is correctly imported.
The Problem seems to be that the date is formated in a way that a german would do. the numbers are formated in a way am english person would do (germans use a comma as a seperator not a dot)

I would like to be able to export the values in a way that I can import it to libre offce
Piping the file through this
`|tr "," ";" | sed 's/;\([0-9]\+\)\.\([0-9]\)/;\1,\2/g'`

Resulted into this
```
"biceps";"bone";"caliper1";"caliper2";"caliper3";"calories";"chest";"comment";"dateTime";"fat";"hip";"lbm";"muscle";"neck";"thigh";"visceralFat";"waist";"water";"weight"
0.0;2,9;0,0;0,0;0,0;0,0;0,0;"";"13.10.2019 06:29";17,6;0,0;0,0;40,5;0,0;0,0;0,0;0,0;62,7;69,3

```

and that I could import using german language settings.

I would like to generate a file with either complete english or complete german layout so that I can ignore the sed system and import it directly

greeting

oliexdev, 2019-10-13:
I understand your problem. The whole date, number representation is quite complex if you want it compatible world wide. You are right it is inconsistent using the German date time format and the English numbers format. But what is the right English date format? British English or American English? What is with the Japanese date format or Arabic?
I could set the date format into the ISO 8601:2004 format `yyyy-MM-dd HH:mm` if that would help?

robinx999, 2019-10-13:
Hi,

> I understand your problem. The whole date, number representation is quite complex if you want it compatible world wide. You are right it is inconsistent using the German date time format and the [...]

Have checked it I converted 

> 0.0,2.9,0.0,0.0,0.0,0.0,0.0,"","13.10.2019 06:29",17.6,0.0,0.0,40.5,0.0,0.0,0.0,0.0,62.7,69.3

to 

> 0.0,2.9,0.0,0.0,0.0,0.0,0.0,"","2019-10-13 06:29",17.6,0.0,0.0,40.5,0.0,0.0,0.0,0.0,62.7,69.3

An Libreoffice did recognized it when I set the import language to 'English (USA)'

I think that solution would be perfect, the rest of the converting is done perfectly by Libreoffice (showing everything in German format), but the import stumbles when languages are mixed, or I had to create a custom language for Libreoffice, that uses the german date format and a dot instead of comma as decimal

greeting

oliexdev, 2019-10-13:
ok, I have only the problem that if I change the date format older CSV files can't be imported anymore. I need to find a way to allow both date formats to be imported or I accept that older CSV needs to be converted manually.

robinx999, 2019-10-13:
Ok that seems to be a problem, and I fear many people would not like it if the csv format needs to be converted. Not sure if a check if a dash is in the date field could be the solution to the problem, if there is a dash the format is "ISO 8601:2004" if there is no dash the format could be treated as the old format, could that work?

oliexdev, 2019-10-13:
Unfortunately this will not work, they would interpret it as an invalid date format.

oliexdev, 2019-10-27:
In the latest dev version the CSV date format is now set to the ISO 8601:2004 format and the old format dd.MM.yyyy will be still supported for backward compatible.

closed 2019-10-27 by oliexdev