oliexdev/openScale #507: Openscale crashes to error message after every reading

barry1945, 2019-10-14:
**Describhe bug*.

**To Reproduce*
1. Open app and wait for scale to recognise user.
2. Take reading.
3. As soon as Bluetooth connection closes app crashes wit an image of a dead bug.

**Debug log**
Build version: 2.1.1-pro 
Build date: 1979-11-30 00:00:00 
Current date: 2019-10-14 08:34:24 
Device: Samsung SM-G920F 
 
Stack trace:  
java.lang.IllegalArgumentException: callback cannot be null
	at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:226)
	at com.welie.blessed.BluetoothCentral.stopScan(BluetoothCentral.java:509)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.disconnect(BluetoothCommunication.java:235)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$4.run(BluetoothCommunication.java:428)
	at android.os.Handler.handleCallback(Handler.java:751)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6682)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)

oliexdev, 2019-10-14:
thanks I already recognized the bug before, should be fixed in the next release

oliexdev, 2019-10-20:
should be fixed in the openScale version 2.1.2

closed 2019-10-20 by oliexdev