oliexdev/openScale #508: No connection to the Sanitas SBF70

lattice191011, 2019-10-14:
I'm using Openscale until version 2.1.0 with Sanitas SBF70 without problems. 
After updating to 2.1.1 I did not got connections to the scale with the comment "maximale Anzahl
gleichzeitiger Benutzer erreicht". I reseted the scale and after I had connected for the first time to the scale datasets were transfered.
With the next connection to the scale Openscale gives me the same user message: "maximale Anzahl gleichzeitiger Benutzer erreicht".

In the debug-log I finded the same comment:
...
2019-10-07 14:22:47.491 Debug [1] BluetoothBeurerSanitas: Cannot create additional scale user (error 0x03)
2019-10-07 14:22:47.497 Debug [1] BluetoothBeurerSanitas: Send disconnect command to scale
...

One time Openscale also crashed with the followed error messages:

Build version: 2.1.1 
Build date: 1980-01-01 00:00:00 
Current date: 2019-10-11 19:47:34 
Device: Samsung SM-G900F 
 
Stack trace:  
java.lang.IllegalArgumentException: callback cannot be null
	at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:185)
	at com.welie.blessed.BluetoothCentral.stopScan(BluetoothCentral.java:509)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.disconnect(BluetoothCommunication.java:235)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$4.run(BluetoothCommunication.java:428)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:158)
	at android.app.ActivityThread.main(ActivityThread.java:7224)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
	
On F-Droid a downgrade to version 2.1.0 was not possible.

oliexdev, 2019-10-14:
please reset the scale first and attach the complete debug log

oliexdev, 2019-10-20:
should be fixed in the openScale version 2.1.2

closed 2019-10-20 by oliexdev