oliexdev/openScale #510: Suport for low cost Dr.meter scale

brantic, 2019-10-19:
I am new at this so please for forgive me if my questions do not make sense.
Can you app connect to Dr.meter scale?
Can I change your code to try to measure weight of my propane tank and be able to change how your app looks?

https://www.amazon.com/Bluetooth-Dr-meter-Composition-Conductive-Surface-Elegant/dp/B07FS5WLXS/ref=asc_df_B07FS5WLXS/?tag=bingshoppinga-20&linkCode=df0&hvadid={creative}&hvpos={adposition}&hvnetw=o&hvrand={random}&hvpone=&hvptwo=&hvqmt=e&hvdev=c&hvdvcmdl={devicemodel}&hvlocint=&hvlocphy=&hvtargid=pla-4584070139979007&psc=1

oliexdev, 2019-10-20:
>  Can you app connect to Dr.meter scale?

Currently not, you have to reverse engineer this scale, see the wiki for more information

> Can I change your code to try to measure weight of my propane tank and be able to change how your app looks?

Sure if you comply with the GPLv3 license.

brantic, 2019-10-21:
Thank you for the answers.  I'll try to do it myself, but if I can't get it done alone perhaps we can work together.

Best regards,

Bob

oliexdev, 2019-10-25:
sure if I have time. I will close it temporally until you have something.

closed 2019-10-25 by oliexdev