oliexdev/openScale #513: Yunmai Mini unreliable connection initialization

palmtop, 2019-11-05:
I am using the Pro version with a Yunmai Mini. The problem is that when I start the program, it often fails to connect to the scale. Then I have to tray again (by clicking the bluetooth icon) and in some cases it takes 3-4 times to get a working connection.

What I figured out, that if before opening openScale I run BLEScan, until the scale appears, then openScale connects at once. May be the program could be improved, by doing a search, before doing the actual connection.

oliexdev, 2019-11-06:
openScale do a BLEScan before a connection if you enable GPS and grant access to the fine location, see the FAQ for more details. So did you do this?

tegger2000, 2019-11-16:
I have the same problem, but I can't connect anytime. I deleted all files and reinstall over fdroid. Fresh Openscale didn't find a device. Samsung s9+ with Android 9

tegger2000, 2019-11-16:
2019-11-16 09:53:40.775 Debug [2] AboutPreferences: Debug log enabled, openScale v2.1.2 (47), SDK 28, samsung SM-G965F
2019-11-16 09:53:40.778 Debug [2] AboutPreferences: Selected user id(1) name...

2019-11-16 09:53:49.018 Info [2] BluetoothCentral: scan started
2019-11-16 09:53:49.056 Error [2] BluetoothCentral: scan failed with error code 2
2019-11-16 09:53:59.169 Debug [2] BluetoothCentral: scanning timeout, restarting scan
2019-11-16 09:53:59.176 Info [2] BluetoothCentral: scan stopped
2019-11-16 09:54:00.191 Info [2] BluetoothCentral: scan started
2019-11-16 09:54:00.194 Error [2] BluetoothCentral: scan failed with error code 2
2019-11-16 09:54:09.028 Info [2] BluetoothCentral: scan stopped
2019-11-16 09:54:12.977 Info [2] BluetoothCentral: no scan to stop because no scan is running
2019-11-16 09:54:16.671 Debug [2] AboutPreferences: Debug log disabled

weliem, 2019-11-16:
You are getting an error: 'scan failed with error code 2'. So the scanner for devices is not starting...

So obviously you cannot connect to the device. Did you grant the Location permission?

Can you publish the full log?

tegger2000, 2019-11-16:
this was the full log, i only cutted the personal part with user data

the location permission is set

[https://www.bilder-upload.eu/bild-0ef4af-1573900569.jpg.html](url)

weliem, 2019-11-16:
Can you try if rebooting your phone resolves the issue?

tegger2000, 2019-11-16:
yes, i've got problems with other bluetooth devices, so i rebooted the phone and found the scale after data reset. So now its working, thx

palmtop, 2019-11-19:
My problem seems to be different from tegger2000-s problem.

The fine location is enabled, that should not be the problem. I am attaching the log, the connection problem line is the following:

2019-11-19 06:34:48.003 Info [18721] BluetoothPeripheral: connection failed with status 'GATT ERROR' (TIMEOUT)

[openScale_2019-11-19_06-34.txt](https://github.com/oliexdev/openScale/files/3862170/openScale_2019-11-19_06-34.txt)

(I have removed my birthday and age from the log)

In this case I had two unsuccessfull connection attempts and the third time it was successful.

If I run BLEscan before openScale the connection is always successful the first time. 

I was thinking that may be, when I do not open BLEScan it starts a scan in the background and connects to the scale, hindering openScale to open the connection?

weliem, 2019-11-20:
Well, it seems the scale simply doesn’t react to the connect request and only the last time it succeeds...

tegger2000, 2019-11-21:
i still have problems after some hours after restart, then i need to restart my s9+, otherwise i can't connect to the scale

oliexdev, 2019-11-23:
Did you always turn on the scale before you connect to the scale?

tegger2000, 2019-11-23:
various, it doesn't matter. I tried also a search and the app didn't find any bluetooth device.

oliexdev, 2019-11-28:
I guess there is something other wrong with your smartphone or your scale The search method use only basic Bluetooth mechanism. Do you have problems with the vendor scale app too? Or other Bluetooth devices?

I will close this issue here until we find out that is really a openScale problem.

closed 2019-11-28 by oliexdev

palmtop, 2019-11-28:
I agree with closing the issue, as I believe that this is more specific to the scale, than to the program.

To your previous question, the Yunmai mini is "always on", I do not need to switch is on. 

Recently I had some cases when even if I ran BLEScan the connection was not successful.

I can not test it with the original application as Yunmai decided to lock this type of device to the Chinese application, which I am not able to handle.

On the program side one improvement could be done, that if it receives a GATT error, then it retries the connection for approximately 30 more seconds, then I don't have to do this by hand.