oliexdev/openScale #514: (wrong) message "Max. concurrent users reached" on Silvercrest / Sanitas SBF75

agrimpelhuber, 2019-11-10:
**Describe the bug**
When the birth year of a named user registered on the scale differs from the birth year of a user with the same (short) name registered on openScale, the syncing of weight values stops with the toast message "Max. concurrent users reached". The log file reveals (in my case) that the user "XXX" (1980) is pulled from the scale, when the user "XXX" registered in openScale was born xx.xx.1968.

Using the original software "Healthforyou" to edit the scale user to birth year 1968 fixed the issue, but I lost weight data, since "Healthforyou" erased it from the scale when editing the user.

I don't know how the change of the user's birth year on the scale took place, because I have been using openScale exclusively and successfully for over 4 months now, when the issue suddenly appeared.

Obviously, the SBF75 and probably other Beurer / Sanitas / Silvercrest scales use the name as the (only) user-identifying data, but openScale doesn't. OpenScale's attempt to create a new user with a different birth year fails, see below.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a user on the scale with birth year 1980 and short name XXX, using HealthForYou
2. Create a user with the same name XXX, but a different birth year 1968 in openScale
3. Try to sync - log should show the following messages, even though only one user exists both in openScale and on the scale:

```
AboutPreferences: Selected user id(1) name(XXX) [...] 1968
[...]
BluetoothBeurerSanitas: Received user 1/1: XXX (1980)
[...]
BluetoothBeurerSanitas: Create user: XXX
[...]
BluetoothBeurerSanitas: Cannot create additional scale user (error 0x03)` 
```

Note: Condition 2 probably only appears in real life when *changing* one of the birth dates, or if they are changed by accident

Reproduced with current "pro" version.

**Expected behavior**
Even though I have fixed this for me, and this might not be strictly a bug, openScale could possibly resolve this issue with a number of software options:
1. create a more concise error message, pointing out the real conflict
2. offer to synchronise the birth years, either way
3. offer to ignore the difference, identify users by name only, and sync without creating a new user

**Debug log**
No debug log available, because I have already fixed the issue. In the debug log that I had when the issue occured, I actually anonymised my personal data - and later realized that the problem *was* in the personal data.

oliexdev, 2019-11-10:
First of all, using two apps for the same Bluetooth scale is never a good idea, both apps are getting confused because you can't track what the other app has changed on the scale and vice versa.

If you creating one user with the HealtForYou app and creating another user with the same name but with different birthday in openScale **it is** actually a new user because there is no reason do not have two users with the same name but with different users. I would recommend to reset the scale before using openScale.

But what currently doesn't work is updating and deleting a user on the scale with openScale. If you have change the user information or delete the user in openScale it has no effect on the scale. So you have to reset the scale.

closed 2019-11-21 by oliexdev

agrimpelhuber, 2019-11-24:
Sorry to contradict you, and sorry for the late reply - I was away. Just trying to make the app better, not criticising, or wishing for something that doesn't make sense.

1. Using the two different apps wasn't planned. I was using HealthForYou just for debugging. As I wrote, I used openScale exclusively and successfully for a few months, and also initialized the scale with it. I have no idea why the user birthdates suddenly differed - it just happened after not using the scale for a week, while I was on holiday.
2. Although I didn't do this myself (why should I?), a user might theoretically have forgotten to set the birth date in openScale first, and later might change it, resulting in 2 different dates.
3. My point was: Even if that problem cannot be solved programmatically, a nice hint would be nice. I'm not much of a programmer, but I guess that if would be just a quick check of the birth years after said error has occurred in order to display a more meaningful message.

But maybe pointing this out here will be enough - people will google the message, come here, and find the solution. 'nough done ...

As I said, no criticism - just trying to make a great app even better :-) Thanks!