oliexdev/openScale #515: Add image (take picture) to measurement

nightbluepn, 2019-11-11:
**Is your feature request related to a problem? Please describe.**
No. It would just be nice to visually track progression as well.

**Describe the solution you'd like**
Maybe a new measurement type 'photo' and a take photo button when adding a new measurement.
Tap on a given measurement to view the photo, maybe add a photo series feature.

**Describe alternatives you've considered**
This is obviously not a must have but would be great i think.

**Additional context**
If implemented, it would be nice if you could choose if the pics get saved in a private directory of the app or if it shows up in the gallery. Maybe export feature.

oliexdev, 2019-11-11:
Yes, I already thought about that feature but the main reason why it is currently not implemented is the backup problem. What is the best strategy? Where should the image saved on the flash or in the database? Should the images also included in the backup, if yes how? Currently the backup is a sql database...
How we can export the images with the measurements? Currently we export the data via CSV. The CSV file contains only text not binary data. Should we refer the links in the CSV to the images?
The backup files would also be much larger than now.

So many questions and yes somehow we can produce a working solution but the implementation should be also maintainable.

nightbluepn, 2019-11-11:
What is the issue with just keeping it very simple - seperate DB from images and store them on flash, just save a reference in the database. So you have DB and an image store. If a referenced file file is missing from the image store, display a toast with image missing or something.

Export as well as backup could have an option to include images, even an option to only export images.
If toggled, export CSV/SQL DB bundled with the image store as a folder, or just the folder.
The CSV can contain file references just like the internal DB.

This way the DB size is kept small and users can just decide to drop the attached pictures if they like without changing the DB itself (you would just get the image missing toast everytime or disable the measurement altogether).

oliexdev, 2019-11-11:
If I have more time I will try to implement it like with the file references but I am unsure also copying the images in a separate folder.