oliexdev/openScale #518: Reconnect to scale / provide

00pi, 2019-11-24:
Hi,

may thanks for the great app, I do like it very much!
I'm using a Mi Body Composition Scale 2 (XMTZC05HM) - the 2019 updated hardware with the glass top. It works as a charm from day one, I never used the vendor's Mi app for any kind of set-up (log attached).

For "data aquisition" I am using a wall mounted Android Tablet, that is permanently connected to the mains. The data is forwarded to my smart home system (fhem) via mqtt.

As discussed in #321 and #335 , openscale won't reconnect - which would be the intended behavior in my case. 
Currently I try to work around this issue using an  [https://play.google.com/store/apps/details?id=com.llamalab.automate](Automate flow) which starts openScale (and thus triggers the connect). Unfortunately starting an other app on my tablet disturbs the intended main use, to permanently show a full screen web browsing app.

-> Is it possible to provide a Android Intent or other kind of hook (sorry for the terminology, I am not an programmer) that can triggered using Automate's [App start block](https://llamalab.com/automate/doc/block/activity_start.html) start block?

Thanks,
Robert
[openScale_2019-11-24_11-04.txt](https://github.com/oliexdev/openScale/files/3883444/openScale_2019-11-24_11-04.txt)

oliexdev, 2019-11-28:
First of all, you can't run in newer versions of Android background task anymore. But maybe I can trigger it via MQTT in openScale sync would that help?

00pi, 2019-11-28:
Hi,

to trigger via MQTT does not help, as the scale needs to be discovered by the tablet - the Raspberry Pi hosting my home automation system is too far away. And I would prefer a "self contained" solution.
If triggering via MQTT (realized in a separate app OpenScale Sync) is possible, wouldn't the same interface be accessible to other apps (in my case: Automate)?
I am not sure I understand your statement "you can't run in newer versions of Android background task anymore" - almost every second app on my phone runs background sync processes to servers or devices.

oliexdev, 2019-11-28:
I don't know the app Automate but can you trigger via Automate or your MQTT Server MQTT packet?
You can read more about background task in Android at https://developer.android.com/guide/background

00pi, 2019-12-01:
Ok, so the Android background processing would be a major change of the OpenScale program, which is of course way beyond what I am looking for.

Your idea to send a MQTT message in case the scale is discovered is perfectly doable. So we could trigger the connect in OpenScaleSync via an incomming MQTT message. 
The outgoing path from OpenScaleSync via MQTT to fhem is already working fine.

oliexdev, 2019-12-04:
If I have time, I will look into this feature request.

00pi, 2019-12-11:
that's awesome. Thank you very much in advance!