oliexdev/openScale #525: "Value is out of range" for muscle mass

woj-tek, 2019-12-07:
(possibly duplicate of #288?)

**Describe the bug**
When entering measurements manually, when it comes to "Muscle mass" (in kg) I can't enter anything above 40,08kg (with weight 60,1kg). It seems like 66,5% is being the limit?

**To Reproduce**
Steps to reproduce the behavior:
- add new measurement with `+`
- type 60,10 (kg) as body weight
- try to enter 48,2 (kg) in muscle weight (as displayed in Mi app)

**Expected behavior**
I should be enter (almost) any value.

**Additional context**
No additional context - I'm just typing in manually measurements from my weight (Mi weight 2)

**Debug log**
Doesn't show anything interesting IMHO:
```
2019-12-07 10:42:07.266 Debug [1] AboutPreferences: Debug log enabled, openScale v2.1.2 (47), SDK 25, OnePlus ONEPLUS A3003
2019-12-07 10:42:07.269 Debug [1] AboutPreferences: Selected user id(1) name(Wojtek) … [details stripped]
```

It does look like form validation and it doesn't even register in the log.

oliexdev, 2019-12-09:
Currently you can enter max. 80% of your kg as muscle mass? Do you really have more muscle mass? :open_mouth:

woj-tek, 2019-12-11:
I wouldn't say so (judging from what I see in the mirror) but MiWeight says as much (60,2kg weight, 48,1kg muscle)... and complain that I have too little of muscle.

I'd say that this is some issue/quirk in MiWeight or MiFit app:
[image]
(apologies for Spanish in the screenshot, but I think it's legible)

woj-tek, 2019-12-12:
I was just pondering the values and I think that MiFit app is just wrong...

With weight of 60,4kg it says: "Body Fat 15,5% | Muscle: 48,3kg | Water: 57,9%"

In absolute values that would be 9,35kg, 48,3kg, 34,91kg, which would then add to 92,56kg. It doesn't add up percent-wise neither.

oliexdev, 2019-12-12:
what does openScale if your weight yourself? It doesn't have to sum up to 100% because in other body metrics are some water percents included.

woj-tek, 2019-12-12:
I just tried it and it says: 61,1kg weight, water 59,06%, muscle 48,88kg (what-the-heck?! I use kg here to match MiFit), Body Fat 13,91%.

In principle - more or less similar to what MiFit reports

woj-tek, 2019-12-13:
It just struck me that it would be good idea to get raw data so here's the debug log from the openScale: https://files.unir.se/f/0637c4f4c40c460ab0ba/?dl=1

I think this would be most relevant:
```
2019-12-13 08:29:04.873 Debug [1] BluetoothMiScale2: DataChange hex data: 02 24 E3 07 0C 0C 10 07 19 FD FF BC 2F
2019-12-13 08:29:04.877 Debug [1] BluetoothMiScale2: Impedance value is zero
2019-12-13 08:29:04.907 Debug [1] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Thu Dec 12 16:07:00 GMT+01:00 2019, WEIGHT: 61,10, FAT: 0,00, WATER: 0,00, MUSCLE: 0,00, LBM: 0,00, WAIST: 0,00, HIP: 0,00, BONE: 0,00, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
2019-12-13 08:29:05.192 Debug [1] BluetoothMiScale2: DataChange hex data: 02 24 E3 07 0C 0C 17 38 09 FD FF 6C 2F
2019-12-13 08:29:05.194 Debug [1] BluetoothMiScale2: Impedance value is zero
2019-12-13 08:29:05.196 Debug [1] BluetoothMiScale2: DataChange hex data: 02 26 E3 07 0C 0D 08 1C 01 17 02 30 2F
2019-12-13 08:29:05.196 Debug [1] BluetoothMiScale2: impedance value is 535.0
2019-12-13 08:29:05.211 Debug [1] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Thu Dec 12 23:56:00 GMT+01:00 2019, WEIGHT: 60,70, FAT: 0,00, WATER: 0,00, MUSCLE: 0,00, LBM: 0,00, WAIST: 0,00, HIP: 0,00, BONE: 0,00, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
2019-12-13 08:29:05.250 Debug [1] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Fri Dec 13 08:28:00 GMT+01:00 2019, WEIGHT: 60,40, FAT: 15,60, WATER: 57,90, MUSCLE: 80,02, LBM: 0,00, WAIST: 0,00, HIP: 0,00, BONE: 2,64, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
```

So it kinda looks similar to what MiFit shows (still weird).

Btw. for some reason to get openScale to get measurements I have to step on the weight, let it turn off and step on it again and only then it gets the measurement.

oliexdev, 2019-12-13:
Maybe your scale battery is low? As you get the same result as the original app you have to ask the vendor xiaomi why you get such wired values?

> Btw. for some reason to get openScale to get measurements I have to step on the weight, let it turn off and step on it again and only then it gets the measurement.

That could be possible because only the history measurements are fetched.

closed 2019-12-13 by oliexdev

woj-tek, 2019-12-13:
I replaced the batteries about 2-3 weeks ago so I don't think they should pose a problem.

At any rate - original question/issue was related to imposed ranges - would it be possible to have option "Ignore ranges" in settings so I could put whatever values I get, ignoring whether they make sense or not?

oliexdev, 2019-12-15:
I increased the max muscle percent to 100% but if a person has 100% muscle mass I want a picture of that person :wink:

woj-tek, 2019-12-15:
Thank you! (also, the current readings are bonkers already ;-) )

Interestingly enough, when reading directly from weight the values are not capped by the limits (or it seems so at least)