oliexdev/openScale #527: Full range finger zooming (day, month, year) in overview graph

francescosoave, 2019-12-10:
**Is your feature request related to a problem? Please describe.**
I don't like how the scaling of the graph in overview works (when changing between week, months, year from the burger menu on the right). When I change it from the menu the whole screen goes fully blank for a long time before showing the updated version with the new setting applied (not very nice, every time it seems like if the app crashed!). In addition I feel like this is not very flexible approach when using it

**Describe the solution you'd like**
I'd like to suggest to increase the zoom out range in the graph in the overview. Basically I'd like to be able to see in the graph the general line of weight over very long period of times just by zooming in/out with fingers (not from menu). I think this would be more flexible than the current solution. Basically a dynamic zoom that covers all the possible ranges just with finger zooming and possibly remembering the state of the last time I've used the app.

**Describe alternatives you've considered**
Haven't thought of other solutions since currently there is a solution but I think it could be improved

**Additional context**

oliexdev, 2019-12-12:
Did you try to zoom out in the overview chart with the fingers when the Rolling chart is enabled? If you set to day you can completely zoom out all your data if you want. If you have a lot of data it could be time consumption.

francescosoave, 2019-12-14:
hello! thanks that actually did it! Sorry I didn't realize there was this option. Tt works when Rolling chart box is unticked. Thanks

closed 2019-12-15 by oliexdev

bobmorane06, 2022-02-21:
I do not find any Rolling chart box on the version I just installed, v2.4.5 (60), is this a feature from a previous version? 

If so, how could I zoom out to see all data rather than by year? 

If it is available in v2.4.5, where do I find the Rolling chart checkbox?

Congrats for a very nice app!