oliexdev/openScale #529: Improved support for quickly entering birthdate

joosbuijsNL, 2019-12-22:
I just downloaded OpenScale and of course wanted to create my profile. However, entering your birth date is not really convenient as you have to scroll month-by-month (in my case 15*12=180 times). 

Ideally I could enter my birth date using the numeric keyboard.
Less ideal but still better would be drop-downs for day/month/year.

oliexdev, 2019-12-22:
No you haven't. Click on the year on the top and then you can select the year. It is the default Android date time picker but maybe I will find a better one.

joosbuijsNL, 2019-12-22:
You're right, of course after I went back to 1984 the hard way already ;)

You can close this one.

oliexdev, 2019-12-22:
I changed the date picker dialog into a spinner view, so it should be more clearly.

closed 2019-12-22 by oliexdev