oliexdev/openScale #531: Support for BF-850

oji, 2019-12-27:
**Scale name**
Beurer BF-850

Hi, 
I got a BF-850 and it's not supported. It would be nice to add it to the supported list. I'll try to get the required traffic scans as soon as I can.

Thanks!

dror3go, 2020-09-27:
Hi @oji - were you able to extract the traffic sample?

krisjans, 2021-07-26:
I made support for BF600 and BF850. 

If anyone interested to test it-  this is updated build of #753: [openScale-2.3.5-debug_bf600_bf850_bf950.zip](https://github.com/oliexdev/openScale/files/6877239/openScale-2.3.5-debug_bf600_bf850_bf950.zip)

@oji Do you still have BF850? Would you like to test debug build (link above) and let me know if it works? (and provide debug log if it does not OR if app crashes).

krisjans, 2021-07-28:
If someone is interested in testing my latest changes related to [BF600](https://github.com/oliexdev/openScale/issues/219), [BF850](https://github.com/oliexdev/openScale/issues/531), [BF950](https://github.com/oliexdev/openScale/issues/502) and [SBF77](https://github.com/oliexdev/openScale/issues/681), then latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk), change log in #753.

If something does not work or you encounter a crash - I would highly appreciate if you leave me a comment with attached openScale debug log (if possible also  btsnoop_hci.log. [Instructions how to get logs.](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic))

oliexdev, 2021-10-10:
see #753

Nataffko, 2024-02-21:
Hello!

Firstly, I want to thank everyone involved in the project. I think this is the best app for scales.

I have a Beurer BF850 and I encountered a small problem with the display of readings. On the scales screen, water is displayed as a percentage, but the scales give the app water values ​​converted to kg. For myself, I created a new driver for BF850 from BluetoothBeurerBF600.java, adding the conversion of kg to percentage from BluetoothBeurerBF105.java. Now the values in the app are slightly different from the value on the scale screen, apparently due to the double conversion percentage-kg-percent, but it looks much better than the 31% water that was in the app.

debug excerpt with values on the new driver:
`2024-02-20 14:18:22.804 Debug [2] BluetoothPeripheral: writing <0201c409> to characteristic <00002a9f-0000-1000-8000-00805f9b34fb>
2024-02-20 14:18:22.836 Debug [2] BluetoothCommunication: SUCCESS: Writing <02 01 C4 09> to <00002a9f-0000-1000-8000-00805f9b34fb>
2024-02-20 14:18:26.084 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() value: [0E EE 32 E8 07 02 14 0E 10 1E 01 E2 00 A4 06]
2024-02-20 14:18:26.086 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() flags: 0x0e [SI, timestamp, userID, bmiAndHeight], reserved flags: 0x00 
2024-02-20 14:18:26.087 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() weight: 65.189995
2024-02-20 14:18:26.087 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() timestamp: Tue Feb 20 14:16:30 GMT+03:00 2024
2024-02-20 14:18:26.088 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() scale user index: 1 (app user id: 1)
2024-02-20 14:18:26.089 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() BMI: 22.6
2024-02-20 14:18:26.089 Debug [2] BluetoothStandardWeightProfile: weightMeasurementToScaleMeasurement() heightInMeters: 1.7
2024-02-20 14:18:26.090 Debug [2] BluetoothStandardWeightProfile: Got weight: 65.189995
2024-02-20 14:18:26.136 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() value: [98 03 2F 01 10 16 4F 01 34 21 66 18 E2 14]
2024-02-20 14:18:26.136 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() flags: 0x398 [SI, bmr, musclePercentage, softLeanMass, bodyWaterMass, impedance], reserved flags: 0x0000 
2024-02-20 14:18:26.136 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() bodyFatPercentage: 30.300001
2024-02-20 14:18:26.137 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() bmrInJoules: 5648 bmrInKcal: 1349
2024-02-20 14:18:26.137 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() musclePercentage: 33.5
2024-02-20 14:18:26.137 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() softLeanMass: 42.5
2024-02-20 14:18:26.138 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() bodyWaterMass: 31.23
2024-02-20 14:18:26.138 Debug [2] BluetoothStandardWeightProfile: bodyCompositionMeasurementToScaleMeasurement() impedance: 534.60004
2024-02-20 14:18:26.138 Debug [2] BluetoothStandardWeightProfile: Got body composition: 98 03 2F 01 10 16 4F 01 34 21 66 18 E2 14
2024-02-20 14:18:26.141 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Tue Feb 20 14:16:30 GMT+03:00 2024, WEIGHT: 65,19, FAT: 30,30, WATER: 47,91, MUSCLE: 33,50, LBM: 45,44, WAIST: 0,00, HIP: 0,00, BONE: 2,94, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT:`

I didn’t fully understand what exactly I was doing, but it works, so I’ll attach my driver)
[BluetoothBeurerBF850.txt](https://github.com/oliexdev/openScale/files/14358446/BluetoothBeurerBF850.txt)