oliexdev/openScale #533: Body fat change in pounds / kg

silberberg, 2019-12-31:
Enhancement suggestion / request:  While openScale calculates changes in body mass and body fat percentage, as far as I can tell it doesn't calculate change in pounds (or kg) of fat.

That is: `delta = (Body Mass 2 * Fat % 2) - (Body Mass 1 * Fat % 1)`
Note: % means percentage, not modulo

I could probably send a pull request and work on this myself, but I don't know Java.

Thank you for creating this code!
Steve

oliexdev, 2020-01-02:
You can change the body fat in % to kg/lb/st if you click on the fat measurement in the settings->measurements and disable the percentage option. Do you mean that?

silberberg, 2020-01-02:
OMG yes thank you!  One more question.  Does the app work with the RENPHO scale?  It's not in the list, but maybe someone has reverse engineered the bluetooth

oliexdev, 2020-01-03:
not yet, see issue #291

closed 2020-01-03 by oliexdev