oliexdev/openScale #535: Graph: 31 december measurement is first of year instead of last

OverkillGuy, 2020-01-10:
In the graph view, 31 December was the firstpoint in the graph instead of last, because it's (incorrectly) marking as Week number 1 (likely a [ISO 8601 week number issue](https://www.youtube.com/watch?v=D3jxx8Yyw1c)).

Steps to reproduce the behavior:
1. Daily weight measurements spanning end of year, including 31st December
2. Look at "month view" weight graph
3. Notice 31 december is FIRST point (left-most, "Week 1" side) instead of right

_Will update bug with screenshot ASAP_

Shown using F-Droid version, not reproduced yet.

**Expected behavior**
31st December is last date = further right in the graph.

**Additional context**
Sounds very related to the "Week Number" of 31st December being marked as Week 1 of the year, which sounds like a small misunderstanding while using ISO 8601 Week number. See [Tom Scott's video](https://www.youtube.com/watch?v=D3jxx8Yyw1c) for some related context.

oliexdev, 2020-01-11:
Thanks, you are absolutely right, so far I have not found an easy solution, except to replace Java Calendar implementation with  the [joda time library](https://github.com/JodaOrg/joda-time).

OverkillGuy, 2020-01-15:
I see. I am perfectly happy wontfix-ing this issue, it's a tiny corner case that doesn't bother people much.
 It just made me smile to notice this curiosity, so I thought to report it. Hope it made you smile too.

oliexdev, 2020-01-19:
true its a tiny problem but this question will be reported every year, so would be great if somebody could fix it.