oliexdev/openScale #543: 2.2.0 release displays wrong data for muscle mass

dobey, 2020-01-19:
**Describe the bug**
After updating to the new 2.2.0 release on my phone, I've noticed the muscle mass column in the table view now shows the exact same value as the weight column, for all scale readings. After exporting the database and examining with sqlite3, the database entry still has a presumably correct value.

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest release from f-droid
2. Load openScale in table view with muscle mass column displayed, with valid readings
2b. Connect to scale for new readings if needed
4. Notice muscle mass now matches weight column

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): No

**Expected behavior**
I expected muscle mass column to continue showing muscle mass data.

oliexdev, 2020-01-22:
@dobey for which scale? Could you attach the openScale log file?

dobey, 2020-01-23:
It is with the Xiaomi Mi Body Composition Scale (which I guess is Mi Scale v2). I don't have a debug log, but I guess I can try to get one; though I'm not sure it would have any useful info.

I tried looking through the changes in git, and couldn't find anything related. The only change I can see was a change to have 100% of body weight possible instead of 80% in the percent related calculations with it. And, as I said, looking at the database itself, the data in the `muscle` column seems correct, in that it varies across the entries, and does not match the `weight` column, though I do not know what units it is stored in.

dobey, 2020-02-06:
I've tried to export a debug log, but it seems there isn't really much debug logging happening, and I see nothing especially related to this issue in the log file. I have no idea what units the muscle mass reading is in, from the scale, though the value seems to be about 150% of the weight kg value from the scale. Though I obviously can't be 150% of my weight in muscle mass alone.

obbardc, 2020-02-11:
For me my weight is 80kg, and the Xiaomi Body Composition Scale 2 shows 70% muscle mass in openScale readings. I'm probably going to attempt to update the scale firmware first.

obbardc, 2020-02-13:
FWIW my Xiaomi Body Composition Scale 2 has firmware version `1.0.0.12`.

The mi fit app output vs openScale:
- weight: 82.20kg, 82.05kg
- body fat: 25.9%, 25.11%
- water 50.7%, 51.37%
- visceral fat 10, 13.89
- muscle 57.7kg, 71.07%of82.05=58.31...kg
- protein 19.4%, n/a
- bone mass 3.1kg, 3.13kg

so, it looks as if openScale is reading the values as per the mi fit app.

dobey, 2020-02-13:
Where can one see the firmware version in openScale? It also seems that openScale does not log any information about the scale itself in the debug log? All I see is the openScale version, my phone model, and the user profile data which I set in openScale. No scale info or the raw readings.

obbardc, 2020-02-13:
That was all taken from the Mi fit app. Agreed that information should probably be logged in openScale, maybe that belongs on a different issue? :-).

oliexdev, 2020-02-27:
@dobey the firmware version is not shown in openScale. You have to read out for each the scale the firmware but the raw readings should be in the debug log file. Which firmware do you use?
@obbardc your are welcome to send a PR

closed 2020-02-27 by oliexdev

dobey, 2020-02-27:
@oliexdev I'm using whatever firmware came on the scale. Since there is no way in openScale to see what that is, I can't tell you a version. If there's no way to upgrade the firmware without installing, running, and logging into, the MiFit app, then that won't change.

The raw readings do not appear in the debug log, or at least it's unclear how to make them be there if they should, in the UI.

dobey, 2020-03-04:
I think maybe for some reason the scale is giving the muscle measurement in either catty or lbs, because if I divide the raw value by 2 or 2.2, then it's around 70-75% of the raw kg value for weight, which makes a lot more sense. Is there some way to determine what actual units the measurements are in?

dobey, 2020-04-09:
So @oliexdev is there any way to figure out which unit is actually being given by the scale, and perform the math to get the correct value, instead of assuming it's in kg?

oliexdev, 2020-04-09:
@dobey sure you could analyze the Bluetooth packets which are send from the scale. You can use wireshark for this.