oliexdev/openScale #544: Support for PICOOC S1 Pro

pililac, 2020-01-20:
Hi together, I'm really liking my PICOOC S1 Pro, but the App is a disaster. The only possibility for export data is saving a JPG of the report. I recorded some BT traffic and 
also the openScale debug log file. I had a first look at the data with Wireshark, but I didn't find the values. Maybe someone with more experience can help?

**Scale name**

PICOOC S1 Pro

https://eu.picooc.com/en/product-s1pro.php

**Log Files**

User settings in the vendor app for all four measurements: 

jacob
male
Feb/10/1989
180 cm
Germany (Region)
White/Caucasian

No Fitness level was asked

**Measurement 1:**

BT Snoop Log: [btsnoop_hci_1.log](https://github.com/oliexdev/openScale/files/4088101/btsnoop_hci_1.log)

Measured true values: [measurement1.txt](https://github.com/oliexdev/openScale/files/4088102/measurement1.txt)

Screenshot: [measurement1](https://user-images.githubusercontent.com/60116646/72760774-2cdfdf80-3bda-11ea-9f51-10676f349818.jpg)

**Measurement 2:**

BT Snoop Log: 
[measurement2.txt](https://github.com/oliexdev/openScale/files/4088115/measurement2.txt)

Measured true values: 
[btsnoop_hci_2.log](https://github.com/oliexdev/openScale/files/4088116/btsnoop_hci_2.log)

Screenshot: [measurement2](https://user-images.githubusercontent.com/60116646/72760758-205b8700-3bda-11ea-9157-ae083e98e1f4.jpg)

**Measurement 3:**

BT Snoop Log: [measurement3.txt](https://github.com/oliexdev/openScale/files/4088120/measurement3.txt)

Measured true values: [btsnoop_hci_3.log](https://github.com/oliexdev/openScale/files/4088121/btsnoop_hci_3.log)

Screenshot: [measurement3](https://user-images.githubusercontent.com/60116646/72760745-16d21f00-3bda-11ea-8613-48e94c8d0596.jpg)

**Measurement 4:**

BT Snoop Log: [measurement4.txt](https://github.com/oliexdev/openScale/files/4088123/measurement4.txt)

Measured true values:[btsnoop_hci_4.log](https://github.com/oliexdev/openScale/files/4088124/btsnoop_hci_4.log)

Screenshot: [measurement4](https://user-images.githubusercontent.com/60116646/72760704-f99d5080-3bd9-11ea-96ab-9cc9478f9fd9.jpg)

**openScale debug log file**

[openScale_2020-01-20_22-29.txt](https://github.com/oliexdev/openScale/files/4088133/openScale_2020-01-20_22-29.txt)

All data in one zip: [PICOOC_S1.zip](https://github.com/oliexdev/openScale/files/4088134/PICOOC_S1.zip)

pililac, 2020-03-06:
Okay, ich found the date and the weight in the BT data. The scale sends data to the receiver via handle 0x0012 in the format `390a5e2614fe06b40f8c`, (taken from the four measurements 1). I found the following format:

`columns
| 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 |

39  0a  5e  26  14  fe  06  b4  0f  8c
39  0a  5e  26  15  ad  07  18  0f  be
39  0a  5e  26  16  52  06  f0  0f  aa
39  0a  5e  26  16  e8  06  ce  0f  c8`

col 3 - 6	[date]:

5e  26  14  fe	=	Mo, 20 Januar 2020 22:00:46 +0100
5e  26  15  ad	=	Mo, 20 Januar 2020 22:03:41 +0100
5e  26  16  52	=	Mo, 20 Januar 2020 22:06:26 +0100
5e  26  16  e8	=	Mo, 20 Januar 2020 22:08:56 +0100

Format: UNIX 32bit hex value big endian

col 7 & 8	

[weight]

06 b4		1716	/20 =  85.8
07 18		1816	/20 =  90.8
06 f0		1776	/20 =  88.8
06 ce		1742	/20 =  87.1

I don't know what the other columns are for, but maybe someone of you has an idea.

oliexdev, 2020-03-10:
From my experience the first two column are mostly the command handles. The last column could be a XOR checksum or with column 9 and 10 the impedance value!?

The body measurements are done in the vendors app. Unfortunately in a .so library.

tobiaspc, 2020-07-29:
Based on the openScale debug log file, to me it seems that this scale and the mini variant, see #401, have the same protocol. I also own the mini scale and I wonder what else needs to be done to support this scale. How can I help?

pililac, 2020-10-27:
It's looking like a dead end to me. When the calculations are done in the app and the .so library cannot be opened, I have no clue how to proceed.

jvkersch, 2023-09-20:
Just a naive question, but based on the research done by @pililac above, would it make sense to offer "partial support" for this scale, by recording only the weight and not the other (calculated) body measurements?

VladislavNikolaev, 2023-12-06:
I've decompiled the Picooc apk and found out the following peace of code. It contains the logic how they parse the byte array and get weight and impedance values, I hope it might help:

`
    private String parseBroadcastDataFrom3E(byte[] bArr) {
        if (bArr[0] - 1 == 8 && bArr[2] == 80 && bArr[3] == 73 && bArr[4] == 67 && bArr[5] == 79 && bArr[6] == 79 && bArr[7] == 67 && bArr[8] == 45 && bArr[9] == 76) {
            int i = bArr[0] + 1;
            byte[] bArr2 = new byte[18];
            for (int i2 = 0; i2 < 18; i2++) {
                bArr2[i2] = bArr[i + 2 + i2];
            }
            if (NumUtils.makeCheckSum(BluetoothUtils.bytesToHexString(bArr2, 18)).toLowerCase().equals(BluetoothUtils.bytesToHexString(new byte[]{bArr[i + 20]}, 1))) {
                byte[] bArr3 = new byte[6];
                for (int i3 = 0; i3 < 6; i3++) {
                    bArr3[i3] = bArr[i + 2 + i3];
                }
                String bytesToHexString = BluetoothUtils.bytesToHexString(bArr3, 6);
                float round = Math.round(((((bArr[i + 13] << 8) & MotionEventCompat.ACTION_POINTER_INDEX_MASK) + (bArr[i + 14] & 255)) / 20.0f) * 10.0f) / 10.0f;
                int roundFloatToInt = NumUtils.roundFloatToInt(((65280 & (bArr[i + 15] << 8)) + (bArr[i + 16] & 255)) / 10.0f);
                float round2 = Math.round(((((bArr[i + 17] << 8) & 32512) + (bArr[i + 18] & 255)) / 10.0f) * 10.0f) / 10.0f;
                byte b = bArr[i + 19];
                this.scannedDeviceForBroadcastDevice.setMac(bytesToHexString);
                this.scannedDeviceForBroadcastDevice.setWeight(round);
                this.scannedDeviceForBroadcastDevice.setImp(roundFloatToInt);
                this.scannedDeviceForBroadcastDevice.setUnit(b);
                PicoocLog.m2433i("lipeng", this.scannedDeviceForBroadcastDevice.getImp() + "imp");
                PicoocLog.m2433i("lipeng", this.scannedDeviceForBroadcastDevice.getWeight() + "weight");
                StringBuffer stringBuffer = new StringBuffer();
                stringBuffer.append("\nmac:").append(bytesToHexString).append(StringUtils.f5723LF);
                stringBuffer.append("体重（kg）:").append(round).append("      体重:(").append(b == 1 ? "lb" : "斤").append(")").append(round2).append(StringUtils.f5723LF);
                stringBuffer.append("电阻：").append(roundFloatToInt).append("      单位：").append((int) b);
                return stringBuffer.toString();
            }
            return "CS校验失败！";
        }
        return "数据不合法";
    }`

4ndv, 2024-03-21:
> I've decompiled the Picooc apk and found out the following peace of code. It contains the logic how they parse the byte array and get weight and impedance values, I hope it might help:

Looks like that piece of code is for "PICOOC-W" and "PICOOC-L" scales. These scales broadcast measurements without the need of first writing timestamp to characteristic. For Mini, S1 Pro, and other "PICOOC-TWO-LETTERS" devices the parsing logic is in BTBle.displayData, and impedance calculations are in WeightingDirect class