oliexdev/openScale #545: Fehler bei Sanitas Waage

WolframSchmidtRS, 2020-01-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

WolframSchmidtRS, 2020-01-25:
Nach der Verbindungsaufnahme kommt folgende Fehlermeldung:

Build version: 2.2.0-pro 
Current date: 2020-01-25 13:11:32 
Device: HUAWEI VOG-L29 
 
Stack trace:  
java.util.UnknownFormatConversionException: Conversion = ')'
	at java.util.Formatter$FormatSpecifier.conversion(Formatter.java:2781)
	at java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2811)
	at java.util.Formatter$FormatSpecifierParser.<init>(Formatter.java:2624)
	at java.util.Formatter.parse(Formatter.java:2557)
	at java.util.Formatter.format(Formatter.java:2504)
	at java.util.Formatter.format(Formatter.java:2458)
	at java.lang.String.format(String.java:2842)
	at com.health.openscale.gui.MainActivity$10.handleMessage(MainActivity.java:568)
	at android.os.Handler.dispatchMessage(Handler.java:112)
	at android.os.Looper.loop(Looper.java:216)
	at android.app.ActivityThread.main(ActivityThread.java:7625)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)

WolframSchmidtRS, 2020-01-25:
Es handel sich um die Sanitas Waage

oliexdev, 2020-02-27:
Could you attach the openScale log file? Check the FAQ to see how it works. Please also indicate which language your application has used.

closed 2020-03-08 by oliexdev