oliexdev/openScale #546: openScale sync regularly stops syncing to GoogleFit (error "GoogleFit sync failed")

PavelGorbanj, 2020-02-01:
After a day, openScale Sync stops syncing to GoogleFit (error "GoogleFit sync failed").
It reproduces on 4 devices for 2 users.

Steps to reproduce the behavior:
1. Configure synchronisation to Google Fit
2. Weight, check that new measure appeared in Google Fit
3. Wait for a day
4. Weight, open Sync and get error "GoogleFit sync failed" (see screenshots).
5. Open configuration screen in Sync, fix issue by clicking "Google sign in", then "Manual full sync"
6. Weight, check that new measure appeared in Google Fit
3. Wait for a day, open Sync and get error "GoogleFit sync failed" (see screenshots).
repeat:)

Reproduced with Sync version 0.3.1, Pro version 2.2.0-pro

**Expected behavior**
Sync doesn't require manual intervention.

**Additional context**
This was checked on 4 different Samsung devices (Android 8, Android 10) with 2 different accounts.
Both openScale Pro and Sync were bought/installed from Google Play.

**Debug log**
... in progress ...

oliexdev, 2020-02-27:
It is not a bug, your google account sign in expired. That has nothing to do with openScale. This could happen if you have more Google Accounts on your phone or you disconnect from your account in another app and so on.

PavelGorbanj, 2020-02-27:
> It is not a bug, your google account sign in expired. That has nothing to do with openScale. This could happen if you have more Google Accounts on your phone or you disconnect from your account in another app and so on.

This happens every other day!
No other app with google sign-in behaves like that on my phone (and I use such sign-in method a lot).

Also, I do not disconnect my account.

Are you still sure it's not a bug?

ivanfor, 2020-04-08:
Hi. I have the same behaviour than @PavelGorbanj 
It happens very frecuently. Probably more than once a day. I don't have any other google account on the phone either.
Is there anything we can do to help you debugging?

PavelGorbanj, 2020-04-08:
@ivanfor , hi, if you don't mind, can you pls re-submit your comment to #568 that I've created just now? assuming it's not only me who experience this bug, I think it should be kept open

oliexdev, 2020-04-09:
you can search for google silent login problem on google. Also attach the debug log to see more details as it is since February in progress...

PavelGorbanj, 2020-07-08:
Dunno if anything has changed in the Sync/Scale app that is published on GPlay, but since some time I see that GoogleFit was updated by Sync app without my intervention. 
At the same time, I still see "sign in failed" in the Sync app.

closed 2020-09-20 by PavelGorbanj