oliexdev/openScale #547: Huawei Body Fat Scale AH100

dmitriytim, 2020-02-02:
datetime 02-02-2020 20:57
Weight  82,6
BMI 24,8
FAT 22,1
Visceral FAT 11,5
Basal metabolic rate 1831
water 54,1
Bone mineral 3,3
protein 19,9
Muscles 62,0
[btsnoop_hci_20200202205551.cfa.txt](https://github.com/oliexdev/openScale/files/4145257/btsnoop_hci_20200202205551.cfa.txt)
[2020-02-02.zip](https://github.com/oliexdev/openScale/files/4145260/2020-02-02.zip)

oliexdev, 2020-02-27:
Please use the correct template and do **all** the steps probably.

KerstinMaur, 2020-06-18:
Hey, I am interested in support for that Huawei AH-100 scale too as I might purchase a smartwatch where they bundle that scale at the moment.
Can you please provide @oliexdev with the provided information, @dmitriytim? 🙏

dmitriytim, 2020-06-18:
edited

fonshartendorp, 2020-06-24:
I'm also interested in added support for the AH-100, but have zero experience in reverse engineering. Is the correct template used now and are all steps done properly? I'd be happy to help

diegoalbertopp, 2020-07-07:
And what happened?

klosteinmann, 2021-01-15:
Oh here's the original. Sorry for duplication so does the scale properly work without workarrounds?

meigor, 2021-03-23:
I also want to contribute to the support of the Huawei AH100 (and Honor CH100 sem to be the same, my is a Huawei AH100 but bordcast the CH100 as model). 
i tried to get the log file with adb bugreport _NAME extract it and convert with btsnooze.py [btsnooze info](https://source.android.com/devices/bluetooth/verifying_debugging#debugging-with-bug-reports) but wireshark can't read the file.

So i tried to get some information with the nRF app and got the following log:

```
nRF Connect, 2021-03-23
CH100 (5C:CA:D3:2C:7F:E4)

D	17:46:16.744	gatt.setCharacteristicNotification(00002a9c-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.745	gatt.setCharacteristicNotification(00002a9d-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.746	gatt.setCharacteristicNotification(00002a99-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.746	gatt.setCharacteristicNotification(00002a9f-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.748	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.748	gatt.setCharacteristicNotification(00002a2b-0000-1000-8000-00805f9b34fb, true)
D	17:46:16.748	gatt.setCharacteristicNotification(0000faa2-0000-1000-8000-00805f9b34fb, true)
I	17:46:16.812	Connection parameters updated (interval: 43.75ms, latency: 0, timeout: 5000ms)
V	17:46:17.891	Disconnecting...
D	17:46:17.891	gatt.disconnect()
D	17:46:17.903	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	17:46:17.903	Disconnected
D	17:46:18.932	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	17:47:49.861	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	17:47:49.861	gatt.close()
D	17:47:49.872	wait(200)
V	17:47:50.072	Connecting to 5C:CA:D3:2C:7F:E4...
D	17:47:50.072	gatt = device.connectGatt(autoConnect = true, TRANSPORT_LE, preferred PHY = LE 1M)
D	17:47:50.083	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	17:47:50.083	Connected to 5C:CA:D3:2C:7F:E4
V	17:47:50.084	Requesting new MTU...
D	17:47:50.084	gatt.requestMtu(517)
I	17:47:50.315	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I	17:47:50.818	MTU changed to: 23
V	17:47:50.818	Discovering services...
D	17:47:50.818	gatt.discoverServices()
D	17:47:50.829	[Callback] Services discovered with status: 0
I	17:47:50.829	Services discovered
V	17:47:50.845	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
Body Composition (0x181B)
- Body Composition Feature [R] (0x2A9B)
- Body Composition Measurement [I] (0x2A9C)
   Client Characteristic Configuration (0x2902)
Weight Scale (0x181D)
- Weight Scale Feature [R] (0x2A9E)
- Weight Measurement [I] (0x2A9D)
   Client Characteristic Configuration (0x2902)
User Data (0x181C)
- Age [R W] (0x2A80)
- Gender [R W] (0x2A8C)
- Height [R W] (0x2A8E)
- Database Change Increment [N R W] (0x2A99)
   Client Characteristic Configuration (0x2902)
- User Index [R] (0x2A9A)
- User Control Point [I W] (0x2A9F)
   Client Characteristic Configuration (0x2902)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Current Time Service (0x1805)
- Current Time [N R W] (0x2A2B)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000faa0-0000-1000-8000-00805f9b34fb)
[... 27 lines cut ...]
I	17:47:53.090	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BD-01-19-18
A	17:47:53.090	"(0x) BD-01-19-18" received
I	17:47:54.996	Connection parameters updated (interval: 20.0ms, latency: 0, timeout: 1000ms)
I	17:48:02.817	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BC-11-0E-9D-0E-DF-CE-4B-01-F2-A0-41-61-B8-36-23-64-41-E5
A	17:48:02.817	"(0x) BC-11-0E-9D-0E-DF-CE-4B-01-F2-A0-41-61-B8-36-23-64-41-E5" received
I	17:48:02.842	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BC-11-8E-EC-AB-06-A3-3B-0E-A7-67-8B-52-F9-DE-73-32-9D-EA
A	17:48:02.842	"(0x) BC-11-8E-EC-AB-06-A3-3B-0E-A7-67-8B-52-F9-DE-73-32-9D-EA" received
I	17:48:03.116	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BD-01-20-21
A	17:48:03.116	"(0x) BD-01-20-21" received
I	17:48:03.276	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BD-01-19-18
A	17:48:03.276	"(0x) BD-01-19-18" received
I	17:48:11.077	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BD-01-01-00
A	17:48:11.077	"(0x) BD-01-01-00" received
I	17:48:12.279	Notification received from 0000faa2-0000-1000-8000-00805f9b34fb, value: (0x) BD-01-19-18
A	17:48:12.279	"(0x) BD-01-19-18" received
```

Here are the UIDs and Datasets one example of the data from the user interface of the nRF app the TimeStamp:
[image]
[image]

here are also the Data from the Original Huawei App (the data measured at 17:48 while recording the bluetooth trace)
[image]

Tell me if I can help with something else.

cybergurke, 2021-12-21:
I'm interested on this scale too. Let me know if I can help

axcro7, 2023-02-19:
Happy to see it is partially supported now. Is there anything I could help with reverse engineering?

oliexdev, 2023-02-20:
@axcro7 sure you could try to reverse engineering the water, bone and muscle, see https://github.com/oliexdev/openScale/pull/873